MarlinFirmware/Marlin #14561: Cannot get TMC2209 to work on mega2560

EvilGremlin, 2019-07-09:
I want to use 3 drivers on single-pin UART, as should be possible with modified SoftwareSerial library (https://reprap.org/forum/read.php?415,848808,848808) but cannot connect even to single driver, getting "Testing X connection... Error: All LOW"
Is AVR initialization broken again in 2.0?

thinkyhead, 2019-07-11:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply. We'll use them in our testing.

teemuatlut, 2019-07-11:
Trying to support modified framework libraries isn't really something I want to get into.
Try to first get your drivers the normal way.

EvilGremlin, 2019-07-11:
Sorry, i was sure that i uploaded file...
Also, it doesn't work with standard library. I didn't get to play with modified yet.
[config_2209.zip](https://github.com/MarlinFirmware/Marlin/files/3383147/config_2209.zip)

pkilar, 2019-07-14:
I got Rumba board + 3x TMC2209 working on single HW serial (Serial3). Remember to configure unique address for each of the TMC2209 drivers using DIP switched on the board. Attached is my config.
[TMC2209_hw_serial.zip](https://github.com/MarlinFirmware/Marlin/files/3389968/TMC2209_hw_serial.zip)

EvilGremlin, 2019-07-14:
What commit/branch you running? Well, i'll try out HW serial anyway.

pkilar, 2019-07-14:
I'm running bugfix-2.0.x:
commit caa87d5fe1d497fadf797091199568d282a797a0 (HEAD)
Author: Giuliano Zaro <3684609+GMagician@users.noreply.github.com>
Date:   Thu Jul 11 05:04:16 2019 +0200

    SD2Card cleanup (#14571)

EvilGremlin, 2019-07-14:
Thanks, will try.

boelle, 2019-09-24:
@EvilGremlin is the issue still there?

boelle, 2019-10-21:
had a chance to play  arround with this one?

EvilGremlin, 2019-10-22:
8.10 software serial still didn't work in any form.

boelle, 2019-10-30:
8.10? is that arduino ide or some other tool i dont know of?

EvilGremlin, 2019-10-30:
it's a date

boelle, 2019-10-30:
ahh... i'm just a stupid european dude, we have date the other way round and almost always include year to make sure people get that its a date

ie 8/10-2019

free laughs on me for being slow :-D

EvilGremlin, 2019-10-30:
Guess i'm also stupid kinda european dude :) I forgot how default date writing customs differ from country to country.

boelle, 2019-10-30:
we could also write the same date as 19-10-8, there are a few other formats but i'm not that bored :-D

boelle, 2019-11-16:
@EvilGremlin i guess none of the updates in the past 17 days has helped any?

@teemuatlut do you have any clue why this fails?

EvilGremlin, 2019-11-16:
didn't update past 17 days, probably won't update for another 17 :)

EvilGremlin, 2019-11-21:
Ok, so i updaed and with all the changes it works now.

closed 2019-11-21 by EvilGremlin