MarlinFirmware/Marlin #14562: M600 or filament change feature activates at the beginning

HellsPriest, 2019-07-09:
Hello.

Recently I installed filament runout sensor.

I have uncommented advanced_pause_feature and nozzle_park_feature to make it work and then I booted it into my ender-3.

Expected behavior: printer start to print without any stops
Actual behavior: printer activates filament change feature after auto home

Steps to reproduce:
1) Start print
2) Wait till print do auto home
3)Check LCD with filament change command

(If you proceed with filament change feature it will purge the filament and then repeat it again.)

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3375020/Marlin.zip)

tonn0297, 2019-07-10:
you need to inverse the filament runout sensor

HellsPriest, 2019-07-10:
> you need to inverse the filament runout sensor

Like in the marlin or real sensor.
Hardware has arrow which shows filament direction.
Inverse is not possible because board won't fit into case.
I have switched 5v pin and sig pins (not on the board, but on the connector) to match it to my melzi board.

ManuelMcLure, 2019-07-10:
This is not a physical inversion - instead you need to set ``FIL_RUNOUT_INVERTING`` to ``true`` so that the logic is inverted.

HellsPriest, 2019-07-11:
> This is not a physical inversion - instead you need to set `FIL_RUNOUT_INVERTING` to `true` so that the logic is inverted.

Thank you, I will try it today.

HellsPriest, 2019-07-11:
Thank you guys, You've helped a lot.

First time it returned with error after compiling on arduino IDE 1.6.9 (something about exit status.
I tried to boot it from arduino 1.8.8 and it worked.

closed 2019-07-11 by HellsPriest