MarlinFirmware/Marlin #14564: [BUG] missing line in Marlin\src\pins\pins_BIGTREE_SKR_PRO_V1.1.h

RudyBenner, 2019-07-10:
I propose..

Marlin\src\pins\pins_BIGTREE_SKR_PRO_V1.1.h

[image]

It compiles much better now.

RudyBenner, 2019-07-10:
I propose:

[image]

RudyBenner, 2019-07-10:
should this be a PR?

thisiskeithb, 2019-07-10:
What would it accomplish?

Note that some pins were just cleaned up to remove the `BOARD_NAME` ifndef statement: https://github.com/MarlinFirmware/Marlin/commit/44c8f29e47bb9fc00dbb42e5305dd463338f4630

RudyBenner, 2019-07-10:
yes, I did see that this morning and the compile suffers, which is what I was trying to clean-up

RudyBenner, 2019-07-10:
try the latest, then try my proposal.

closed 2019-07-11 by thinkyhead