MarlinFirmware/Marlin #14565: [BUG] adding M43 will result in compile error

ShadowOfTheDamn, 2019-07-10:
[compile error.txt](https://github.com/MarlinFirmware/Marlin/files/3376040/compile.error.txt)

### Description

In the recent update of marlin after adding M43 to the configuration_adv.h I get compile error.

Just use the attached config files.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[compile error.txt](https://github.com/MarlinFirmware/Marlin/files/3376042/compile.error.txt)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3376053/Configuration_adv.zip)

Ludy87, 2019-07-10:
Bugfix: https://github.com/MarlinFirmware/Marlin/pull/14554

PR is released

ShadowOfTheDamn, 2019-07-10:
Yes, That works now.
thanks and closing the issue.

closed 2019-07-10 by ShadowOfTheDamn