MarlinFirmware/Marlin #14567: [BUG] RADDS + Due - Won't compile with SSD1306 OLED

DerThommi, 2019-07-10:
### Description

I want to add an I2C OLED Display so I uncommented the "#define U8GLIB_SSD1306" line and installed the library. It didn't compile so I reinstalled Arduino IDE and redownloaded Marlin 2.0, only changed the board to RADDS and uncommented the define line.

`Arduino: 1.8.9 (Windows 10), Board: "Arduino Due (Programming Port)"

sketch\src\lcd\ultralcd.cpp: In static member function 'static void MarlinUI::abort_print()':

sketch\src\lcd\ultralcd.cpp:1450:24: error: 'return_to_status' was not declared in this scope

       return_to_status();

                        ^

exit status 1
`

DerThommi, 2019-07-10:
When I uncomment #define OLED_PANEL_TINYBOY2:

```
Arduino: 1.8.9 (Windows 10), Board: "Arduino Due (Programming Port)"

In file included from sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi.cpp:65:0:

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:26: error: variable or field 'u8g_SetPIOutput_DUE' declared void

 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);

                          ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:26: error: 'u8g_t' was not declared in this scope

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:33: error: 'u8g' was not declared in this scope

 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);

                                 ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:46: error: expected primary-expression before 'pin_index'

 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);

                                              ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:28:25: error: variable or field 'u8g_SetPILevel_DUE' declared void

 void u8g_SetPILevel_DUE(u8g_t *u8g, uint8_t pin_index, uint8_t level);

                         ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:28:25: error: 'u8g_t' was not declared in this scope

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:28:32: error: 'u8g' was not declared in this scope

 void u8g_SetPILevel_DUE(u8g_t *u8g, uint8_t pin_index, uint8_t level);

                                ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:28:45: error: expected primary-expression before 'pin_index'

 void u8g_SetPILevel_DUE(u8g_t *u8g, uint8_t pin_index, uint8_t level);

                                             ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:28:64: error: expected primary-expression before 'level'

 void u8g_SetPILevel_DUE(u8g_t *u8g, uint8_t pin_index, uint8_t level);

                                                                ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi.cpp: In function 'uint8_t u8g_com_HAL_DUE_sw_spi_fn(u8g_t*, uint8_t, uint8_t, void*)':

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi.cpp:85:42: error: 'u8g_SetPIOutput_DUE' was not declared in this scope

       u8g_SetPIOutput_DUE(u8g, U8G_PI_SCK);

                                          ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi.cpp:90:44: error: 'u8g_SetPILevel_DUE' was not declared in this scope

       u8g_SetPILevel_DUE(u8g, U8G_PI_SCK, 0);

                                            ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

```

DerThommi, 2019-07-14:
No comment on this?

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

DerThommi, 2019-07-21:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3414492/Marlin.zip)

Don't really necessary because I only changed the board to RADDS and uncommented line 1978 in configuration.h to activate a generic OLED SSD1306, but here you have it. This is also the newest version of Marlin 2.0 I just download 5 minutes before posting this. The bug also appeared with the download from 11 and 7 days ago. A Marlin version from last year works fine.

So I'm using Arduino IDE 1.8.9 with U8glib from Oliver version 1.19.1, Arduino Due (Programming Port) Board selected.

boelle, 2019-07-21:
have you tried arduino ide 1.9 ? or vs code with platform io?

DerThommi, 2019-07-21:
VScode error:

```
Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing DUE (platform: atmelsam; board: due; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/due.html
PLATFORM: Atmel SAM 3.8.0 > Arduino Due (Programming Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinosam 4.3.190711
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing SlowSoftI2CMaster
SlowSoftI2CMaster @ 3a18be5 is already installed
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.4.5
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.1.3
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD> #6f53c19
|-- <SailfishRGB_LED> #4d0c888
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster> #3a18be5
|-- <SPI> 1.0
|-- <Wire> 1.0
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\DebugMonitor_Due.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\EepromEmulation_Due.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\G2_PWM.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\HAL.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\HAL_spi_Due.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\HAL_timers_Due.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\InterruptVectors_Due.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\MarlinSerialUSB_Due.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\MarlinSerial_Due.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\Servo_Due.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\Tone.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\persistent_store_eeprom.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\u8g_com_HAL_DUE_shared_hw_spi.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\u8g_com_HAL_DUE_st7920_sw_spi.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\usb\ctrl_access.c.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\usb\sysclk.c.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\usb\udc.c.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\usb\udi_cdc.c.o
In file included from Marlin\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi.cpp:65:0:
Marlin\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:26: error: variable or field 'u8g_SetPIOutput_DUE' declared void
 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);
                          ^~~~~
Marlin\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:26: error: 'u8g_t' was not declared in this scope
Marlin\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:26: note: suggested alternative: 'uid_t'
 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);
                          ^~~~~
                          uid_t
Marlin\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:27:33: error: 'u8g' was not declared in this scope
 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);
                                 ^~~
[... 91 lines cut ...]

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi_shared.h:28:64: error: expected primary-expression before 'level'

 void u8g_SetPILevel_DUE(u8g_t *u8g, uint8_t pin_index, uint8_t level);

                                                                ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi.cpp: In function 'uint8_t u8g_com_HAL_DUE_sw_spi_fn(u8g_t*, uint8_t, uint8_t, void*)':

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi.cpp:85:42: error: 'u8g_SetPIOutput_DUE' was not declared in this scope

       u8g_SetPIOutput_DUE(u8g, U8G_PI_SCK);

                                          ^

sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_sw_spi.cpp:90:44: error: 'u8g_SetPILevel_DUE' was not declared in this scope

       u8g_SetPILevel_DUE(u8g, U8G_PI_SCK, 0);

                                            ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

```

DerThommi, 2019-07-21:
In u8g_com_HAL_DUE_sw_spi.cpp located in Marlin\src\HAL\HAL_DUE moving line 65 `#include "u8g_com_HAL_DUE_sw_spi_shared.h"
`

to line 71 (after the last include) makes it compile again. Unfortunately I broke my OLED display during the troubleshooting process so I can't test if it actually works. Arduino IDE gives no error, vs code a couple of warnings.

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle