MarlinFirmware/Marlin #14569: [BUG] Cant compile 2.0 with #define AZSMZ_12864

wailord284, 2019-07-10:
On the latest version of Marlin 2.0, defining the screen AZSMZ_12864  prevents me from compiling with either Arduino IDE or PlatformIO in visual studio code. By disabling AZSMZ_12864, I can compile normally. I am using a Smart Ramps with an Arduino DUE on a delta printer.
Here are my config files. I have just updated them to be on the newest config file layout. Thank you.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3379095/configs.zip)

DerThommi, 2019-07-10:
Seems related to my Issue #14567

wailord284, 2019-09-13:
> If this has been resolved, please share what you did and close the issue. Thank you.

I haven't been able to test recently, but as of not long ago it was still not fixed.
However, someone with a similar issue proposed a solution that I have not yet tried.
https://github.com/MarlinFirmware/Marlin/issues/14567#issuecomment-513578304

boelle, 2019-10-12:
@wailord284 still having issues?

closed 2019-10-12 by wailord284