MarlinFirmware/Marlin #14570: [BUG] fail build in platformio (4.0?): lib_ldf_mode error in some environment

sl1pkn07, 2019-07-10:
### Description

platformio output error when try to build firmware for the SM32F1 board like `fysetc_STM32F1`, `JGAURORA_A5S_A1`,  about bad `lib_ldf_mode` option

### Steps to Reproduce

- setup vscode and install platformio ide
- build `env:fysetc_SM32F1` or `env:JGAURORA_A5S_A1`
- see the terminal

**Expected behavior:** build OK

**Actual behavior:**  fail build

#### Additional Information

VScode and platformio-core 4.0.0

> Executing task: platformio run --target clean --environment fysetc_STM32F1 <

~~~
Error: Invalid value: invalid choice: 1. (choose from off, chain, deep, chain+, deep+) for option `lib_ldf_mode` in section [env:fysetc_STM32F1]
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
~~~

Ccleorina87, 2019-07-11:
inside PlatformIO.ini change ( for 4.0 PIO )

env_default to default_envs

thinkyhead, 2019-07-12:
> env_default to default_envs

We'll keep this unchanged until after PIO 4.0 has been out for several months. "Deprecated" only means it will go away someday.

closed 2019-07-12 by thinkyhead

Ccleorina87, 2019-07-12:
Agree