MarlinFirmware/Marlin #14575: [BUG] The printer will reset when performing UBL of 3 points.

satiromarra, 2019-07-11:
When try gcode `G29 J`
```
>>> G29 J
SENDING:G29 J
Tilting mesh (1/3)
Error:MAXTEMP triggered, system stopped! Heater_ID: bed
[ERROR] Error:MAXTEMP triggered, system stopped! Heater_ID: bed

Error:KILL caused by KILL button/pin
[ERROR] Error:KILL caused by KILL button/pin

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!
```
The both temperatures (hotend and bed) are ok, 210º and 60º respectively.
And I have not pressed any reset button.

thinkyhead, 2019-07-11:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2019-07-11:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply. We'll use them in our testing.

satiromarra, 2019-07-11:
I use already the branch bugfix-2.0.x and I have it updated today.
ramps 1.4 + NO SD card + UBL
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3382460/Configuration.zip)

satiromarra, 2019-07-11:
@thinkyhead 
I have tried different "trees" and the last one that works is this: https://github.com/MarlinFirmware/Marlin/tree/f7de453da89091d162f01ddddeb259bf8a682472

Works
https://github.com/MarlinFirmware/Marlin/tree/ab99a807938b7fd8c5eafe2018b23ff876e765e6
https://github.com/MarlinFirmware/Marlin/tree/3ae3bf5d33cc3c6a3d250cc19a3de0b9e22fadae
https://github.com/MarlinFirmware/Marlin/tree/5519054f50c2e0e532f8a541f156ef350aaecc9c
https://github.com/MarlinFirmware/Marlin/tree/9817976375509b9d2928f7c1cf615b3949f7e026
https://github.com/MarlinFirmware/Marlin/tree/881e5a48bcdd2f78528b4966ce3bd286b8efe653
https://github.com/MarlinFirmware/Marlin/tree/828cc71847190f61e2d9991147091c010d121745
https://github.com/MarlinFirmware/Marlin/tree/f7de453da89091d162f01ddddeb259bf8a682472

Don't works
https://github.com/MarlinFirmware/Marlin/tree/cf9ac4c84751507d09ccb9379bf366cee897229f
https://github.com/MarlinFirmware/Marlin/tree/ad1c061e7b4522f30e8bc4deac565baa0c4b6568

jokejoke, 2019-07-12:
I have the same problem with the last branch bugfix-2.0.x . (The heatbed was preheated to 70ºC).

`>>> G29 P1 T`
`SENDING:G29 P1 T`
`Default storage slot 0 selected.`
`Mesh invalidated. Probing mesh.`
`Bed Topography Report:`
`    ( 15,205)                                      (205,205)`
`        0       1       2       3       4       5       6`
` 6 |   .       .       .       .       .       .       .`
`   |`
` 5 |   .       .       .       .       .       .       .`
`   |`
` 4 |   .       .       .       .       .       .       .`
`   |`
` 3 |   .       .       .    [  .   ]   .       .       .`
`   |`
` 2 |   .       .       .       .       .       .       .`
`   |`
` 1 |   .       .       .       .       .       .       .`
`   |`
` 0 |   .       .       .       .       .       .       .`
`        0       1       2       3       4       5       6`
`    ( 15, 15)                                      (205, 15)`
`Probing mesh point 1/49.`
`Error:MAXTEMP triggered, system stopped! Heater_ID: bed`
`[ERROR] Error:MAXTEMP triggered, system stopped! Heater_ID: bed`

satiromarra, 2019-07-12:
@jokejoke try this tree: https://github.com/MarlinFirmware/Marlin/tree/f7de453da89091d162f01ddddeb259bf8a682472

jokejoke, 2019-07-12:
@satiromarra I tried it. I have no problem with this tree (UBL works fine).

satiromarra, 2019-07-14:
Why this change make Marlin fail? [cf9ac4c](https://github.com/MarlinFirmware/Marlin/commit/cf9ac4c84751507d09ccb9379bf366cee897229f)

satiromarra, 2019-07-15:
I have tried the latest version and (apparently) it works correctly.
@thinkyhead @jokejoke

lterrill, 2019-07-15:
I am seeing this issue as well on a GT2560v3 board. So I tried the latest marlin 2 version (93bc1b17f) as suggested by @satiromarra . However in this version, the problem appears and disappears depending on my particular configuration options so I don't think it is completely fixed. With my 'normal' configuration, it crashes. When I turn on DEBUG_LEVELING_FEATURE to see exactly where it is failing, the issue goes away.  I then turn off Z_MIN_PROBE_REPEATABILITY_TEST to save a bit of memory since level debug puts me at 82% program storage, the crash comes back. Here is my console with level debug on (after issuing G28, G29 L1, and G29 J with the crash. I have started the log with G29 L1...

Mesh loaded from slot 1
Done.
ok
  current_position=(154.00, 111.00, 3.00) : sync_plan_position
Tilting mesh (1/3)
xi out of bounds in z_correction_for_y_on_vertical_mesh_line(ry0=0.00, xi=188, y1_i=0)
echo: cold extrusion prevented
xi out of bounds in z_correction_for_y_on_vertical_mesh_line(ry0=0.00, xi=187, y1_i=0)
echo: cold extrusion prevented
xi out of bounds in z_correction_for_y_on_vertical_mesh_line(ry0=0.00, xi=186, y1_i=0)
echo: cold extrusion prevented
...repeats 16 times total as xi counts down by 1 each time....
Error:MAXTEMP triggered, system stopped! Heater_ID: bed
Error:Printer halted. kill() called!

Just let me know if you need something from me. I'm happy to help. I've included my last config with debug, no Z_MIN_PROBE_REPEATABILITY_TESTThanks!

satiromarra, 2019-07-15:
In my case, the error is repeated every time I performed `G29 J`

@lterrill and this? This is the branch with which I have had no problems: [f7de453](https://github.com/MarlinFirmware/Marlin/tree/f7de453da89091d162f01ddddeb259bf8a682472)

I've been with this tree for 6 hours and I have no problems at the moment: [93bc1b1](https://github.com/MarlinFirmware/Marlin/tree/93bc1b17f9c633643aeac6772929ae8d497a6113)

I have made changes to my current configuration. `RAMPS 1.4 + SD + UBL`
[Configuration-20190715.zip](https://github.com/MarlinFirmware/Marlin/files/3393992/Configuration-20190715.zip)

Roxy-3D, 2019-07-15:
G29 J should have very low memory requirements compared to other G29 sub commands (such as mesh generation and mesh editing).   It is strange that this triggers an instability.    I'm wondering if you are running low on RAM (variable) memory.   

How much memory does Arduino say you have free?   Can you turn on M100 and let us know what the numbers are that it reports?

lterrill, 2019-07-15:
From M100:
bss_end               : 0x18DF
free_memory_start     : 0x18DF
free_memory_end       : 0x216C
Stack Pointer         : 0x216C
Initializing free memory block.

1939 bytes of memory initialized.

lterrill, 2019-07-15:
Responding to @satiromarra, the identical configuration that crashes on 93bc1b1 does seem to work fine on f7de453. However, by simply changing a seemly unrelated option (I enable or disable M48 tests on mine), it appears to change the 'G29 J' crash behavior on later revisions. Very strange. If you have time, you might try turning your SD card off again and see your config crashes again under the current revision. Your recent success may have been due to the config change and not new code.

Roxy-3D, 2019-07-16:
@lterrill   That should be plenty of free memory to not have stack problems.   Something else is causing weird behavior!

jokejoke, 2019-07-16:
The problem still presists in tree:
https://github.com/MarlinFirmware/Marlin/tree/93bc1b17f9c633643aeac6772929ae8d497a6113

`>>> M111 S32`
`SENDING:M111 S32`
`echo:DEBUG:LEVELING`
`>>> G29 P1`
`SENDING:G29 P1`
`Default storage slot 0 selected.`
`Mesh invalidated. Probing mesh.`
`  current_position=(83.00, 110.00, 4.00) : set_probe_deployed`
`deploy: 1`
`do_probe_raise(4.00)`
`>>> do_blocking_move_to(83.00, 110.00, 4.00)`
`<<< do_blocking_move_to`
`Probing mesh point 1/100.`
`start` **<--The printer has restarted itself**
`echo:Marlin bugfix-2.0.x`

My configuration:
[Configuration-20190715.zip](https://github.com/MarlinFirmware/Marlin/files/3395444/Configuration-20190715.zip)

satiromarra, 2019-07-16:
@lterrill 
Using the tree: [93bc1b1](https://github.com/MarlinFirmware/Marlin/tree/93bc1b17f9c633643aeac6772929ae8d497a6113) `#define SDSUPPORT`

Test M100
```
>>> M100
SENDING:M100
bss_end               : 0x13F7
free_memory_start     : 0x13F7
free_memory_end       : 0x2187
Stack Pointer         : 0x2187
Initializing free memory block.
3222 bytes of memory initialized.
```
Test M48
```
>>> M48
SENDING:M48
M48 Z-Probe Repeatability Test
Finished!
Mean: -0.001500 Min: -0.003 Max: 0.000 Range: 0.003
Standard Deviation: 0.001225
```
Test UBL (**Probing mesh point 102/100. ?**)
```
>>> G28
SENDING:G28
>>> G29 A
SENDING:G29 A
Unified Bed Leveling System v1.01 active
>>> G29 L0
SENDING:G29 L0
Done.
>>> G29 J
SENDING:G29 J
Tilting mesh (1/3)
Probing mesh point 102/100. 
Error:Thermal Runaway, system stopped! Heater_ID: bed
[ERROR] Error:Thermal Runaway, system stopped! Heater_ID: bed

Error:MAXTEMP triggered, system stopped! Heater_ID: bed
[ERROR] Error:MAXTEMP triggered, system stopped! Heater_ID: bed

Error:Thermal Runaway, system stopped! Heater_ID: bed
[ERROR] Error:Thermal Runaway, system stopped! Heater_ID: bed
```
Test `G29 P1 T` the command works fine, but by mistake run `G29 P` and it has been deleted. :`(

Using the same tree: [93bc1b1](https://github.com/MarlinFirmware/Marlin/tree/93bc1b17f9c633643aeac6772929ae8d497a6113) but `//#define SDSUPPORT`
`DATA:    [======    ]  56.1% (used 4599 bytes from 8192 bytes)`
`PROGRAM: [=======   ]  73.3% (used 186078 bytes from 253952 bytes)`

Test M100
```
>>> M100
SENDING:M100
bss_end               : 0x13F7
free_memory_start     : 0x13F7
free_memory_end       : 0x2187
Stack Pointer         : 0x2187
Initializing free memory block.
3222 bytes of memory initialized.
```
Test M48 without homing (reset?)
```
>>> M48
SENDING:M48
echo:Home XYZ primero
Bed Topography Report:
    ( 15,205)Error:Thermal Runaway, system stopped! Heater_ID: bed
```
Test M48 with homing
```
>>> G28
SENDING:G28
>>> M48
SENDING:M48
M48 Z-Probe Repeatability Test
Finished!
Mean: 0.002750 Min: 0.000 Max: 0.005 Range: 0.005
Standard Deviation: 0.001750
```
Test UBL (**Probing mesh point 10281/100.**?)
```
>>> G28
SENDING:G28
>>> G29 A
SENDING:G29 A
Unified Bed Leveling System v1.01 active
>>> G29 L0
SENDING:G29 L0
Done.
>>> G29 J
SENDING:G29 J
Tilting mesh (1/3)
Bed Topography Report:
    ( 15,205)Error:Thermal Runaway, system stopped! Heater_ID: bed
...
Probing mesh point 10281/100.
start
```

Test `G29 P1 T` the command works fine.

I will perform some more test, but i don't know why now fails `G29 J`

satiromarra, 2019-07-16:
@lterrill 
Using the tree: [93bc1b1](https://github.com/MarlinFirmware/Marlin/tree/93bc1b17f9c633643aeac6772929ae8d497a6113) with 
```
//#define SDSUPPORT
#define M100_FREE_MEMORY_WATCHER
#define Z_MIN_PROBE_REPEATABILITY_TEST
```

`DATA:    [======    ]  56.1% (used 4599 bytes from 8192 bytes)`
`PROGRAM: [=======   ]  73.3% (used 186078 bytes from 253952 bytes)`

Test M100
```
>>> M100
SENDING:M100
bss_end               : 0x13F7
free_memory_start     : 0x13F7
free_memory_end       : 0x2187
Stack Pointer         : 0x2187
Initializing free memory block.
3222 bytes of memory initialized.
```
Test M48
```
>>> M48
SENDING:M48
M48 Z-Probe Repeatability Test
Finished!
Mean: -0.001500 Min: -0.003 Max: 0.000 Range: 0.003
Standard Deviation: 0.001225
```
Test UBL (**Probing mesh point 102/100. ?**)
```
>>> G28
SENDING:G28
>>> G29 A
SENDING:G29 A
Unified Bed Leveling System v1.01 active
>>> G29 L0
SENDING:G29 L0
Done.
>>> G29 J
SENDING:G29 J
Tilting mesh (1/3)
Probing mesh point 32809/100.
Error:Thermal Runaway, system stopped! Heater_ID: bed
[ERROR] Error:Thermal Runaway, system stopped! Heater_ID: bed

Error:MAXTEMP triggered, system stopped! Heater_ID: bed
[ERROR] Error:MAXTEMP triggered, system stopped! Heater_ID: bed

Error:Thermal Runaway, system stopped! Heater_ID: bed
[ERROR] Error:Thermal Runaway, system stopped! Heater_ID: bed
```
Test `G29 P1 T` the command works fine, but by mistake run `G29 P` and it has been deleted. :`(

Using the same tree: [93bc1b1](https://github.com/MarlinFirmware/Marlin/tree/93bc1b17f9c633643aeac6772929ae8d497a6113) with
```
#define SDSUPPORT
#define M100_FREE_MEMORY_WATCHER
#define Z_MIN_PROBE_REPEATABILITY_TEST
```
`DATA:    [========  ]  77.4% (used 6341 bytes from 8192 bytes)`
`PROGRAM: [========  ]  80.9% (used 205326 bytes from 253952 bytes)`

Test M100
```
>>> M100
SENDING:M100
bss_end               : 0x1AC5
free_memory_start     : 0x1AC5
free_memory_end       : 0x216D
Stack Pointer         : 0x216D
Initializing free memory block.
1454 bytes of memory initialized.
```
Test M48 without homing (reset?)
```
>>> M48
SENDING:M48
echo:Home XYZ primero
?(F)ade height for Bed Level Correction not plausible.
start
```
Test M48 with homing
```
>>> G28
SENDING:G28
>>> M48
SENDING:M48
M48 Z-Probe Repeatability Test
Finished!
Mean: -0.003750 Min: -0.008 Max: 0.000 Range: 0.008
Standard Deviation: 0.002016
```
Test UBL 
```
>>> G28
>>> G28
SENDING:G28
>>> G29 A
SENDING:G29 A
Unified Bed Leveling System v1.01 active
>>> G29 L0
SENDING:G29 L0
Done.
>>> G29 J
SENDING:G29 J
Tilting mesh (1/3)
Tilting mesh (2/3)
Tilting mesh (3/3)
```

Test `G29 P1 T` the command works fine.

I will perform some more test.

satiromarra, 2019-07-16:
`G29 J` without homing restart the printer.
```
>>> G29 J
SENDING:G29 J
echo:Home XYZ primero
?(F)ade height for Bed Level Correction not plausible.
start 
```

satiromarra, 2019-07-21:
@jokejoke you have enabled `PIDTEMPBED`?

jokejoke, 2019-07-21:
@satiromarra yes - for a long time.

satiromarra, 2019-07-21:
I activated the `PIDTEMPBED` (for test) parameter and Marlin is frozen when realize `G29 J`.
```
G28
G29 A
G29 L0
G29 J 
```
I have to leave the configuration.h like this:
```
//#define PIDTEMPBED
#define BED_LIMIT_SWITCHING
```

jokejoke, 2019-07-21:
I activated  `BED_LIMIT_SWITCHING` instead `PIDTEMPBED` and the result is the same -> restart after 
`G29 P1` :(

satiromarra, 2019-07-21:
With the last commit [e5aa453](https://github.com/MarlinFirmware/Marlin/commit/e5aa45329305be8ac5e277305d52a449a41bfc98)?
I have seen your configuration and it seems to be fine.

`G29 P1` after `G28`?
```
G28
G29 A
G29 P1 T
```
In [some tests](#issuecomment-511858459) `G29 ...` fails if you do not `G28` first.

jokejoke, 2019-07-21:
Yes with the last commit https://github.com/MarlinFirmware/Marlin/commit/e5aa45329305be8ac5e277305d52a449a41bfc98.

1) `M502`
2) `M500`
3) `G28`
4) I use also `G34` for Z align, but doesn't depend on it.
5) `G29 P1 T` -> cause restart

tpruvot, 2019-07-21:
else you can do "G28 XY" before G29 to avoid an useless Z probe

Chalmes, 2019-07-21:
Hi all.

I'm also experiencing this issue on a RAMPS 1.4/Atmega2560 configuration, and i've been looking into it a bit more and I notice that if i compile and upload with UBL enabled in the configuration.h, then I don't get a response back when I send through an M502.  The code doesn't appear to get through to the line that pumps out the usual message:
DEBUG_ECHOLNPGM("Hardcoded Default Settings Loaded");

If all i do is disable UBL and reupload, then that message comes back to Pronterface with no issues.

For those experiencing the printer restarts on G29 J, can you please check if the M502 command is returning the expected message.

jokejoke, 2019-07-22:
@Chalmes I noticed that the M502 does not return expected message and also `G28` -> `G29 P1` cause restart (with commit https://github.com/MarlinFirmware/Marlin/commit/e5aa45329305be8ac5e277305d52a449a41bfc98).

satiromarra, 2019-07-22:
@Chalmes 
```
>>> M502
SENDING:M502
```
Does not return nothing but it works.

lterrill, 2019-07-22:
Following up on @satiromarra examination of the commit after which this issue occurred, I found after updating to current HEAD (a589456a1) and manually removing this code block:

In Marlin/src/HAL/shared/Marduino.h
#ifndef vsnprintf_P
	  #define vsnprintf_P vsnprintf
#endif 

that was added in commit cf9ac4c84751507d09ccb9379bf366cee897229f
my crashing behavior issue went away.

I realize this is not a proper fix as it will likely break the 32-bit build due to other changes made in the same commit, but I am running on a atmega2650 and this appears to correct this issue for me. 

If someone with stronger Marlin skills could take a look at commit cf9ac4c84751507d09ccb9379bf366cee897229f, specifically where vsnprintf_P is redefined in Marduino.h and fix this properly, I'd be grateful. It seems to be the cause of the instability on the atmega2650 boards.

FYI: I am using Arduino IDE 1.8.9 in case that may be a factor

satiromarra, 2019-07-22:
I have also done that test, modifying the files one by one and see if I could get an error; however when enabling `SDSUPPORT` the error seems to" disappear "
On the other hand, when performing a `G29 J` or` G29 P1 T` without doing `G28` before, the printer resets.
In case you may be interested, the G29 commands allow verbosing, Vx 0 to 4, ex: `G29 P1 T V4` `G29 J V4`
I use Platformio via "Sublime text" to build and upload and atmega2650
Now i use the last commit and works fine, but using the same configuration.

satiromarra, 2019-07-22:
@Chalmes `DEBUG_ECHOLNPGM` (and others debug functions for leveling) works if `#define DEBUG_LEVELING_FEATURE` is enabled in `Configuration.h`

satiromarra, 2019-07-22:
@jokejoke 

File `src/feature/bedlevel/ubl/ubl_G29.cpp`
line 310:
`    if (g29_parameter_parsing()) return; // Abort on parameter error`

lines 1077-1080 in function g29_parameter_parsing: 
```
    g29_x_flag = parser.seenval('X');
    g29_x_pos = g29_x_flag ? parser.value_float() : current_position[X_AXIS];
    g29_y_flag = parser.seenval('Y');
    g29_y_pos = g29_y_flag ? parser.value_float() : current_position[Y_AXIS];
```
whats happend if `current_position` has not values? I think that in this case a `G28` should be done to complete the values.

And these changes? 
File `src/feature/bedlevel/ubl/ubl_G29.cpp` lines 1074-1083 
```
    if (axis_unhomed_error()) gcode.home_all_axes(); //Do homing now?
    //if (axis_unhomed_error()) return UBL_ERR; //Or check if homing before and return?

    g29_constant = 0;
    g29_repetition_cnt = 0;

    g29_x_flag = parser.seenval('X');
    g29_x_pos = g29_x_flag ? parser.value_float() : current_position[X_AXIS];
    g29_y_flag = parser.seenval('Y');
    g29_y_pos = g29_y_flag ? parser.value_float() : current_position[Y_AXIS];

``` 
I don't tried this changes yet.

Chalmes, 2019-07-22:
I'm not sure the magic that went into a589456, but that's fixed both the eeprom response messages when UBL is enabled, as well as the UBL G29 J crashing.

jokejoke, 2019-07-23:
@lterrill  @satiromarra 
I can confirm that if I comment this piece of code (in the last commit) in Marlin/src/HAL/shared/Marduino.h:

`#ifndef vsnprintf_P`
`#define vsnprintf_P vsnprintf`
`#endif`

And then:
`G28`
`G29 P1 T`

UBL works without restart:)

satiromarra, 2019-07-23:
@jokejoke with `SDSUPPORT` disabled?

jokejoke, 2019-07-23:
@satiromarra with `SDSUPPORT` enabled.

satiromarra, 2019-07-23:
When I have `SDSUPPORT` disabled it fails me. I have to enable it so that it doesn't happen, I really don't need the SD. I lose memory just by having it enabled.

satiromarra, 2019-07-23:
@jokejoke I tried comment in src/lcd/ultralcd.cpp:1373
`//vsnprintf_P(status_message, MAX_MESSAGE_LENGTH, fmt, args);`
And all the tests worked without problems.

I have also tried changing the line `vsnprintf_P(status_message, MAX_MESSAGE_LENGTH, fmt, args);` to `vsnprintf(status_message, MAX_MESSAGE_LENGTH, fmt, args);`

There may be a problem with vsnprintf

jokejoke, 2019-07-23:
@satiromarra Now we know where the problem is :) Who can fix it? :)

lterrill, 2019-07-24:
I believe https://github.com/MarlinFirmware/Marlin/pull/14712 is related to this issue and is a better explanation of the root cause. A fix has just been committed, so I suggest that anyone seeing this issue pull 585a8a96b4dea316f97a22f24baa166f138fd20f or later and test again.

jokejoke, 2019-07-24:
I have no problem with `G29 P1` with the last commit [1088846](https://github.com/MarlinFirmware/Marlin/commit/1088846caef6f660d5af65e1725a6a6deeef16fd). For me is the problem solved. Thank you :)

jokejoke, 2019-07-24:
@lterrill 
I tried some LCD menu items (with the last commit [1088846)](https://github.com/MarlinFirmware/Marlin/commit/1088846caef6f660d5af65e1725a6a6deeef16fd) and now if I want to use "Filament change" on LCD i get:

`echo:enqueueing "M600 B0 T0"`
`echo:Home XYZ first`
`Error:Heating failed, system stopped! Heater_ID: 0`
`[ERROR] Error:Heating failed, system stopped! Heater_ID: 0`

Load and Unload filament work without problem. UBL works without problems too.

satiromarra, 2019-07-25:
I tried the last commit [ead6bb2](https://github.com/MarlinFirmware/Marlin/commit/ead6bb267dd17b5053426e4129467373e64c164a) and the problem persists if I don't do it before `G28`.
```
>>> G29 J V4
SENDING:G29 J V4
echo:Home XYZ primero
?Error probing point. Aborting operation.
start
echo:Marlin bugfix-2.0.x
```

Roxy-3D, 2019-07-26:
@satiromarra    I think you might have the root cause:  https://github.com/MarlinFirmware/Marlin/issues/14575#issuecomment-513972562

You said:

> whats happend if current_position has not values? I think that in this case a G28 should be done to complete the values.

You might be correct.    But if this is the case, why did the problem just show up now?   This code has been this way for a long time?   Something else is factoring into why this problem is just showing up now.

satiromarra, 2019-07-26:
@Roxy-3D 
I think that error leads to another error which causes the printer to reset.
The change of comment [#14575 (comment)](#issuecomment-513972562) forces a `G28` if it is not done, but has not resolved other printer resets.

Another test I did was in: `src/lcd/ultralcd.cpp:1373`
`//vsnprintf_P(status_message, MAX_MESSAGE_LENGTH, fmt, args);`
And all the tests I did showed no problem (and neither did any message).
With this change it is no longer necessary to modify the `src/feature/bedlevel/ubl/ubl_G29.cpp` file and is able to perform `G28` on its own.

I think the problem comes from `vsnprintf_P` or `vsnprintf`, however in the documentation of the `vsnprintf` function it does not seem that this throws an error, it only returns -1.

I am not sure but it may not be due directly to `vsnprintf`, but to the continuity of the function, (`finish_status, ExtUI::onStatusChanged, refresh,` ...)

I have tried both changes in the printer without problems.

lterrill, 2019-07-26:
Just an FYI, the previous fix #14712 I referenced as fixing my inability to call G29 after a G28 was found to not completely address the entirety of original issue. A second fix #14725 is currently 'in review' and has not been committed at this time. I manually applied this second fix and, for me, it addresses the crash associated with the 'Home XYZ first' message issued by vsnprintf_P when G29 or M48 is issued without first doing a G28. So look for a fix soon for #14725 to address this issue.

jscaltreto, 2019-07-27:
I've been having a similar issue with the printer resetting on G29 P1. #14725 appears to solve the issue for me.

satiromarra, 2019-07-28:
I tried today the last commit [fec52e6](https://github.com/MarlinFirmware/Marlin/commit/fec52e61ea22640d70d7aaa0c8c235958cfe4b24) and appears works fine.

If everything is correct, I close the issue.

closed 2019-07-28 by satiromarra