MarlinFirmware/Marlin #14579: SKR 1.3 TMC 2130 SPI mode EEPROM_ENABLE causes TMC CONNECTION ERROR

ChrisS888, 2019-07-11:
Config set for TMC 2130 SPI, BLTouch, define EEPROM_SETTINGS. Printer boots fine but reports TMC_CONNECTION  ERROR on LCD and will halt if any attempt made to move X, Y, Z or E. M122 reports fault status also.

Comment out EEPROM_SETTINGS and all seems OK. Test print fine, M122 all good.

This was Marlin 2.0.x dowloaded on 1st July.

Tried again today with latest version, fault now different. Will report elsewhere.

Thank you.

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

ChrisS888, 2019-07-21:
Apologies - this was my first bug report, and as is no doubt typical, I didn't read the instructions :(

This issue has been resolved in as much as the version from 13th July is working fine for me now, but for completeness, I have attached the configs. I left this open as I saw other posts elsewhere about this. Please feel free to close it if you feel that's best.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3414424/Archive.zip)

boelle, 2019-07-21:
people are still able to find this even if closed, and if its solved then the best thing is to close it

we can always open it again if that is needed

closed 2019-07-21 by boelle

ChrisS888, 2019-07-21:
Thank you - again, apologies for totally messing up the protocol.

milkpirate, 2019-10-04:
Anything new on this one? I seem to have the same problem.

@ChrisS888 did you find a solution to have both (TMCs & EEPROM) working?

ChrisS888, 2019-10-05:
@milkpirate   I just persisted with later and later releases until it came good. That said, not totally convinced it wasn't some other factor - perhaps some corrupt data in the SKR EEPROM that eventually got cleared??

Not a satisfactory diagnosis ultimately, but it is now working great and has been for a while.