MarlinFirmware/Marlin #14581: [BUG] PlatformIO exception with Marlin defined board

GMagician, 2019-07-11:
BigTree_SKR_Pro.json defines "cmsis-dap" as upload protocol, this lead to exception error in PIO when called with BOARDS argument. Really sure it's correct?

thisiskeithb, 2019-07-11:
@Msq001 might know.

Msq001, 2019-07-12:
hello.  I have downloaded the latest version of Marlin 2.0，and it can be compiled successfully. What kind of exception error did you encounter? Can you provide the log information of exception error?

GMagician, 2019-07-12:
I have to check, error is when you execute PIO command like PIO boards or PIO init --ide .....
it seems that removing "cmsis-dap" solve it. Maybe something related with PIO 4.0 strict parser control?

GMagician, 2019-07-12:
here is log

> $ pio boards
   Error: Traceback (most recent call last):
  File "c:\python27\lib\site-packages\platformio\__main__.py", line 88, in main
    cli(None, None, None)
  File "c:\python27\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\python27\lib\site-packages\platformio\commands\__init__.py", line 37, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python27\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python27\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\python27\lib\site-packages\platformio\commands\boards.py", line 33, in cli
    for board in _get_boards(installed):
  File "c:\python27\lib\site-packages\platformio\commands\boards.py", line 75, in _get_boards
    return pm.get_installed_boards() if installed else pm.get_all_boards()
  File "c:\python27\lib\site-packages\platformio\managers\platform.py", line 191, in get_all_boards
    boards = self.get_installed_boards()
  File "c:\python27\lib\site-packages\platformio\util.py", line 70, in wrapper
    self.cache[key] = (time.time(), func(*args, **kwargs))
  File "c:\python27\lib\site-packages\platformio\managers\platform.py", line 180, in get_installed_boards
    for config in p.get_boards().values():
  File "C:\Users\Giuliano\.platformio\platforms\atmelsam\platform.py", line 68, in get_boards
    result[key] = self._add_default_debug_tools(result[key])
  File "C:\Users\Giuliano\.platformio\platforms\atmelsam\platform.py", line 111, in _add_default_debug_tools
    assert openocd_chipname
AssertionError

Msq001, 2019-07-13:
hello， thanks for your test and feedback!
I have confirmed this issue, but I don't know why.  I've created a PR to delete cmsis-dap

closed 2019-07-13 by GMagician