MarlinFirmware/Marlin #14582: X axis motor is skipping steps...

SRIRAM134, 2019-07-11:
### X axis motor is skipping steps...

1. When I move the X axis motor sometimes it skips steps.
2. While in the print also the X axis motor skips steps.

**Expected behavior:** [What you expect to happen]: Expectation is that the motor must work properly without skipping steps.

**Actual behavior:** [What actually happens]: But the X axis motor is skipping steps during printing....

#### Additional Information

* My settings for X axis are as follows:
Default steps per mm: X1031.43
Default feedrate: X8
Default Max. Acceleration: X300
Default  acceleration: P150
                                       R17
                                       T300
Default X jerk: 17

* 
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-07-11:
Your numbers seem off.
Typical axis_steps_per_unit (unit is in mm by default) for X and Y is 80~200 or thereabout. Use this to calculate your steps. https://prusaprinters.org/calculator/
You may need to refine it: https://e3d-online.dozuki.com/Guide/Titan+Marlin+Configuration/35   --see steps 8-10, you can adapt the same method to any other axis

Acceleration of 500 or more - I find that 1000-3000 is fine on most machines, Marlin's default config is set at 3000.

All that said: this Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

SRIRAM134, 2019-07-12:
Okay thanks for your help...

closed 2019-07-12 by SRIRAM134