MarlinFirmware/Marlin #14590: Remove Configuration.h and Configuration_adv.h in project root from git

marchelh, 2019-07-12:
All,

Please remove the Configuration.h and Configuration_adv.h from the default location, so that pulling the latest version does not conflict your local configuration setting. Also add these to the .gitignore file, so that user/developers are not adding them by accident.

Kind regards,
Marcel

reloxx13, 2019-07-12:
this would make it harder to merge if new defines get added.

i use marlin 2.0.x for a year now and had never merging issues.

yes i get conflicts, but this is correct for being uptodate with the configs.

e.g. bltouch configs got moved to config_adv. with ignoring them u need to check the hole config urself on every pull.

sl1pkn07, 2019-07-12:
or directly in the main if you don't want push changes to upstream:

- clone the repo
~~~
git clone https://github.com/MarlinFirmware/Marlin.git
cd Marlin
git checkout <your preferable branch, you can use TAB for the list>
~~~

- change you code and commit it with:
~~~
commit -am 'you changes'
~~~
you not need add more commits with you changes, you can add them in the created commit, see below:

- if you need add more changes, edit the files you want and apply it in the repo with:
~~~
git add *
git commit --amend --no-edit
~~~
the new changes is applied automagically in the commit created by you with 'git commit' in the earlier step

- when exist new update in github:
~~~
git fetch origin
git rebase origin/<the branch you use>
~~~

- if have more updates, repeat steps above

- if the git rebase make a file conflicts, resolve it with the editor text. add the changes and continue with:
~~~
git add *
git rebase --continue
~~~

marchelh, 2019-07-13:
Of cause, you can do some stuff via git commands, and when new options are added, you have to add them in the config. But there you have the config/defaults files for here you place the new option every time there is an update. I really don't like the fact that changeable "config" file are part of the git repository.

reloxx13, 2019-07-13:
read this, it got declined already

https://github.com/MarlinFirmware/Marlin/pull/12541

closed 2019-07-13 by marchelh

reloxx13, 2019-07-13:
if u did not use git till now, use it and tell us ur experience back here after 1-2 month active using/updating marlin 2.0.x.

im updating at least once a week.