MarlinFirmware/Marlin #14594: Z default estep is 4000 in config file.

SFRSKZ, 2019-07-12:
i guess noone noticed or didnt tell anything about it yet. xy is set to 80 but z is 4000. so i guess one 0 need to be removed. i didnt notice that (e step is added to starting script)  and damaged my powder coated pei :( 

this is my first github post so if i did something wrong , forgive me.

Ludy87, 2019-07-12:
Where?

yet-another-average-joe, 2019-07-12:
This is true ! Never noticed before...

#define DEFAULT_AXIS_STEPS_PER_UNIT  { 80, 80, 4000, 500 } 

4000 is a lot

with direct drive 2mm pitch 4 start leadscrews (=8mm/rev), we need 400 microsteps (1/8 rev, 16 µsteps, 200 steps/rev)

4000 is suitable for M5 threaded rods (single start, 0.8mm pitch, 0.8mm/rev)

phizz166, 2019-07-13:
Always double-check your configuration! Removing the hotend and any delicate glass/PEI bed surfaces is a good idea when first testing a new firmware...

4000 steps is an artifact from the old days when most printers used threaded rods and nuts instead of actual leadscrews. With essentially all new machines using proper mechanisms, this probably should be updated to reflect a more realistic value, yes.

zillarob, 2019-07-13:
I dont know if many use it, but default filament width is 3.0, when my guess is 1.75 is waayyy more popular nowadays.

sjasonsmith, 2020-12-02:
The default values really are not very important. Most users are going to start with example for their base printer anyway.
No matter what these values change to, they are always going to be wrong for some or most users.

If you feel strongly about any of these, I suggest you submit the change as a pull request, and the merits of the change can be discussed there.

closed 2020-12-02 by sjasonsmith