MarlinFirmware/Marlin #14596: [BUG] UBL wierd behaviour on edge  Top and right

CrNMGuy, 2019-07-12:
Hi !

I noticed a strange behaviour on UBL active correction 
Context : CoreXY with BLTouch clone, Measure with indicator (accuracy on the mount 100%)
See behaviour here [https://youtu.be/0KD9YAJmi8o](https://youtu.be/0KD9YAJmi8o)
(1mm X incrementation)

Works great on all the bed surface, except last 1 mm on Y and last 2 mm on X

Here is measurement  with an indicator (plus see video) :

```
        with compensation UBL  G29 A
                without  G29 D
X@261    0.09    0.51
X@262    0.09    0.51
X@263    0.09    0.52
X@264    0.09    0.52
X@265    0.10    0.52
X@266    0.53    0.53
X@267    0.53    0.53

```

UBL Initiated by (http://marlinfw.org/docs/gcode/G029-ubl.html) :
```
G28           ; Home XYZ.
G29 P1        ; Do automated probing of the bed.
G29 P3 T      ; Repeat until all mesh points are filled in.
G29 S1        ; Save UBL mesh points to EEPROM.
G29 A         ; Activate the UBL System.
M500          ; Save current setup. WARNING - UBL will be active at power up, before any G28.
```

Configuration extract : 
```
#define X_BED_SIZE 267
#define Y_BED_SIZE 320

#define AUTO_BED_LEVELING_UBL

  #define ENABLE_LEVELING_FADE_HEIGHT

  // For Cartesian machines, instead of dividing moves on mesh boundaries,
  // split up moves into short segments like a Delta. This follows the
  // contours of the bed more closely than edge-to-edge straight moves.
  #define SEGMENT_LEVELED_MOVES
  #define LEVELED_SEGMENT_LENGTH 5.0 // (mm) Length of all segments (except the last one)

  #define MESH_INSET 0              // Set Mesh bounds as an inset region of the bed
  #define GRID_MAX_POINTS_X 15      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle
  #define UBL_SAVE_ACTIVE_ON_M500   // Save the currently active mesh in the current slot on M500

#define X_PROBE_OFFSET_FROM_EXTRUDER 0  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 50  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
#define MIN_PROBE_EDGE 0
```

Bed Topography Report:
```
    (  0,320)                                                                                                      (266,320)
        0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
14 | +0.548  +0.481  +0.427  +0.368  +0.303  +0.279  +0.223  +0.234  +0.196  +0.235  +0.279  +0.342  +0.377  +0.473  +0.556
   |
13 | +0.418  +0.356  +0.315  +0.257  +0.185  +0.153  +0.154  +0.120  +0.113  +0.155  +0.188  +0.260  +0.331  +0.410  +0.490
   |
12 | +0.301  +0.223  +0.220  +0.161  +0.139  +0.095  +0.084  +0.059  +0.057  +0.113  +0.149  +0.195  +0.277  +0.372  +0.458
   |
11 | +0.256  +0.220  +0.188  +0.140  +0.109  +0.090  +0.079  +0.057  +0.088  +0.106  +0.156  +0.204  +0.291  +0.374  +0.438
   |
10 | +0.258  +0.235  +0.203  +0.159  +0.121  +0.091  +0.103  +0.105  +0.105  +0.130  +0.199  +0.274  +0.330  +0.382  +0.478
   |
 9 | +0.285  +0.221  +0.204  +0.175  +0.168  +0.141  +0.105  +0.124  +0.149  +0.176  +0.243  +0.317  +0.356  +0.444 [+0.538]
   |
 8 | +0.302  +0.269  +0.264  +0.187  +0.170  +0.142  +0.148  +0.160  +0.184  +0.197  +0.283  +0.357  +0.411  +0.521  +0.600
   |
 7 | +0.304  +0.270  +0.230  +0.193  +0.162  +0.159  +0.154  +0.167  +0.201  +0.218  +0.294  +0.365  +0.456  +0.537  +0.634
   |
 6 | +0.286  +0.252  +0.209  +0.160  +0.154  +0.124  +0.136  +0.140  +0.186  +0.232  +0.268  +0.366  +0.472  +0.555  +0.641
   |
 5 | +0.199  +0.166  +0.150  +0.118  +0.095  +0.086  +0.089  +0.102  +0.150  +0.188  +0.253  +0.360  +0.444  +0.555  +0.673
   |
 4 | +0.158  +0.134  +0.117  +0.079  +0.088  +0.063  +0.081  +0.085  +0.135  +0.202  +0.278  +0.361  +0.465  +0.574  +0.653
   |
 3 | +0.176  +0.166  +0.106  +0.085  +0.089  +0.073  +0.107  +0.105  +0.183  +0.228  +0.309  +0.406  +0.506  +0.608  +0.699
   |
 2 | +0.193  +0.199  +0.106  +0.091  +0.090  +0.083  +0.132  +0.124  +0.230  +0.255  +0.341  +0.451  +0.548  +0.641  +0.745
   |
 1 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
   |
 0 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
        0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
    (  0,  0)                                                                                                      (266,  0)

```

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

CrNMGuy, 2019-07-21:
Sure, here it is
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3414976/Marlin.zip)

shitcreek, 2019-08-31:
In the video, is the head stuttering at the probe points in question? Have you tried with less probe points?

Roxy-3D, 2019-08-31:
Using a 15 x 15 mesh can exhaust memory.   How much memory do you have after you finish the compile and start the down load?    It is important to have 1500 bytes of variable memory free.

But independent of that...   Your back row is +.5mm high.  Can we lower that .5mm ???   That will make things much easier to debug.

boelle, 2019-10-12:
@CrNMGuy fixed your problem?

CrNMGuy, 2019-10-12:
> 
> 
> @CrNMGuy fixed your problem?

Just got back home, will check things next 15 days

Thanks !

CrNMGuy, 2019-10-16:
>  It is important to have 1500 bytes of variable memory free.
Memory Usage 
DATA:    [=======   ]  72.2% (used 5914 bytes from 8192 bytes)
PROGRAM: [========  ]  79.3% (used 201424 bytes from 253952 bytes)

2278 bytes available

> Using a 15 x 15 mesh can exhaust memory.
trying now with 9x9

CrNMGuy, 2019-10-16:
Here is topografy in 9x9
`Bed Topography Report:
    (  0,320)                                                      (266,320)
        0       1       2       3       4       5       6       7       8
 8 | +0.570  +0.512  +0.427  +0.393  +0.352  +0.369  +0.405  +0.512  +0.614
   |
 7 | +0.423  +0.362  +0.293  +0.263  +0.234  +0.262  +0.351  +0.447  +0.575
   |
 6 | +0.251  +0.234  +0.201  +0.165  +0.180  +0.204  +0.299  +0.387  +0.501
   |
 5 | +0.266  +0.268  +0.241  +0.223  +0.228  +0.277  +0.362  +0.478  +0.567
   |
 4 | +0.306  +0.288  +0.279  +0.272  +0.300  +0.334  +0.457  +0.570  +0.695
   |
 3 | +0.233  +0.213  +0.197  +0.211  +0.240  +0.302  +0.443  +0.570  +0.729
   |
 2 | +0.170  +0.148  +0.151  +0.166  +0.226  +0.297  +0.427  +0.593  +0.737
   |
 1 |[  .   ]   .       .       .       .       .       .       .       .
   |
 0 |   .       .       .       .       .       .       .       .       .
        0       1       2       3       4       5       6       7       8
    (  0,  0)                                                      (266,  0)`

CrNMGuy, 2019-10-16:
And works fine with 14x14
(then tried again with 15x15 and going wrong again)

boelle, 2019-10-22:
so maybe there is just to little memory left with 15x15 ?

CrNMGuy, 2019-10-22:
> Using a 15 x 15 mesh can exhaust memory. How much memory do you have after you finish the compile and start the down load? It is important to have 1500 bytes of variable memory free.

With 15x15, free memory is 2278 bytes 
```
Memory Usage 
DATA:    [=======   ]  72.2% (used 5914 bytes from 8192 bytes)
PROGRAM: [========  ]  79.3% (used 201424 bytes from 253952 bytes)

```

boelle, 2019-10-22:
use 14x14 ? 2278 might be to little

CrNMGuy, 2019-10-22:
See last messages, 14x14 is working great, no issue
I will stay 14x14 
but about the Marlin Code, is there a notice to write ? 
is it supposed low memory that do a bug or the code ?

CrNMGuy, 2019-10-22:
> But independent of that... Your back row is +.5mm high. Can we lower that .5mm ??? That will make things much easier to debug.

Sorry miss that. 
Don't understand the action needed. Would you have any direction for me ?

boelle, 2019-10-22:
adjust the bed at the back row so its 0.5mm lower

CrNMGuy, 2019-10-22:
Oh yes, will do !

boelle, 2019-10-22:
the goal is to have things as flat as you can without using UBL

the better the hardware the less the software has to compensate and the better the prints will be

boelle, 2019-10-30:
i will close this one and assume that @CrNMGuy has solved his problems

we can always reopen if that is not the case

closed 2019-10-30 by boelle