MarlinFirmware/Marlin #14599: [BUG] [bigfix-2.0.x]  Re-arm 32 and Ramps1.6 hot bed still on after M112

KrX3D, 2019-07-12:
### Description

I got an re-arm 32 board:
https://www.aliexpress.com/item/33007283020.html?ws_ab_test=searchweb0_0%2Csearchweb201602_10_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103%2Csearchweb201603_53%2CppcSwitch_0&algo_expid=0eeedca5-5fbc-41cf-9b42-d849ba235c2e-0&algo_pvid=0eeedca5-5fbc-41cf-9b42-d849ba235c2e

and an ramps 1.6 board:
https://www.aliexpress.com/item/32919533439.html?spm=2114.search0104.3.1.78ee12a28zqul9&ws_ab_test=searchweb0_0%2Csearchweb201602_10_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103%2Csearchweb201603_53%2CppcSwitch_0&algo_expid=739bf55a-89db-44a9-9096-aeec618caf62-0&algo_pvid=739bf55a-89db-44a9-9096-aeec618caf62

i use on D8 (HOT BED) an Spindle and i power it on with "M106 P1 S255" to power it of i use "M106 P1 S0" that works.

but when i use "M112" the  Spindle does not power off. the LCD shows printer halted, and all movements are stopped only the spindle does not power off.
hitting the reset button on the re-arm resets eveything and the spindle goes off

### Steps to Reproduce

1. "M106 P1 S255"
2. "M112"
3. all movement stops only spindle is still ON

**Expected behavior:** [What you expect to happen]
after M112 the spindle (or hot bed) should also power off

**Actual behavior:** [What actually happens]
after the emergency M112 spindle (hot bed) is still powered on

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

KrX3D, 2019-07-21:
[Marlin Config.zip](https://github.com/MarlinFirmware/Marlin/files/3414463/Marlin.Config.zip)

KrX3D, 2019-08-22:
hi, anything i can help with? or any idea what could be wrong?

boelle, 2019-09-24:
@KrX3D is the issue still there?

KrX3D, 2019-09-25:
@boelle i downloaded the newest version and tried it today but it still does not work

since i maybe changed some things since last time here the newest config files:

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3651853/Marlin.zip)

boelle, 2019-09-25:
maybe @shitcreek knows, i think he have just looked at the spindle part of marlin

shitcreek, 2019-09-25:
Currently, as far as I can tell from the code, there is nothing that explicitly tells the spindle to turn off when issue M112 (kill).
Inside `temperature.cpp`:
```
void Temperature::disable_all_heaters() {
...
...
  #if HAS_HEATED_BED // this doesn't get defined unless a bed temp sensor was provided
    setTargetBed(0);
  #endif...
```
A condition needs to be added for the case of a laser/spindle.

I'm currently testing out the new laser/spindle codes (#15335) but they have yet to be merged to the main branch and there is nothing in it that addresses this issue with M112.

Another similar issue is to turn off the laser/spindle when a job is canceled.

boelle, 2019-10-21:
i added the confirm label to make the issue stand out on the list
and i think there might be a relation to https://github.com/MarlinFirmware/Marlin/issues/14863

boelle, 2019-11-03:
@KrX3D many updates have passed since 25 sep, is this still an issue?

KrX3D, 2019-11-04:
hi, i will need to check on saturday when im back home. i tried it again like 2.5 weeks ago and it didnt work. i will report back when i tested it on saturday

KrX3D, 2019-11-08:
hi, i tried it with the newest firmware of marlin. but it still does not work. when i enter M112 i get an printer halted message also the display shows it, i lose connection to my printer or cnc machine but the spindle is still working

boelle, 2019-12-07:
@KrX3D since 2.0 was just released a few days ago has this changed this issue at all?

KrX3D, 2019-12-21:
hi,
 i tested the newest version and it is still happening.

Like i said before the spindle is connected to D8, but when i send "M106 P1 S10" i can see than an LED on the ramps 1.6 board is switching to ON and next to it it says D6 (dont know why i would think that an led on D8 should power on?)

or is it happening because of my ramps 1.6?

[image]

i cant remember if this was happening before but i noticed than when i send M112 and the printer "resets" that the spindle "powers off" or "stops" for a friction of a second . as if power shuts down and than powers on again.

i also got a question, each time i use M112 i need to manually reset the rearm board via the button, cant it reset itself?

boelle, 2020-01-17:
@KrX3D @Vertabreak just did some test and the only thing that does not work as the wiki says are steppers, those are not disabled but everything else is

he will look at code and maybe comment on this later

boelle, 2020-01-17:
#16592

boelle, 2020-01-20:
@KrX3D this was fixed ~7 hours ago, please retest with a new copy of bugfix 2.0.x

boelle, 2020-01-24:
will close as it was fixed with a PR by @thinkyhead

closed 2020-01-24 by boelle

KrX3D, 2020-01-24:
Hello,

i downloaded today the newest version and modified the config files.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4110421/Marlin.zip)

but it still does not work. like i said before when i hit M112 i can hear for a friction of a second the spindle to lose speed and than it starts again at full speed.