MarlinFirmware/Marlin #14600: [BUG] Inverted directions: Y_HOME_BUMP and LEVEL_BED_CORNERS wrong direction

Ionic97, 2019-07-12:
When homing on an Ender 5. the HOME_BUMP prodecure is executed properly.
On the Y-axis meanwhile, the carriage is getting pressed further into the endstop.

Also LEVEL_BED_CORNERS seems to crash the carriages into the frame as well. 

It seems like there are some problems with the INVERT settings.

Related settings:
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG

#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true

#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

thisiskeithb, 2019-07-12:
Are you using the latest configs from this Marlin repo and not somewhere else? This is what's set in the [Ender-5 configs](https://github.com/MarlinFirmware/Marlin/tree/82cafc9b1992541ecc61a5fb0ca426ea6e96ccc8/config/examples/Creality/Ender-5), not what you have listed:

https://github.com/MarlinFirmware/Marlin/blob/82cafc9b1992541ecc61a5fb0ca426ea6e96ccc8/config/examples/Creality/Ender-5/Configuration.h#L604-L606

https://github.com/MarlinFirmware/Marlin/blob/82cafc9b1992541ecc61a5fb0ca426ea6e96ccc8/config/examples/Creality/Ender-5/Configuration.h#L999-L1001

https://github.com/MarlinFirmware/Marlin/blob/82cafc9b1992541ecc61a5fb0ca426ea6e96ccc8/config/examples/Creality/Ender-5/Configuration.h#L1024-L1026

As for `LEVEL_BED_CORNERS`, that is not enabled by default, so you might have to increase the value for `LEVEL_CORNERS_INSET`.

Ionic97, 2019-07-12:
These settings are not compatible with the Ender 5. The Z-axis is homing in the wrong direction and thy Y-xis still bumps into the frame. Also the homed position is at X220 Y220 and not X0 Y0 like stock.
These settings could perhaps be based on an earlier unit of the Ender 5. I think creality changed the endstop/homing position at some point.

thisiskeithb, 2019-07-12:
> I think creality changed the endstop/homing position at some point.

Oh, great. Gotta love undocumented changes.

If you change the `*_HOME_DIR` settings to:

```c++
 #define X_HOME_DIR -1 
 #define Y_HOME_DIR -1 
 #define Z_HOME_DIR 1
```

...does it home correctly?

Ionic97, 2019-07-12:
ok, now this is weird.
So first, these
```
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG

#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1
```
should be the right settings. This way it homes to the back right corner with X0 Y0 Z0.

Except, the Y axis seems to be bugged in my case. It doesn't matter in which direction I turn the dial in manual movement, it only moves towards the back of the printer. `INVERT_Y_DIR` doesn't seem to make the difference. 
I'm gonna download a fresh repo and compare if I maybe killed some code...

thisiskeithb, 2019-07-12:
I'm looking at some other Ender-5 source code zips floating around and it looks like the `INVERT_Z_DIR` should be `TRUE` and `Z_HOME_DIR` should be `-1`.

The `*_PLUG` settings seem to match across each config:
```c++
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMIN_PLUG
```
So I'm not sure how or why `USE_XMIN_PLUG` & `USE_YMIN_PLUG` works for you unless Creality is using a totally different board now. Which board do you have?

Here's a Marlin 1.1.9 config from Michael over at Teach Tech if you want to give it a try: 
[Ender-5-Marlin-1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/3387478/Ender-5-Marlin-1.1.9.zip)

Ionic97, 2019-07-12:
Oh yeah, seems like I've missed something there. I have searched for documentation on Creality's 1.1.4 board, but haven't found anything. It seems like this board uses a XMAX and YMAX plug.
I've got the FYSETC Cheetah V1.1 yesterday and try to modify it for the Ender 5. It uses all MIN plugs as you can see in the [Board Layout](https://wiki.fysetc.com/images/Cheetah_Real.jpg).

Sadly, this doesn't clear up my original problem for which I haven't found a fix yet.

thisiskeithb, 2019-07-12:
How old is your printer? Is there a sticker with the SN/date?

You might try [TH3D's fork of Marlin](https://github.com/houseofbugs/TH3D-Unified-U1.R2) as well since they support the Ender-5. If your printer still doesn't home correctly, then there's a possibility that this is a new hardware revision or something weird with your particular printer (this is the first issue reported for these configs, so it's difficult to say).

Ionic97, 2019-07-12:
The original box which went to Comgrow has a shipping label from Shenzen. It states 26.04.19, so it should be from around that time.
I have modified the [FYSETC fork](https://github.com/FYSETC/Marlin-2.0.x-FYSETC) to work for now, where it homes correctly (including the bumpers), but this one is build upon the master from July 2, so not really up-to-date anymore.

Ionic97, 2019-07-12:
Alright. Fresh repo didn't solve the problem. A search in the FYSETC forum showed, that other people seem to have problems with the official fork as well. Using and modifying the FYSETC fork resolves the problem.
I will try to further look into this board-specific issue and maybe a resolution can be implemented into the official fork. In the meantime this workaround has to be enough, so I'm gonna close the issue.

closed 2019-07-12 by Ionic97

thisiskeithb, 2019-07-12:
I'll be able to get my hands on an Ender-5 early next week, so I'll check as well. I'll probably put in a PR to update the z homing/inverting & I'll see what else needs fixing in the Marlin configs.

>  This way it homes to the back right corner with X0 Y0 Z0.

I don't understand this, though. Wouldn't that be X235/Y235 if it were the back right corner since you're at the max X & Y? X0/Y0 should be front left corner of the build plate.

Ionic97, 2019-07-12:
Well this depends on the perspective, but you are not wrong. My stock Ender-5 has it's X0 Y0 in the back right corner and I believe that's the standard on newer units. There are some other settings concerning feedrates etc. which are a bit different from the config on stock settings. I might open a PR for them once I've got everything up and running again.

thisiskeithb, 2019-07-12:
Not all settings will be a 1:1 transfer, but I'd be inclined to push for the proper config and fix the home position to the front left. This would also rotate your prints around the correct way to match what you see in your slicer.