MarlinFirmware/Marlin #14601: [BUG]Miss-guided G1 command crashes latest BF2

oneofnine, 2019-07-12:
Due to a problem in Cura 4 (just going to look for it), it inserted some crazy G1 commands into it's GCode output file which Marlin can't deal with as Marlin just crashes out and reboots part way through printing.

Looks like you need to add some sanity checking when receiving some GCode commands, namely G1 (but no doubt G0 etc).

Here's a line from Cura's output that gets sent to Marlin and bombs it out ..

G1 F17987547480000 Z10.25

OK, yes, the 'F' param is extremely wrong, but Marlin really ought to deal with it by limiting the parameter values.

thisiskeithb, 2019-07-12:
See https://github.com/MarlinFirmware/Marlin/pull/13929.

oneofnine, 2019-07-12:
I'll close this but add to the other threads.

closed 2019-07-12 by oneofnine