MarlinFirmware/Marlin #14604: [BUG] (short description)

boubouexpress, 2019-07-12:
hello

after activating #define Z_DUAL_STEPPER_DRIVERS the engine of the   of the driver  Z2 (E1)works but makes a hissing sound

I switched the cables and changed the driver no result

#define X_DRIVER_TYPE  TMC2130
#define Y_DRIVER_TYPE  TMC2130
#define Z_DRIVER_TYPE  TMC2130
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2130
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE LV8729
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988

merci

InsanityAutomation, 2019-07-12:
Sounds like vref needs to be adjusted, is there a POT near the driver? Compare to the other z driver that isnt hissing.

boubouexpress, 2019-07-12:
I forgot to say that I checked the driver adjustments in relation to the amperage of the motors

boubouexpress, 2019-07-18:
Résolu
J’avais oublier activer pour le Z2 le stepping mode 
Désolé

closed 2019-07-18 by boubouexpress