MarlinFirmware/Marlin #14607: [FR] Alternate pattern for UBL mesh generation

phizz166, 2019-07-13:
### Description

When I run a G29 P1 on my cartesian printer with a 4x4 grid, starting from the center as that is the G28 z-probing location, it probes in the following pattern:

```
[13]  [12]  [11]  [16] 
[5]   [4]   [3]   [10]
[6]   [1]   [2]   [9]
[14]  [7]   [8]   [15]
```

The comments in the code suggest that it was designed this way so that you can ensure that your settings are right in the center without wasting time at the edges. That's fine for people with delta printers or those who are doing the initial calibration, but this method is actually more time-consuming than a simple zigzag once you know you want to run the whole pattern. Lots of time is spent on travel moves when it doesn't need to be.

I prefer to run the calibration routine anew before every print, so I would like to have the printer just do a rapid move to the corner and then go row-by-row:

```
[16]  [15]  [14]  [13]
[9]   [10]  [11]  [12]
[8]   [7]   [6]   [5]
[1]   [2]   [3]   [4]
```
By my calculations this will only travel 75% of the distance and thus should be 33% faster. 

Yes, a few extra seconds at the beginning of a print is not really a huge deal, but I think this is a simple optimization to make that does save time -- and it just feels more logical for a rectangular bed than the pseudocircle. It appears from the code in /src/core/feature/bedlevel/ubl/ubl_G29.cpp that there may already be logic to do it, but I am not sure if it's possible to just enable it or if more work is required.

Roxy-3D, 2019-07-13:
There is an addition reason for that probing pattern.   It is done that way for the safety of the machine.    If a point has been probed successfully...   The points close to it probably are not that far off (in Z-Height) from that value.    So by probing outward from the starting location gives the user time to react if the nozzle starts getting too close to the bed.

Incidentally...   Bob-the-Kuhn did the work to add the more traditional grid probe pattern to UBL.   When it was all said and done, it only saved a few seconds.    It is true the nozzle travels much more with the expanding outward pattern.   But the bulk of the time is spent doing the actual probes.

gloomyandy, 2019-07-13:
It is also worth noting that you can "fine tune" a mesh by running a G29 J at the start of the print. This will handle any tilt issues that may have developed after the original mesh was created (and/or any small height changes). I find that with a removable spring steel magnetic bed small changes can occur. I usually load my saved mesh and then run G29 J3 (a 3x3 square pattern) at the start of each print.

phizz166, 2019-07-13:
Both very helpful posts. I agree that ideally the mesh regeneration should not have to be done every time....but a bunch of the printers I'm maintaining are in a school lab and they get a decent amount of abuse from the students. One common problem is that a student will inadvertently push on the Z mechanism when removing a print and get the Z leadscrews slightly out of sync, which ruins the next print until they are recalibrated. Generating the mesh is quicker than the calibration process for the leadscrews, so I figured that would be the best move.

However, @Roxy-3D 's points about fine-tuning and safety are well-taken, and I hadn't realized that you could combine the 3x3 and full mesh methods as @gloomyandy says. That seems like it would be the best of both worlds -- a high resolution base mesh that can be tweaked just before the print to account for something like the Z axis being slightly desynchronized. Thanks!

EvilGremlin, 2019-07-13:
If you have dual Z screws - you should consider modding printer to use dual Z motors and endstops instead of UBL.

closed 2019-07-15 by phizz166