MarlinFirmware/Marlin #14609: [BUG] RESTORE_LEVELING_AFTER_G28 keeps MBL OFF

Zemistr, 2019-07-13:
### Description

I have enabled:
```
EEPROM_SETTINGS

MESH_BED_LEVELING 
PROBE_MANUALLY 
LCD_BED_LEVELING

RESTORE_LEVELING_AFTER_G28 
```
but I am not able to have `Mesh Bed Leveling ON` without `M420 S1`.

### Steps to Reproduce

1. G29 S0 (Mesh Bed Leveling OFF)
2. G28
3. G29 S0 (Mesh Bed Leveling OFF)
4. M420 S1 (echo:Bed Leveling ON)
5. G29 S0 (Mesh Bed Leveling ON)
6. M500 (echo:Settings Stored (606 bytes; crc 19598)
7. G29 S0 (echo:Bed Leveling ON)
8. RESET PRINTER
9. G29 S0 (Mesh Bed Leveling OFF)
10. G28
11. G29 S0 (Mesh Bed Leveling OFF) ???

**Expected behavior:** [What you expect to happen]
Mesh Bed Leveling **ON** after `G28`.

**Actual behavior:** [What actually happens]
Mesh Bed Leveling **OFF** after `G28`.

**Temporary fix:**
Add `M420 S0` to your start G-Code after `G28`.

#### Additional Information
* Configs: [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3388536/Marlin.zip)
* Marlin version: https://github.com/MarlinFirmware/Marlin/commit/91bb0f54395eca7b8898cb94e7aed30c7b6a949d
* Video: https://youtu.be/TsWabkw5COs

gloomyandy, 2019-07-13:
I think you are misunderstanding what this option does. It keeps the previous state of mesh levelling after G28 (by default it will always turn it off). 
```C
**
 * Normally G28 leaves leveling disabled on completion. Enable
 * this option to have G28 restore the prior leveling state.
 */
//#define RESTORE_LEVELING_AFTER_G28
```
In your example above you said that before the G28 levelling was off, the G28 will simply have preserved that state, so it will still be off after G28.

BETLOG, 2019-07-13:
Memory is fuzzy now, so maybe I'm wrong, but iirc you need UBL on... As in: one of the more advanced types, not just bilinear.

Zemistr, 2019-07-13:
Yeah, I think it's just my misunderstanding.
I expected the leveling state is stored in EEPROM.
So after G28, it would recover the state from EEPROM, not just from SRAM.
But I understand, those are different things.
Is there some way how to keep leveling ON also after restart?

LastDragon-ru, 2019-07-13:
> Is there some way how to keep leveling ON also after restart?

I'm using folowing gcodes with `AUTO_BED_LEVELING_UBL` + `RESTORE_LEVELING_AFTER_G28`

```
# G29 P1       ; Bed Leveling
# G29 P3       ; Fill Unpopulated regions of the mesh (smart fill)
# G29 S1       ; Save UBL mesh points to EEPROM
# G29 A        ; Activate the UBL System
# M500         ; Save firmware
```

closed 2019-07-14 by Zemistr