MarlinFirmware/Marlin #14610: Hot end not reaching target temperature. Steppers won't move.

ChrisS888, 2019-07-13:
SKR 1.3
TMC2130 SPI mode
BLTouch
Ender 3
EEPROM_SETTINGS enabled
Download of latest update @ 13-7-19

M122 reports all good
M500 works
Steppers refuse to move - any attempt to move any axis causes fault condition, printer locks out.

From reset (i.e. not after a fault lockout).
If hot end commanded to heat, will get to within a few degrees of target then heater goes off, temp drops, heater comes back on, temp climbs but not quite as high as before, then alarm triggers and printer locked out. Seems to be an offset somewhere??

Tried 180 target, reached 174.0, then 167.7 then 173 then failed with alarm.

Tried 80 target, reached 77.7 then 74.4 then failed with alarm.

Heated bed works nominally. Tried target temps of 40 and 80, both worked as expected. No alarms.

Config.h and config_adv.h attached in archive.zip

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3388673/Archive.zip)

This exact hardware works nominally with previous releases apart from the EEPROM_SETTTINGS issue. (Reported in case #14579)

Thank you.

gloomyandy, 2019-07-13:
I'm totally confused by this part of your report... "Steppers refuse to move - any attempt to move any axis causes fault condition, printer locks out". What exactly do you mean by this? What "fault condition" do you see? Without doing any sort of heating are you able to move the steppers? 

Are you saying that your previously reported eeprom settings are resolved?

MarkMan0, 2019-07-13:
For the hotend, try PID tuning. 
What fault condition?

ChrisS888, 2019-07-13:
> I'm totally confused by this part of your report... "Steppers refuse to move - any attempt to move any axis causes fault condition, printer locks out". What exactly do you mean by this? What "fault [...]

If I try to HOME the printer, the LCD screen blanks then displays:

"Homing failed

PRINTER HALTED

Please reset"

If I try to manually move an axis either using the LCD or via Octoprint, the commands are ignored.

EEPROM_SETTINGS issue is now fixed, in as much as if it is enabled, and I issue an M122 command the report shows all is well with the drivers. They won't actually move though, as above. Previous versions, if EEPROM_SETTINGS enabled would result in an on-LCD display of TMC DRIVER ERROR right from boot, M122 reported bad drivers.

Re the hot end warmup issue, I've compared a working version (bar the ability to enable EEPROM_SETTINGS)  with the latest one referred to on this report and the values are identical, hence I reported it as a bug. Same settings, same hardware, different result - has to be a bug?  Heated bed is fine.

Hope that clarifies, please ask if more testing/detail required, and thanks for the thoughts/comments. All welcome.

gloomyandy, 2019-07-13:
With that combination of problems I'd be taking a hard look at things like the power supply and wiring.

ChrisS888, 2019-07-13:
I get why you say that, but with the 'right' firmware (and no other change whatsoever), it prints just fine. 

Admittedly a significant change, but this machine has been working 100% on many prints, some 15 hour plus, with an MKS-GEN-L board, right up until I started messing with the SKR & Marlin 2. Other than the main board, no other changes to the environment.

These things make me think it's firmware, not hardware. That, plus the fact it seems different almost every release, but I can reliably create the issues on a given build. Others seem to be using this board without major problems though, so yeah, puzzling.

gloomyandy, 2019-07-13:
Have you gone back to the "good" firmware and checked it still works? 

Oh and make sure that when you install a version of the firmware that has eeprom enabled that you are doing an m502 followed by m500. You might also want to try the latest firmware but do not enable eeprom.

ChrisS888, 2019-07-14:
I have, and yes, it does work still. At least the SKR is quick & easy to re-flash....  :)

Thanks for the M502 hint - when(!) I get one fully working I'll make sure to use that.

Latest version with EEPROM_SETTINGS disabled - check, tried that too, no change.

Appreciate the thoughts.

ChrisS888, 2019-07-16:
During further investigation, I have concluded that this isn't a Marlin bug, it was caused by corruption of the flash process due the Raspberry Pi Octoprint host being connected to the USB port of the SKR.

Exact same firmware file was re-flashed to the exact same hardware but by chance, Pi had been unplugged to allow direct connection to computer.

Of interest, even if it's academic, is that the corruption is totally repeatable and consistent. I must have flashed the same firmware at least 6 times with the Pi connected and got identical results.

Printer now functions nominally, communication with Octorpint/Pi is also nominal.

TFL

closed 2019-07-16 by ChrisS888

thisiskeithb, 2019-07-16:
Very odd. I flash my SKR 1.3 multiple times a week via OctoPrint and have not run into that issue.

Have you tried a different USB cable?

ChrisS888, 2019-07-17:
Humpf. This just gets weirder. I tried again just now - exact same firmware file I've been using since i generated it on the 13th, from the build I pulled that day. With the Pi connected but a different USB cable. For clarity, I'm using a direct transfer using a micro SD card from computer to SKR, not uploading via USB/Pi.

So, usual routine as used many times up until now. Power up, insert SD card, hit reset button on SKR, wait for flash to complete and Marlin to boot, remove SD card, test. This time, it works.

Put old USB cable back, repeat process. It works.

From the start, I have always made the assumption that I'm doing something wrong. I began to suspect a hardware issue when I tried a known-to-work firmware.bin file that still didn't work, then I went back to suspecting I was making mistakes when I got a partially working version, then I tended toward just thinking it was down to the very fluid state of Marlin at this time and I'd just wait a bit.

All through this was the nagging thought that others seem to have this board working fine, so kept coming back to my original thought, what was I doing wrong.

I've still no idea! But, I do now have a working SKR board, so I'll take it.

Only thing I can think of that I have done along the way recently that I didn't do before was to reset the EEPROM, but I haven't been rigorous nor made notes so cannot say at what point in the process I did this. Could it be that simple? Corrupt data in there causing all the grief?

Anyhoo, thank you to everyone involved with Marlin. I'm new to the whole open source ethos - you folks are amazing!