MarlinFirmware/Marlin #14611: you can not remove splash lines from boot screen

gemakochi, 2019-07-13:
you can not disable STRING_SPLASH_LINE1 with any lcd controller enabled.

minimal changes that produce error in bugfix-2.0.x
```
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 6880d7bbe..aba663a99 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -78,2 +78,2 @@
-#define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
-#define STRING_SPLASH_LINE2 WEBSITE_URL         // will be shown during bootup in line 2
+//#define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
+//#define STRING_SPLASH_LINE2 WEBSITE_URL         // will be shown during bootup in line 2
@@ -1920 +1920 @@
-//#define MKS_MINI_12864
+#define MKS_MINI_12864
```

arduino ide output
```
sketch/src/lcd/dogm/ultralcd_DOGM.cpp: In static member function 'static void MarlinUI::show_bootscreen()':
sketch/src/lcd/dogm/ultralcd_DOGM.cpp:165:55: error: 'STRING_SPLASH_LINE1' was not declared in this scope
                         text_width_1 = uint8_t(sizeof(STRING_SPLASH_LINE1) - 1) * uint8_t(MENU_FONT_WIDTH),
                                                       ^
sketch/src/lcd/dogm/ultralcd_DOGM.cpp:172:59: error: 'text_width_2' was not declared in this scope
     constexpr uint8_t text_max_width = _MAX(text_width_1, text_width_2),
                                                           ^
sketch/src/lcd/dogm/ultralcd_DOGM.cpp:178:26: error: 'rspace' was not declared in this scope
     if (text_max_width < rspace) {
                          ^
exit status 1
```

i managed to solve issue with following changes.
```
diff --git a/Marlin/src/lcd/dogm/ultralcd_DOGM.cpp b/Marlin/src/lcd/dogm/ultralcd_DOGM.cpp
index c978a07d0..022681733 100644
--- a/Marlin/src/lcd/dogm/ultralcd_DOGM.cpp
+++ b/Marlin/src/lcd/dogm/ultralcd_DOGM.cpp
@@ -160,16 +160,23 @@ void MarlinUI::set_font(const MarlinFont font_nr) {
     constexpr uint8_t width = LCD_PIXEL_WIDTH, height = LCD_PIXEL_HEIGHT;
 
     // Determine text space needed
-    #ifndef STRING_SPLASH_LINE2
-      constexpr uint8_t text_total_height = MENU_FONT_HEIGHT,
-                        text_width_1 = uint8_t(sizeof(STRING_SPLASH_LINE1) - 1) * uint8_t(MENU_FONT_WIDTH),
-                        text_width_2 = 0;
+    #ifdef STRING_SPLASH_LINE1
+      constexpr uint8_t text_width_1 = uint8_t(sizeof(STRING_SPLASH_LINE1) - 1) * uint8_t(MENU_FONT_WIDTH);
+      #define SPLASH_LINE1_HEIGHT MENU_FONT_HEIGHT
     #else
-      constexpr uint8_t text_total_height = uint8_t(MENU_FONT_HEIGHT) * 2,
-                        text_width_1 = uint8_t(sizeof(STRING_SPLASH_LINE1) - 1) * uint8_t(MENU_FONT_WIDTH),
-                        text_width_2 = uint8_t(sizeof(STRING_SPLASH_LINE2) - 1) * uint8_t(MENU_FONT_WIDTH);
+      constexpr uint8_t text_width_1 = 0;
+      #define SPLASH_LINE1_HEIGHT 0
     #endif
-    constexpr uint8_t text_max_width = _MAX(text_width_1, text_width_2),
+    #ifdef STRING_SPLASH_LINE2
+      constexpr uint8_t text_width_2 = uint8_t(sizeof(STRING_SPLASH_LINE2) - 1) * uint8_t(MENU_FONT_WIDTH);
+      #define SPLASH_LINE2_HEIGHT MENU_FONT_HEIGHT
+    #else
+      constexpr uint8_t text_width_2 = 0;
+      #define SPLASH_LINE2_HEIGHT 0
+    #endif 
+    
+    constexpr uint8_t text_total_height = SPLASH_LINE1_HEIGHT + SPLASH_LINE2_HEIGHT,
+                      text_max_width = _MAX(text_width_1, text_width_2),
                       rspace = width - (START_BMPWIDTH);
 
     int8_t offx, offy, txt_base, txt_offx_1, txt_offx_2;
@@ -197,11 +204,13 @@ void MarlinUI::set_font(const MarlinFont font_nr) {
     do {
       u8g.drawBitmapP(offx, offy, (START_BMPWIDTH + 7) / 8, START_BMPHEIGHT, start_bmp);
       set_font(FONT_MENU);
-      #ifndef STRING_SPLASH_LINE2
-        u8g.drawStr(txt_offx_1, txt_base, STRING_SPLASH_LINE1);
-      #else
-        u8g.drawStr(txt_offx_1, txt_base - (MENU_FONT_HEIGHT), STRING_SPLASH_LINE1);
-        u8g.drawStr(txt_offx_2, txt_base, STRING_SPLASH_LINE2);
+      #define TXT_OFFY 0
+      #ifdef STRING_SPLASH_LINE2
+        u8g.drawStr(txt_offx_1, txt_base, STRING_SPLASH_LINE2);
+        #define TXT_OFFY MENU_FONT_HEIGHT
+      #endif
+      #ifdef STRING_SPLASH_LINE1
+        u8g.drawStr(txt_offx_1, txt_base - (TXT_OFFY), STRING_SPLASH_LINE1);
       #endif
     } while (u8g.nextPage());
     #ifndef BOOTSCREEN_TIMEOUT
```

gloomyandy, 2019-07-13:
If you don't want any text to be displayed wouldn't it have been easier just to have defined the text as...
```C
-#define STRING_SPLASH_LINE1 "" // will be shown during bootup in line 1
-#define STRING_SPLASH_LINE2 ""         // will be shown during bootup in line 2
```
or something similar?

But anyway is there a reason why you do not want to display the Marlin information? In general it is preferred to do so.

gemakochi, 2019-07-13:
no intention to hide marlin info, i was messing around and find out i can comment out STRING_SPLASH_LINE2 but not STRING_SPLASH_LINE1. there is already controls for line2 but not for line1 so i want to share.

if you ask me these should not be configurable, there is already a custom boot screen option.

InsanityAutomation, 2019-07-14:
But if you dont want to add the time of a second boot screen and merely want to show a version number on boot, its perfect.

Roxy-3D, 2019-07-14:
@gemakochi   The Marlin Contributors have literally donated over 100,000 hours of work to produce the current code base.   The Splash screens are put up while other things are happening in the firmware.   They are not delaying the responsiveness of the firmware getting to the point of where it can accept commands from the user.

Why do you want to hide the fact that Marlin is running on your printer?

InsanityAutomation, 2019-07-14:
@Roxy-3D I think he just wants to eliminate the text that usually just contains the version number or website url. Logo would still be shown above so its not hiding whats on it, just a slightly cleaner picture screen.In ultralcd_dogm.cpp it actually has code around if line 2 is disabled : 

`#ifndef STRING_SPLASH_LINE2
      constexpr uint8_t text_total_height = MENU_FONT_HEIGHT,
                        text_width_1 = uint8_t(sizeof(STRING_SPLASH_LINE1) - 1) * uint8_t(MENU_FONT_WIDTH),
                        text_width_2 = 0;
    #else`

So it can handle no website, but version number is always displayed.

I think to alleviate the original issue, a line in conditionals_post

#ifndef STRING_SPLASH_LINE1
  #define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION
#endif

would mean if either line doesnt exist in the configuration it still happily builds.

Roxy-3D, 2019-07-14:
Thank You @InsanityAutomation !   But with no disrespect intended...   Let's hear his explanation.   
Admitably, I am overly sensitive on this topic.   But if somebody wants to strip the Marlin Logo out of the boot up sequence, it seems pretty brash to ask us for help doing that.   Especially since it doesn't slow anything down.

If there is a REAL reason for doing that...  I'm all ears.   But if somebody wants to hide the fact they are using 100,000 man hours of expertise coding skills on a printer they are selling...  Too bad!  They can pay somebody $75 an hour to get that done.

And especially because of this fact:   Marlin is the best Open Source 3D-Printer firmware out there.   Why would you want to hide the fact you are using the best firmware available?

gemakochi, 2019-07-14:
> I think hes just pointing at the lack of consistency in behavior between the two values.

thank you @InsanityAutomation, that was the issue i want to pointout.

ofcourse lines can be left empty but logo can get bigger instead of displaying empty lines. i would prefer fullscreen logo.

boelle, 2019-09-24:
@gemakochi is the issue still there?

gemakochi, 2019-09-24:
yes, i didn't notice any changes.

closed 2019-09-24 by gemakochi

nkuck, 2020-04-14:
Can anyone explain why "#define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION" displays the wrong installed fw version? Shows a previous one but not the current 2.0.

thinkyhead, 2020-04-26:
@nkuck — 
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y