MarlinFirmware/Marlin #14612: [BUG] view error message

Ludy87, 2019-07-13:
### Description

before homing try 'Set home offsets'

normally LCD show error message with 'Home XYZ first'.
 
### Steps to Reproduce

1. enable `ANET_FULL_GRAPHICS_LCD`
2. LCD: Menu->Configuration->Advanced Settings->Set home offsets

**Expected behavior:** Last line show 'Home XYZ first'

**Actual behavior:** The last line is empty

#### Additional Information

`ui.status_printf_P` don't work
`ui.status_printf_P(0, PSTR(MSG_HOME " %s%s%s " MSG_FIRST), xx ? MSG_X : "", yy ? MSG_Y : "", zz ? MSG_Z : "");`

shitcreek, 2019-07-19:
Looking through the codes for v1.x and v2.x, 'Home XYZ first' is related to UBL and isn't even called anywhere.
And since M206 and M428 set the current position as home and so there wouldn't be a need to home.
http://marlinfw.org/docs/gcode/M206.html

Ludy87, 2019-07-19:
Console message `home XYZ first` in the following line` ui.status_printf_P`

closed 2019-07-28 by Ludy87