MarlinFirmware/Marlin #14614: Creatbot Violation of license terms

ffnull, 2019-07-13:
Hello there. Im recently buy CreatBot f430 and found that this printer uses modified Marlin firmware. CreatBot won't provide sources and firmware. What we can do?

coreyfaure, 2019-07-13:
It looks like this company is no stranger to either violating open-source licenses or just straight up ripping content from other websites. Case in point:
Original article: https://www.simplify3d.com/support/print-quality-troubleshooting/
Their FAQ: https://www.creatbot.com/en/faqs.html

ffnull, 2019-07-13:
how can we punish them.  and make source code available?

thisiskeithb, 2019-07-13:
> how can we punish them.

We don’t buy their products 🙂

OneOfEleven, 2019-07-14:
They have some lovely printers though, the mechanics of their machines look really nice.

ffnull, 2019-07-14:
Im buy printer for 8000$ in my country. And it doesnt work good. I need source codes to fix. But manufacturer is asshole

OneOfEleven, 2019-07-14:
> There's probably not much that can be done with these Chinese companies. Naomi Wu had luck convincing Creality to respect the GPL after a lengthy explanation on what it is and how it would benefit the company to adhere to it, but there's nothing much stopping them from doing this kind of thing unfortunately.

All (chinese, western or where ever) companies use and rip open source code and us it in their products and never share anything, a lot of my open source code is being used in some rather high priced equipment, they never ask, never say, never mention, never share, they just use it, that's how companies are I'm afraid, the only thing they want is profit, nothing else matters.

Such is the human race.

boelle, 2019-07-14:
yep, its not the first time i see a post like this and not the last either

the only way you could prevent this to prevent the firmware working if a license key is entered either with lcd or via seriel.

or if printer can phone home we could tell it to not work 

but the most bullet proff way is just forget about it and go on, china does not give a rats arse about it

boelle, 2019-07-14:
@ffnull 

"Im buy printer for 8000$ in my country. And it doesnt work good. I need source codes to fix. But manufacturer is asshole"

send it back

InsanityAutomation, 2019-07-14:
> yep, its not the first time i see a post like this and not the last either
>
> the only way you could prevent this to prevent the firmware working if a license key is entered either with lcd or via [...]

It being open source, that is guaranteed to annoy users more than manufacturers who skirt the license. Theyll simply remove it. Itll slow down one of their developers for maybe a day.

UAEChan, 2019-07-15:
Their machines have very good mechanics. It's not important that the code is open source and not . they will provide a variety of firmware if you want

UAEChan, 2019-07-15:
I configure Marlin using the configuration tool, or command line

ffnull, 2019-07-15:
Here to many futures that i cant provide while porting marlin 2. So without
source code, i cant make port.

UAEChan, 2019-07-15:
@ffnull I can ask them for you, you want the pins list ?

UAEChan, 2019-07-15:
marlin 2 have too many bugs

ffnull, 2019-07-15:
Yes. Pin map will be good. And how the make usb flash over spi

closed 2019-07-15 by ffnull