MarlinFirmware/Marlin #14615: [BUG] SKR Pro v1.1 and TMC 2209 in UART mode compilation error

biskero, 2019-07-13:
Hello,

I am trying to compile Marlin 2.0.x from yesterday for the SKR Pro v1.1 with TMC2209 in UART mode and I am getting some errors:
[image]

also the Software Serial Pins in pins_BIGTREE_SKR_PRO_V1.1.h should be these, got them from SKR Pro facebook page;
[image]
 
any ideas? I can test things if needed, compiling on Ubuntu 18 in Atom/Platformio.
thx

EvilGremlin, 2019-07-13:
Zip configuration files and attach to next message.

biskero, 2019-07-13:
I have TMCStepper 0.4.5 in Arduino from github.com/teemuatlut/TMCStepper

Here are the config files:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3389207/configuration.zip)

Ludy87, 2019-07-13:
I think @teemuatlut has to work on it again.

Ludy87, 2019-07-13:
Marlin can be compiled with this library FORK (https://github.com/Msq001/TMCStepper.git).

zillarob, 2019-07-13:
I couldnt figure it out the single pin way that is currently in the pins file (Tx and Rx are the same pin) so changed to what is above and seems to work. M122 looks good for all of them anyway.
The board is setup for the 2 pin way and those pin numbers are off the btt pin diagram.

EvilGremlin, 2019-07-13:
Single pin mode depends on support by either hardware port or software library. You cannot just put same ports in config and hope it'll work.

Ludy87, 2019-07-13:
```c++
tmc_util.h:166:40: error: no matching function for call to 'TMC2208Stepper::TMC2208Stepper(const uint16_t&, const uint16_t&, const float&, const bool&)'
       TMC2208Stepper(RX, TX, RS, has_rx)```

zillarob, 2019-07-13:
@EvilGremlin  The pins file in Marlin currently is setup for single pin. I have never used it and not familiar with it so just changed the pins to what is called for in the pin diagram.
Single pin prob works or I would guess the pins file wouldnt be the way it is, but the board is setup for 2 pin so I gave it a shot that way and it worked.

eightheads, 2019-07-13:
> @EvilGremlin The pins file in Marlin currently is setup for single pin. I have never used it and not familiar with it so just changed the pins to what is called for in the pin diagram.
> Single pin [...]

The latest Marlin does not compile with 2 pins defined either. I'm getting the same error
`tmc_util.h:166:40: error: no matching function for call to 'TMC2208Stepper::TMC2208Stepper(const uint16_t&, const uint16_t&, const float&, const bool&)'
       TMC2208Stepper(RX, TX, RS, has_rx)`

zillarob, 2019-07-13:
Good point. I was using the one on btt github and dont remember straying out of the shallow end =D

biskero, 2019-07-14:
yes, I tried compiling both single pins or the above configuration but the error persists.

Ludy87, 2019-07-14:
@biskero 
'platformio.ini'
Change TMCStepper in https://github.com/Msq001/TMCStepper.git

biskero, 2019-07-14:
It worked ! compiled successfully !
So it's the TMCStepper library that needs to be an update.

teemuatlut, 2019-07-14:
SoftwareSerial is not supported internally by the STM32Duino framework.

biskero, 2019-07-15:
@teemuatlut so does it means that the above pin layout is not ok? and the single pin layout is ok?
thx

teemuatlut, 2019-07-15:
No SW Serial means HW Serial only.

Alternatively you can use the Bigtreetech fork or modify upstream Marlin.

eightheads, 2019-07-16:
> No SW Serial means HW Serial only.
> 
> Alternatively you can use the Bigtreetech fork or modify upstream Marlin.

I assume SoftwareSerial for STM32 chipsets will be added sometime relatively soon in the official TMCStepper library?

teemuatlut, 2019-07-16:
TMCStepper will not be adding any internal Serial implementations. That's firmly in the realm of either the platform framework or a supporting library for the platform.
When that exists I can easily add STM into the supported list.

Evg33, 2019-07-18:
#TMCStepper@<1.0.0
  https://github.com/bigtreetech/TMCStepper

boelle, 2019-09-24:
@biskero is the issue still there?

hackebike, 2019-09-24:
@boelle issue is not there anymore

boelle, 2019-09-24:
oki, then i will close, we can always reopen :-D

Evg33, 2019-09-24:
@RudyBenner can you show me yours platformio.ini Configuration.h and Configuration_adv.h  ?

closed 2019-09-24 by boelle

Evg33, 2019-09-24:
PR successfully tested with 2208 uart and 5160 spi.
At 2208, you need to solder 2 blue pins in the middle.
[image]