MarlinFirmware/Marlin #14617: [BUG] Controller fan timeout secs not honored

OneOfEleven, 2019-07-13:
No matter what we set **CONTROLLERFAN_SECS** too the controller fan always times out (turns off) approx 7 seconds after the steppers are deactivated.

```cpp
/**
 * Controller Fan
 * To cool down the stepper drivers and MOSFETs.
 *
 * The fan will turn on automatically whenever any stepper is enabled
 * and turn off after a set period after all steppers are turned off.
 */
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  //#define CONTROLLER_FAN_PIN -1
  #define CONTROLLER_FAN_PIN SERVO0_PIN
  #define CONTROLLERFAN_SECS 10           // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 70          // 255 == full speed
#endif
```

boelle, 2019-09-24:
@OneOfEleven is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle