MarlinFirmware/Marlin #14622: [BUG bugfix-2.0.x] SKR v1.3 Freeze/Reboot on Toolchange

Chris00223, 2019-07-14:
I have a Hypercube using a SKR 1.3, TMC2208 drivers and a Chimera+ hotend. Nozzle changes are handled by firmware. When printing using Cura gcode it pauses on the first toolchange  and reboots. When using Prusa Slicer gcode of the same STL the print completes.  

The last command issued before the reboot is T1, the only differences I can see between Cura and Prusa Slicer is that Cura issues a M105 to report temperature before heating and lowering the temperature after the toolchange, rather than before in Prusa Slicer.

If I use a single nozzle to print using Cura, the print completes. I have changed from absolute extrusion to relative in Cura, to match Slicer, but the problem still happens.

I have attached a zip with the config files and the gcode from Cura and Prusa Slicer. 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3389749/Marlin.zip)

shitcreek, 2019-07-19:
I feel like this might be a slicer issue and not Marlin. As per the documentation, it looks like M105 requires extra process which I assume is the cause. http://marlinfw.org/docs/gcode/M105.html
Try M155 instead - must be configured to use.

Chris00223, 2019-07-19:
The printer rebooted while printing from the LCD SD card without a host connected to USB and behaved the same when connected to Octoprint. 

I commented the 5 instances of M105 out of the file attached and it rebooted in the same place, after looking at the gcode again I noticed this on line 3299:

"G1 F17987547480000 Z1.5"

I changed this to F300 and the file has printed fine, I assume that it is this extreme number that caused the reboot, it seems that Cura believes printers won't be exceeding the speed of light for Z moves. 

The speed of light is 299792458000 mm/s or 17987547480000 mm/min which must be how they arrived at that number, I have the maximum Z speed set to 0, which Cura states will use firmware max speed for Z moves. 

Entering this number for a feedrate causes the printer to reboot, I assume this is causing an overflow of the datatype for feedrates or calculations used by Marlin and the reboot. 

While this number is perhaps an extreme way for Cura to ensure that the printer will use the max firmware feedrate, should this be handled by Marlin to drop it to the max feedrate?

shitcreek, 2019-07-19:
Max feed rates are set in two places: first in your config file: DEFAULT_MAX_FEEDRATE, which gets overridden by the max feed rates set inside your slicer/host

Marlin's default feed rate for Z is 5mm/s - mine in my slicer is set at 12mm/s, but I kept the default max feed rate in the config file at 5mm/s

Chris00223, 2019-07-19:
Yes, mine is set to 5mm/s. But the issue I am talking about is Cura inputting a large number which causes Marlin to reboot, while I can set this to a reasonable number, and I have. My question is whether or not the developers of Marlin wish to handle the error caused by entering a number that large for the feedrate.

Personally I would prefer an error to be displayed or to handle the error silently and drop 17987547480000 or whatever the user/slicer has entered to the max feedrate defined in the firmware, as a reboot doesn't provide any information to the user other than something has caused a reboot and no idea as to what has caused it.

shitcreek, 2019-07-19:
That would be easy enough. I'll change this to a feature request since it isn't a Marlin bug per se

thisiskeithb, 2019-07-20:
Sounds like a dupe of https://github.com/MarlinFirmware/Marlin/issues/14454

Chris00223, 2019-07-20:
Yes, it is a dupe, I had expected that it was a dupe but couldn't find it after finding the cause yesterday. I don't feel that this would be a feature request, as I agree with what OneOfEleven in #14454 said about sanity checking input. I will close this, thanks.

closed 2019-07-20 by Chris00223