MarlinFirmware/Marlin #14624: Custom control panel

Adik22, 2019-07-14:
Hello,

how lcd,tft communicate with board ?
I have SKR v1.3 board and I want to make my own control panel(touch screen) for it.

Thanks :)

Adrian

LastDragon-ru, 2019-07-15:
Probably you can use serial port like mks tft does.

tpruvot, 2019-07-16:
beware: the handled mks tft is not over serial... but some others are using a dedicated system/firmware and act as a remote controller (like octoprint can do over web)

boelle, 2019-07-21:
We're not a support group
This Issue Queue is for Marlin bug reports and development-related
issues, and we prefer not to handle user-support questions here. (As
noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/
1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-ijust-have-a-question).) For best results getting help with configuration
and troubleshooting, please use the following resources:
RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/
groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://
www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?
search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a
bug in Marlin, then you should post a [bug report](https://github.com/
MarlinFirmware/Marlin/issues/new/choose).

closed 2019-07-21 by boelle