MarlinFirmware/Marlin #14626: [BUG] (SKR v1.3 Homing-G28 Causes PID_BED_DEBUG Issues)

Mbombace, 2019-07-14:
### Description

Marlin Bugfix 2.0 Current Branch crashes (Software Crash) when homing.

### Steps to Reproduce

1. Connect Terminal
2. G28 (Home)
3. Marlin Crash (echo: PID_BED_DEBUG : Input 21.25 Output 0.00 pTerm -212.50 iTerm 0.00 dTerm 13.35)

**Expected behavior:** Printer to Home

**Actual behavior:** Marlin Crashes

#### Additional Information
[image]

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-07-19:
says BLTouch error. Check if it is configured properly and the connection is good.

boelle, 2019-09-24:
@Mbombace is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

yaoyaobutao, 2019-12-21:
I'm using the skr v1.3 board on anycubic kossel ,having the same issue when sending G28 and it keep echoing pid_bed_debug

jrprintin3d, 2020-03-03:
I`m using also the skr v1.3 board with TMC2208 driver and have the same issue at my mendel90.