MarlinFirmware/Marlin #14629: LV8728 support

jornada812, 2019-07-15:
I found and ordered drivers - lv8728, which work with 3.3 volts logic and can be used with more powerful motors, unlike lv8729. Do you need any changes in the firmware? What to write in the section steppers?

RudyBenner, 2019-07-16:
They are supported

jornada812, 2019-07-16:
> They are supported

but they are not in the list of drivers

sl1pkn07, 2019-07-16:
> > They are supported
> 
> but they are not in the list of drivers

like the ST820

RudyBenner, 2019-07-16:
My bad, not on the list of supported.

sl1pkn07, 2019-07-16:
where is the list? you mean the list in the configure.h?

jornada812, 2019-07-16:
Yes, in configuration.h. Steps for lv8729 are processed separately in stepper.h. And for lv8728? Is this actually the same driver, only the currents are different? Or am I wrong?

sl1pkn07, 2019-07-16:
same question, but about ST820 comes with Fysetc AIO II board. what driver i need set?. the configuration/_adv for the board say A4988, but is correct?

greetings