MarlinFirmware/Marlin #14630: [FR] Update Creality Ender 3/5 configuration from original source

LinoBarreca, 2019-07-15:
### Description

In the configuration folder for ender 3 they say that the configuration is based on reverse engineering from firmware because creality didn't release the source code.
There's the source code from apr.2019 on the official website. You should take the official configuration from there to be as much close to the official firmware.

### Steps to Reproduce

None.

#### Additional Information
Firmware source code at:
https://www.creality3d.cn/download/source-code_c0001

thisiskeithb, 2019-07-15:
That readme should probably be deleted, especially since the bit about U8GLIB no longer applies 🙂

As for updating from Creality’s version, that’s not the best idea since the configs in Marlin’s repo have been improved over stock.

LinoBarreca, 2019-07-15:
> As for updating from Creality’s version, that’s not the best idea since the configs in Marlin’s repo have been improved over stock.

...not really since enabling bltouch gives compilation errors complaining about SERVO0_PIN undefined.

thisiskeithb, 2019-07-15:
@LinoBarreca Can you zip your Configuration.h & Configuration_adv.h files up and attach them for review?

Which pins are you connecting the BLTouch to? Due to the limited pins on the stock boards, there are a couple options, but depending on how you connect it, you have to update your configs/pins definition for the board.

LinoBarreca, 2019-07-15:
From the original source code we can see that they defined
#define SERVO0_PIN      27 in pins_MELZI_CREALITY.h

thisiskeithb, 2019-07-15:
Did you comment out `#SPEAKER`?

LinoBarreca, 2019-07-15:
> 
> @LinoBarreca Can you zip your Configuration.h & Configuration_adv.h files and attach them for review?
> 
> Which pins are you connecting the BLTouch to? Due to the limited pins on the stock boards, there are a couple options, but depending on how you connect it, you have to update your configs/pins definition for the board.

I connect it according to the original instructions.
https://drive.google.com/file/d/1-3PtiTeqI25Coe9icmI6_saBq0BQ-Q3f/view?usp=sharing

LinoBarreca, 2019-07-15:
> 
> 
> Did you comment out `#SPEAKER`?
> 
> https://github.com/MarlinFirmware/Marlin/blob/93bc1b17f9c633643aeac6772929ae8d497a6113/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h#L56-L58

well...I took the source from 1.1.9 bugfix, moved configuration.h and configuration dv replacing defaults, activated bltouch and compiled.....

LinoBarreca, 2019-07-15:
> 
> 
> Did you comment out `#SPEAKER`?
> 
> https://github.com/MarlinFirmware/Marlin/blob/93bc1b17f9c633643aeac6772929ae8d497a6113/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h#L56-L58

What you mentioned isn't contained in 1.1.9-bugfix.

thisiskeithb, 2019-07-15:
1.1.x is no longer being developed/only receiving critical patches, so please test with the latest bugfix-2.0.x branch.

To use a BLTouch on Creality machines with a "Pin 27" board, make sure to disable `#SPEAKER`.

LinoBarreca, 2019-07-15:
> 1.1.x is no longer being developed/only receiving critical patches, so please test with the latest bugfix-2.0.x branch.
> 
> To use a BLTouch on Creality machines with a "Pin 27" board, make sure to disable `#SPEAKER`.

Man... the 1.1.9 should be the most stable version around...
I understand it is not maintained anymore but still it should compile..also I want to run it as stock as possible. original firmware is on 1.1.6 that's why I choose 1.1.9. it should be the most stable version which should be able to run original config files. Also...our ARM has very little space so 2.0 should be bigger and most likely it won't fit as easy as 1.1.x....am I wrong?

LinoBarreca, 2019-07-15:
Compiling 1.1.9 with bltouch support takes 99.7% of program space.
[image]
I think 2.0 will not fit...but I can try.

thisiskeithb, 2019-07-15:
You may have to define `SLIM_LCD_MENUS` to get enough free space.

thisiskeithb, 2019-07-15:
I just built a stock Ender-3 config from the latest bugfix-2.0.x with BLTouch enabled and SPEAKER disabled and it compiled successfully:

```
Checking size .pioenvs/sanguino_atmega1284p/firmware.elf
Building .pioenvs/sanguino_atmega1284p/firmware.hex
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  28.1% (used 4610 bytes from 16384 bytes)
PROGRAM: [==========]  97.8% (used 127230 bytes from 130048 bytes)
```

LinoBarreca, 2019-07-15:
I did it too (successfully) with 1.1.9-bugfix, stock (creality) config files and the define of pin 27.
I don't really like the custom config here because they define different parameters for movement and PID constants from creality defined ones. I understand that those here might be better but unless I have an explanation about what is modified and why I'd rather stick to stock values (I winmerged both)
Heating elements are critical (and print quality for stock is good). if prints aren't as good or something catches fire it's a problem.

LinoBarreca, 2019-07-15:
Just in case someone needs it:
Configuration.h
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010109

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// example_configurations/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
[... 1633 lines cut ...]
 *  34 RAMPS_14    : Analog input 5 on the AUX2 connector
 *  81 PRINTRBOARD : Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 RAMBO       : Analog input 3
 *
 * Note: May require analog pins to be defined for other boards.
 */
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75   // (mm) Diameter of the filament generally used (3.0 or 1.75mm), also used in the slicer. Used to validate sensor reading.

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0    // Index of the extruder that has the filament sensor (0,1,2,3)
  #define MEASUREMENT_DELAY_CM        14    // (cm) The distance from the filament sensor to the melting chamber

  #define MEASURED_UPPER_LIMIT         3.30 // (mm) Upper limit used to validate sensor reading
  #define MEASURED_LOWER_LIMIT         1.90 // (mm) Lower limit used to validate sensor reading
  #define MAX_MEASUREMENT_DELAY       20    // (bytes) Buffer size for stored measurements (1 byte per cm). Must be larger than MEASUREMENT_DELAY_CM.

  #define DEFAULT_MEASURED_FILAMENT_DIA DEFAULT_NOMINAL_FILAMENT_DIA // Set measured to nominal initially

  // Display filament width on the LCD status line. Status messages will expire after 5 seconds.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H

sl1pkn07, 2019-07-15:
archive, pliz, no paste

ManuelMcLure, 2019-07-15:
For heating parameters you should never depend on stock settings - you should always run a PID tune (M303) to get the best performance for your specific printer.

reloxx13, 2019-07-15:
orginal ender3 config was bad in past and u should rather trust marlins default example as those chinese configs...

e.g. thermal runaway, jerks values

LinoBarreca, 2019-07-15:
> 
> 
> archive, pliz, no paste

sorry. zip here.
Warning I also changed the language.
[Stock Configuration 1.1.9 with bltouch.zip](https://github.com/MarlinFirmware/Marlin/files/3393790/Stock.Configuration.1.1.9.with.bltouch.zip)

LinoBarreca, 2019-07-15:
> 
> orginal ender3 config was bad in past and u should rather trust marlins default example as those chinese configs...
> 
in past. those config files source are from april 2019. I feel safer using theirs. since I've run my printer for months with their configuration. in my job they say "if something works, don't fix it". I have to reflash for bltouch....the less I change the better.

thisiskeithb, 2019-07-15:
> in past. those config files source are from april 2019. I feel safer using theirs. since I've run my printer for months with their configuration. in my job they say "if something works, don't fix it". I have to reflash for bltouch....the less I change the better.

That's the beauty of Marlin (and open source software in general). You're free to use whichever fork fits your particular config/preference.

LinoBarreca, 2019-07-15:
>You're free to use whichever fork fits your particular config/preference.
true but then maybe the readme.md which has been deleted should be included again with what has been changed and why it is better over the stock config.

thisiskeithb, 2019-07-15:
> true but then maybe the readme.md which has been deleted should be included again with what has been changed and why it is better over the stock config.

Someone would have to maintain the list of changes over what Crealtiy has released.

Keep in mind that what Creality released doesn't match what they are shipping because they are *still* shipping printers & replacement boards with an old version of Marlin & thermal runaway protection disabled.

If you're looking for the "most up to date" version, you're going to find it within this repo since configs are updated en masse as new features are implemented.

LinoBarreca, 2019-07-15:
> If you're looking for the "most up to date" version, you're going to find it within this repo since configs are updated en masse as new features are implemented.

okay mate I'll give 2.0bugfix a try.
keep fingers crossed.

LinoBarreca, 2019-07-15:
okay. It's definitely a no-go.
With ender3 config files (BLTOUCH enabled) I see these problems:
1) I don't have the "bed-level" menu option (1.1.9 has it)
2) if I start printing from the sd card the bltouch doesn't home properly. it tries to sense the bed outside the bed itself.
The bltouch on marlin 2.0 / ender 3 is definitely configured wrong. (I don't know on 1.1.9 marlin). on 1.1.6creality it works okay.

thisiskeithb, 2019-07-15:
These aren't bugs. You need to enable those options.

> 1. I don't have the "bed-level" menu option (1.1.9 has it)

Pick one of the `AUTO_BED_LEVELING_*` options & enable them

> 2. if I start printing from the sd card the bltouch doesn't home properly. it tries to sense the bed outside the bed itself.

enable `Z_SAFE_HOMING`

I'd recommend that you give [Marlin's Automatic Bed Leveling documentation](http://marlinfw.org/docs/features/auto_bed_leveling.html) a read before proceeding further.

LinoBarreca, 2019-07-15:
...and that's why I didn't want to start with default config files.
I mean....I should be supposed to #define bltouch and all the related setting should take a working default....the maintainer of config files should set those default without leaving the advanced customization to the user (as every other setting). I'll take a look at the configuration and change it to a working default.

is there any other setting that I should look at because it's different from the default before I break the printer? (like endstop setting or so?)

LinoBarreca, 2019-07-15:
this is definitely wrong.

#define X_PROBE_OFFSET_FROM_EXTRUDER 10  // X offset: -left  +right  [of the nozzle]
the probe is 45.5mm to the left (looking at the printer, closer to 0.0)

thisiskeithb, 2019-07-15:
There are no standard probe offsets because there are many different mods that place the probe in various locations in relation to the nozzle. You have to set those options depending on which mod/mount you install on your printer.

LinoBarreca, 2019-07-15:
it makes sense but if the configuration is for ender 3 the default should be for ender 3 official BLTOUCH kit..anyway..too much hassle with configuration. I revert to stock config and 1.1.9.
Use this issue if you want to plan adapting the configuration to something more useful for the standard machine/bltouch or just close it as won't fix.

bye.

ManuelMcLure, 2019-07-15:
Or, _you_ could help the _volunteers_ that put their time and effort into providing this firmware by expending a bit of effort and producing a new, better configuration file.

LinoBarreca, 2019-07-15:
> 
> 
> Or, _you_ could help the _volunteers_ that put their time and effort into providing this firmware by expending a bit of effort and producing a new, better configuration file.

I am signaling what FOR ME is wrong and should be changed. if the project maintainer says that "we don't change 1.1.9" or "it makes no sense because there are many configurations" it means that what I think is logical and should be changed isn't as logical for the one maintaining the software...so I already put time and effort in downloading 1.1.9, changing the config to make it run properly on ender-3+standard bltouch. they said to download version 2 because it's better and more up-to-date. I did. I compiled, I flashed, I found things which didn't work "out of the box" I signaled it and it was deemed logical that it didn't work. okay then. I gave the suggestions. take them or reject them. if you reject them what am I doing here? wasting anyone's time? I go back to what I need, I fix it for my use and leave the decisions to someone else here.

have a nice day.

InsanityAutomation, 2019-07-16:
@LinoBarreca If you want something more specifically tuned and maintained, can look at the Creality branch that I maintain with Tinymachines - https://github.com/InsanityAutomation/Marlin/tree/Creality_2.0_Bleeding

It turns many options on or off depending on what actually fits on the tiny board. Its optimized to fit as many functions as possible on the board, note that this means trimming many of the options on the LCD in favor of getting the gcode behind functions.

Given that the Ender 3 is sold in versions with and without the BLTouch, maybe we could toss in a separate config for both this and the CR20.

That said, remember the original Creality versions had most safeties like thermal runaway disabled. They got a working config for one machine and most things were kept identical with only changes to build volume or other machine specific things. They are definitely not the most well tuned values out there.

thisiskeithb, 2019-07-16:
> Given that the Ender 3 is sold in versions with and without the BLTouch, maybe we could toss in a separate config for both this and the CR20.

This would be a good idea. I had planned on submitting one for the Ender-3/"Official" Creality BLTouch kit, but feel free to put in a PR if you have something. This might be tricky depending on which version of the BLTouch the user receives, but we can cross that bridge later 🙂

InsanityAutomation, 2019-07-16:
> This would be a good idea. I had planned on submitting one for the Ender-3/"Official" Creality BLTouch kit, but feel free to put in a PR if you have something. This might be tricky depending on [...]

Yeah, ive supported it with toggles in my configs since before an official creality kit existed! They saw what we were doing at ERRF last year then about a month later their kits showed up... lol

sjasonsmith, 2020-07-18:
As I read through this, I think the primary BLTouch concern has been addressed, and there isn't really anything actionable here to justify keeping it open.

If somebody has specific improvements they would like to make to examples or pins file, they can go ahead submit a pull request in the configurations repo: https://github.com/MarlinFirmware/Configurations.

closed 2020-07-18 by sjasonsmith