MarlinFirmware/Marlin #14634: problem with SKR1.3 TMC2208 & linear advance

gravit0n78, 2019-07-15:
Use:
marlin -bugfix - 2.0.x  (27.06.19)
Ender 3 pro
Skr 1.3 + TMC2208 standalone
Machine components: initial
Cura 4.1.0
Printing method: sd

Hello, please help with the problem, the extruder motor does not work during printing, what could be the problem? Skr 1.3 + TMC2208 standalone, X/Y/Z axis moves correctly, E-axis stop moving after two initial lines. 

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3393428/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3393429/Configuration_adv.txt)

teemuatlut, 2019-07-15:
Plenty of discussion about it here and elsewhere.
The current solutions are to use `SQUARE_WAVE_STEPPING` or to increase `MINIMUM_STEPPER_PULSE`.

thisiskeithb, 2019-07-15:
> The current solutions are to use `SQUARE_WAVE_STEPPING` or to increase `MINIMUM_STEPPER_PULSE`.

Would it make sense to add some conditionals/errors/warnings alerting you to enable those options if TMC steppers are used?

teemuatlut, 2019-07-16:
I'll see if I can put together a small PR this week.
Adding compile warnings is more of a philosophical question and needs approval from powers that be.

gloomyandy, 2019-07-16:
Is there a reason why MINIMUM_STEPPER_PULSE is simply not set to 1 (or 2) when using these drivers? After all we set minimum pulse values for other drivers.

gravit0n78, 2019-07-16:
> Is there a reason why MINIMUM_STEPPER_PULSE is simply not set to 1 (or 2) when using these drivers? After all we set minimum pulse values for other drivers.

I tryed the values 1/2/4,  it did not help

EvilGremlin, 2019-07-16:
Disable stealthchop or linear advance, it's widely known issue, 2208 don't like rapid direction changes and falls into protection. 
And try other slicer first, Cura is swarming with bugs. Maybe it's just trying to sent your printer to the next galaxy, judging from speeds in 4.1's gcode. Of course, printer don't like that. And generally remember: suspect Cura before anything on any occasion.

gravit0n78, 2019-07-16:
i dont use stealthchop, only  S-curve accseleration with 2208 wuthout UART/SPI, just STEP/DIR control, when LA is active - E-axis dont work

EvilGremlin, 2019-07-16:
Still don't work even in spreadcycle? Well, then your luck is bad. Use another driver, or higher voltage. This bug seems to be fixed in 2209, though i didn't try 'em on bowden yet.

LastDragon-ru, 2019-07-16:
> i dont use stealthchop ... with 2208 **wuthout UART/SPI**
> #define E0_DRIVER_TYPE TMC2208_STANDALONE

nope, you use stealthchop 😄

LastDragon-ru, 2019-07-16:
UART (recommended):
```
// Required
#define MINIMUM_STEPPER_PULSE 1 (or 2)
#define E0_DRIVER_TYPE TMC2208

// Comment to disable stealthchop (there is no other way for this...)
// #define STEALTHCHOP_E
```

If UART is not possible: 
- `#define MINIMUM_STEPPER_PULSE 1 (or 2)`
- reduce feedrates and jerks until it will work without stops (this can be done easily in slicer...)

macasero, 2019-07-27:
I am running in sort of the same problem.
With an old version of marlin 2 

SKR 1.3, 2208 uart, s-curve, LA, enabled
Old version works
New version doesn´t

[E0 2208 uart.zip](https://github.com/MarlinFirmware/Marlin/files/3437997/E0.2208.uart.zip)

einencool, 2019-07-29:
I got the same Problems with a MKS Gen L with many try&errors, although the minimum_step_pulse 1/2/4 didn‘t help.
When LA is switched off, everything works good, but the print quality is not so good. Now I switched for the extruder to an A4988 and everything works well. It‘s a little bit noisier on the extruder, but now it works great and that‘s worth it....

gravit0n78, 2019-07-29:
I finally switched for the extruder to a LV8729 and everything works , with microstep 1/128 it is equal noise level tmc2208

palladin82, 2019-07-31:
SKR 1.3 TMC2130 same problem.  

But E0 allways going backwards!
Disabling LA, changing stepper pulse, disabling stealthchop does not solve the problem.

OLD ver marlin2 works
NEWer marlin2 doesn't

ModMike, 2019-08-11:
Linear advance does not work with 2208 in stealth chop. Disable it and it works.

Anyone know if this was resolved in 2209s?

EvilGremlin, 2019-08-11:
Yeah, it's fixed in 2209. Though 2208 also works in most cases, you just need right settings.

ModMike, 2019-08-11:
I was going nuts trying to get linear advance to work about a year ago until I finally figured out it was stealth chop. I was about to give it a go with 2209s on my Fysetc Cheetah board, thank you for confirming it's been fixed.

talldonkey, 2019-08-15:
> Yeah, it's fixed in 2209. Though 2208 also works in most cases, you just need right settings. And problem isn't exactly in stealthchop, it also fails in spreadcycle.

Hi @EvilGremlin , Can you fill us in on the right settings?  Or at least settings that you know to work.  I'm hoping to find that magic setting to keep a 2208 in stealthchop with LA.  I already have square wave stepping on.

EvilGremlin, 2019-08-15:
Can't do, from what i observed it's unique settings for every motor/vlotage/load combination. I didn't got it to work in stealthchop, and i don't remember any settings.

BlackNet, 2019-08-23:
I have this very same issue on my Creality 1.1.5 'silent board' with my Ender-5.  it's using 2208 chips and I have linear_advance enabled and the value is 0.6.  I was able to do 1-2 small prints and the K test print at first but the next day it just would not print at all.  I was able to disable linear_advance and able to print again.

Also, noteworthy is on this firmware index 0 is 2208_standalone, index 1 is A4988.  I can print with E1 linear_advance set to 0

ChristoffL, 2019-08-23:
I just troubleshooted this issue on my delta. But I did not have the issue only when printing, also during manual extrusions it failed. Guess I just undefine it for now.
Also did not work in Spreadcycle or stealthchop btw.

boelle, 2019-09-24:
@gravit0n78 is the issue still there?

closed 2019-09-24 by gravit0n78

arminth, 2019-09-29:
@boelle  
This issue still persists.
I can't get TMC2208 UART working on an skr1.3 on Extruder. Spreadcycle doesn't do the trick.The moment you enable LA, any movement on Extruder is not doing anything but just enabling the stepper.
Same happens when trying to move the E-stepper manually.

arminth, 2019-09-29:
Enabling SQUARE_WAVE_STEPPING and spreadcycle now did the trick for me. But this is nothing you would come across automatically....

thinkyhead, 2019-09-29:
I have found that `SQUARE_WAVE_STEPPING` is essential, in-general. We should definitely make a point to let people know to try it if they have issues with LA and Trinamics. Meanwhile, it sounds like a step pulse issue of some kind. Did you ever experiment with the pulse durations to see if longer ones cleared things up (at possible sacrifice of top speeds)?

arminth, 2019-09-30:
Hi Scott,
I never tried it as I have also no idea, which values could be reasonable. I will try it know by increments of 2 and let you know the outcome. Will disable SQUARE_WAVE_STEPPING while testing to avoid "crosscontamination".

Cheers
Armin

arminth, 2019-09-30:
Here are my testing results:

Setup: SKRV1.3 with BTT TMC2208 UART drivers, Controller Fan enabled
Printer: Anycubic Kossel
LA: 0.27

Test Method: Increase pulse length by increment of 2 starting at 0. Try to move Extruder 1mm manually. Extruder doesn't move: NOT OK, Extruder move: OK - stop increasing pulse length
 
Test in **Spreadcycle** with SQUARE_WAVE_STEPPING disabled:
//#define MINIMUM_STEPPER_PULSE 0      NOT OK
#define MINIMUM_STEPPER_PULSE 0      NOT OK
#define MINIMUM_STEPPER_PULSE 2      OK

Test in **Stealthchop** with SQUARE_WAVE_STEPPING disabled:
//#define MINIMUM_STEPPER_PULSE 0      NOT OK
#define MINIMUM_STEPPER_PULSE 0      NOT OK
#define MINIMUM_STEPPER_PULSE 2      NOT OK
#define MINIMUM_STEPPER_PULSE 4      OK

With SQUARE_WAVE_STEPPING **enabled** and MINIMUM_STEPPER_PULSE commented out, extruder moves both in **stealthchop** and in **spreadcycle**.

Hope this helps a bit!

Cheers
Armin

talldonkey, 2019-10-01:
Hi arminth,

FWIW, I found out (a few months ago) that being able to move extruder 1mm
does not indicate that it will continue working for the duration of a
print.  I've had successful extruder moves with LA enabled,  completes the
first few layers and looks promising, but then sometime later the extruder
shuts off and starts air printing.

I currently have it working with:
*Spreadcycle* with SQUARE_WAVE_STEPPING
and
/#define MINIMUM_STEPPER_PULSE 0
(Commented out)

When I tested back then -  when I enable Stealthchop, I am able to move
extruder 1mm manually, but I can't get it to complete any meaningful print
(sometimes I can get a calicube to complete successfully, but that's not
too useful)

Setup: SKRV1.3 with FYSETC TMC2208 UART drivers, Controller Fan enabled
Printer: Ender-3
LA: 0.15

So I'm eager to see if your test results hold up to overnight-type of
print, fingers crossed marlin development in the months since has fixed it
somewhere.

blejdacik, 2019-11-05:
Hello! 
PROBLEM with linear advance. Driver tmc2208, cheetah 1.2, Titan extruder. 
Extruder do retract at end of line, then hotend moves to next line and extruder retract again. 
Video here. 
https://drive.google.com/file/d/1n5HXvY2dTtWvLYJFYXQUAlujS76-b6wh/view?usp=drivesdk
Sorry for my English but I hope you understand me. Thank you