MarlinFirmware/Marlin #14635: Is there a way to have Marlin default to a specific servo angle when starting.

wookie666, 2019-07-15:
I cant figure out where to ask this 

Is there a way i can default the servo angle to 25 when starting. 
I run an electronc speed control on a needle cutter and it doesn't initialise untill the angle is set to 25, then anything above that is the speed i want it to run at. 

Is there a way that when marlin starts, it sets the servo angle to 25 automatically. 

I thought i could use EDITABLE_SERVO_ANGLES but i have no idea how this works.

AnHardt, 2019-07-15:
As far i can remember -
No. The servos are currently initialized with the first use.
The predefined servo angles are only used with serve-probes and BL-Touch.

wookie666, 2019-07-16:
Thanks. Ill work around it

closed 2019-07-16 by wookie666