MarlinFirmware/Marlin #14636: Marlin hangs when very large feed rates are used

BDFife, 2019-07-16:
### Description

Marlin hangs ('hang' defined as the printer stopping mid-print and issuing echo:busy: processing continuously) when very large feed rates are provided. 

Cura (incorrectly) in some cases uses a default feedrate of F17987547480000 that was triggering this issue for me. While this is primarily a Cura problem, it would be nice if Marlin handled this more elegantly, or threw an error in the console.

### Steps to Reproduce

Using the Marlin bugfix-2.0.x code branch, my build was from 2019-06-26. 
A gcode file that triggers this issue is attached. 

#### Additional Information

My config files are located here: https://github.com/BDFife/MarlinConfigs/tree/master/Model3Rambo

[PS_xyzCalibration_cube_CuraFail.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/3394704/PS_xyzCalibration_cube_CuraFail.gcode.zip)

BDFife, 2019-07-16:
This issue is a dupe of #14454

reloxx13, 2019-07-16:
dupe, so close it

closed 2019-07-16 by BDFife