MarlinFirmware/Marlin #14637: How to move this custom code to marlin 2.0

wookie666, 2019-07-16:
i have this custom code 
`
 case 7: // G7 Dock Tool 
        tool_dock_undock(false);
        break;

      case 8: // G8 Undock Tool
        tool_dock_undock(true);
        break;
`

and 

`
inline void tool_dock_undock(bool undock) {
  //if undock is set to false. we are docking. otherwise we are undocking
  #define IS_EMPTY(PIN)   (!digitalRead(PIN))
  
  int tool = -1;
  int dock = -1;
  
  

  int dockPin[] = DOCKN_PIN;
  float dockX[][MOVES_IN_DOCK_SEQUENCE] = DOCK_X;
  float dockY[][MOVES_IN_DOCK_SEQUENCE] = DOCK_Y; 
  float dockFR[][MOVES_IN_DOCK_SEQUENCE] = DOCK_FR; 

  if (! (axis_known_position, X_AXIS && axis_known_position, Y_AXIS ) )
  {
   SERIAL_ECHO_START();
    SERIAL_ECHOLN("Please home before calling");
    return; // abort, since we don't know where we are
  }
  
  if (parser.seenval('T')) {
     tool = parser.value_int();
     
     
  } else {
    //if no tool is specified we send a message and exit here
    SERIAL_ECHOLN("Please specify a tool using the T parameter.");
    return;
  }
  //check that the selected tool is valid (i.e T5 is invalid if we only have two tools)
  if(tool < 0 || tool >= TOOL_NUM) {
    //warn and exit if invalid
    SERIAL_ECHO("Warning! Please pick a tool from 0 to ");
    SERIAL_ECHOLN(TOOL_NUM - 1);
    return;
  }

  

  //check if the dock is empty (if docking) or full (if undocking)
  if(undock ? !IS_EMPTY(dockPin[tool]) : IS_EMPTY(dockPin[tool])) {
    dock = tool;
  } else {
    //that tool is already docked, error and exit
    if(!undock) {
      SERIAL_ECHOLN("Tool is currently in that dock!");
    } else {
      SERIAL_ECHOLN("No tool detected in that dock!");
    }
    return;
  }

  //if dock is empty, dock there
  if(dock != -1) {

    //"Docking T0..."
    if(undock) {
      SERIAL_ECHO("Undocking ");
    } else {
      SERIAL_ECHO("Docking ");
    }
    SERIAL_ECHO("T");
    SERIAL_ECHO(tool);
    SERIAL_ECHOLN("...");
    
    
    //Lift z by 10mm or Drop z by 10mm for beds that move
    do_blocking_move_to_z(current_position[Z_AXIS]+Z_LIFT_HEIGHT, Z_LIFT_SPEED);

    
    //do blocking moves to dock

    int loopStart = (undock ? (MOVES_IN_DOCK_SEQUENCE-1) : 0);
    int loopEnd   = (undock ? 1 : MOVES_IN_DOCK_SEQUENCE-2);
    int loopInc   = (undock ? -1 : 1);
    
    for(int i = loopStart; i != loopEnd; i+=loopInc) {                  //Perform Movements in X,Y and Feedrates
      do_blocking_move_to_xy(dockX[dock][i], dockY[dock][i], dockFR[dock][i]);      
    }  

if(undock) {
  tool_change(tool);
       set_destination_from_current();
            planner.buffer_line(
                 current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS],
                 current_position[E_AXIS]+= E_PRIME, MMM_TO_MMS(E_PRIME_FEEDRATE), active_extruder );
           planner.synchronize();
           set_destination_from_current();
}
      loopStart = (undock ? (MOVES_IN_DOCK_SEQUENCE-3) : 2);
      loopEnd   = (undock ? -1 : MOVES_IN_DOCK_SEQUENCE);
      loopInc   = (undock ? -1 : 1);
    
    for(int i = loopStart; i != loopEnd; i+=loopInc) {                  //Perform Movements in X,Y and Feedrates
      do_blocking_move_to_xy(dockX[dock][i], dockY[dock][i], dockFR[dock][i]);      
    }   

                                                                     
    //Drop z by 10mm or Lift z by 10mm for beds that move                                                              
   do_blocking_move_to_z(current_position[Z_AXIS]-Z_LIFT_HEIGHT, Z_LIFT_SPEED);

    if(undock) {
      SERIAL_ECHO("Undocking done.");
    } else {
      SERIAL_ECHO("Docking done.");
    }
  }
}

/**
 * End Dock Command
 */
`

Sorry if this is the wrong space, but how would i fork this over to marlin 2.0, ive tried to find the same file and i am struggling, 
Any help would be appreciated. 

Thanks.

shitcreek, 2019-07-19:
Not sure if you've got it all sorted out, but you need to do a Pull Request.

boelle, 2019-07-21:
closing as all the OP needs to do is submitting an PR

closed 2019-07-21 by boelle

wookie666, 2019-07-21:
Sorry for the late reply. But yes sorted and pull request submitted.

boelle, 2019-07-21:
what is the PR number ?

wookie666, 2019-07-21:
It's already been sorted out and fixed.

boelle, 2019-07-21:
yes but i wanted to look up  your pull request
has it been merged?

wookie666, 2019-07-21:
Im working with the current magnetic docking feature, and made some changes to that, instead of integrating my code in which basically does the same thing #14663