MarlinFirmware/Marlin #14651: Request a Feature

tuyiwen, 2019-07-17:
If I want to further develop marlin, what software should I use?

EvilGremlin, 2019-07-17:
Whatever you can.

Ketchu13, 2019-07-17:
@tuyiwen 
Hello did your read this ?
[How to build Marlin](bugfix-2.0.x/README.md)

## Building Marlin 2.0

To build Marlin 2.0 you'll need [Arduino IDE 1.8.8 or newer](https://www.arduino.cc/en/main/software) or [PlatformIO](http://docs.platformio.org/en/latest/ide.html#platformio-ide). We've posted detailed instructions on [Building Marlin with Arduino](http://marlinfw.org/docs/basics/install_arduino.html) and [Building Marlin with PlatformIO for ReArm](http://marlinfw.org/docs/basics/install_rearm.html) (which applies well to other 32-bit boards).

RudyBenner, 2019-07-17:
https://github.com/MarlinFirmware/Marlin/issues/bugfix-2.0.x/README.md

seems to be a bad link.

boelle, 2019-07-21:
We're not a support group
This Issue Queue is for Marlin bug reports and development-related
issues, and we prefer not to handle user-support questions here. (As
noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/
1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-ijust-have-a-question).) For best results getting help with configuration
and troubleshooting, please use the following resources:
RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/
groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://
www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?
search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a
bug in Marlin, then you should post a [bug report](https://github.com/
MarlinFirmware/Marlin/issues/new/choose).

boelle, 2019-09-24:
will close this one as its not a bug

please use the following resources:

-RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
-[Tom's 3D Forums](https://forum.toms3d.org/)
-Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
-Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
-[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-09-24 by boelle