MarlinFirmware/Marlin #14652: [Q] SET_INPUT_PULLDOWN() in fastio_avr.h

Ketchu13, 2019-07-17:
Hello,
SET_INPUT_PULLDOWN () is not defined in fatio_avr.h, is this normal?
https://github.com/MarlinFirmware/Marlin/blob/523feeee3c9769b2d095f8e41b73bf3bf47f5114/Marlin/src/HAL/HAL_AVR/fastio_AVR.h#L100-L104
uncomment `#define FIL_RUNOUT_PULLDOWN` and the compilation fails indicating that the SET_INPUT_PULLDOWN definition is nonexistent.

by adding
```c++
#define SET_INPUT_PULLDOWN (IO) do {_SET_INPUT (IO); _WRITE (IO, LOW); } while (0)
```
the compilation ends successfully

AnHardt, 2019-07-17:
The AVRs don't have internal pull down resistors. So the command does not make sense.
What your define does, is only switching off the internal pull-up resistor - not activating a pull-down resistor, because it does not exist.

Ketchu13, 2019-07-17:
ok, thanks

closed 2019-07-17 by Ketchu13