MarlinFirmware/Marlin #14653: [BUG] Delta-SKR1.3-TMC2208(UART) Auto bed leveling: probe crash on last point

maxarchx, 2019-07-17:
### Description
My printer: Anycubic Kossel Linear Plus with SKR1.3-TMC2208(UART) with lastest Marlin bugfix 2.0.
Trying to perform Auto Bed Leveling with probe from LCD screen. Probing starts and hits all the points, but when it reaches last probe point it is crashes probe to bed.

### Steps to Reproduce
1. Run Auto Bed Level from LCD screen.
2. Wait for bed leveling finishes.

**Expected behavior:** Probe checks all probe points and go to home position.

**Actual behavior:** Probe checks all probe points except last one and crashes to bed.

#### Additional Information
Configs:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3401443/Marlin.zip)

Bjerg, 2019-07-17:
I am having exactly the same problem do you also use bilinear bed level or?

But i am using MKS SGEN 

my Configs:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3403852/Marlin.zip)

maxarchx, 2019-07-18:
I have tried linear and bilinear. Results are same :(
Just noticed, that bed leveling starts in wrong position, with some offset from front bed edge. About 15-30mm. Hmmm.

boubouexpress, 2019-07-18:
Modifier
#define Z_probe_low_point -2   (-8)

maxarchx, 2019-07-20:
> Modifier
> #define Z_probe_low_point -2 (-8)
Did not help.

maxarchx, 2019-07-20:
Just tried UBL bed leveling, and it failed too. It looks nice on start, and after 15-20 point checks probe begins move really weird and stucks out off bed.

maxarchx, 2019-07-20:
Fixed it.
Commented this line:
//#define Z_AFTER_PROBING            8 // Z position after probing is done

There is some problem with **move_z_after_probing()** function on delta printer.

boubouexpress, 2019-07-20:
j'ai mi a 40 au-lieu de 1 par défaut 
#define MESH_INSET 40              // Set Mesh bounds as an inset region of the bed

boubouexpress, 2019-07-20:
video

https://www.youtube.com/watch?v=ONpKxkil16Q&t=1097s

boubouexpress, 2019-07-20:
for the process of G29 p1 to work until the end I had to modify   #define MESH_INSET 45

luxflow, 2019-08-22:
I also having this issue too on 41b7f16ab84908544091882a924a80aa4201b3d6
I use LV8729 driver so I think driver doesn't affect regarding this problem

I didn't try Z_AFTER_PROBING yet, I will comment again after uncomment

luxflow, 2019-08-22:
I confirmed that after I commented //#define Z_AFTER_PROBING 8
bed leveling works

boelle, 2019-09-24:
@maxarchx this @luxflow suggest work for you too ?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle