MarlinFirmware/Marlin #14654: Marlin 2.0 MKS Sbase Home bump not working for x and y

wookie666, 2019-07-17:
Marlin 2.0, with the following setup 
```

// Homing hits each endstop, retracts by these distances, then does a slower bump.
#define X_HOME_BUMP_MM 10
#define Y_HOME_BUMP_MM 10
#define Z_HOME_BUMP_MM 5
#define HOMING_BUMP_DIVISOR { 4, 4, 4 }  // Re-Bump Speed Divisor (Divides the Homing Feedrate)
//#define QUICK_HOME                     // If homing includes X and Y, do a diagonal move initially
//#define HOMING_BACKOFF_MM { 2, 2, 2 }  // (mm) Move away from the endstops after homing

// When G28 is called, this option will make Y home before X
#define HOME_Y_BEFORE_X
```

the X and Y axis do not home bump, but the Z does. 
cant figure this one out. 
Using the latest 2.0 downloaded today. probably a setting i have wrong somewhere. 
 and this is my config file

```
// @section homing

#define NO_MOTION_BEFORE_HOMING  // Inhibit movement until all axes have been homed

//#define UNKNOWN_Z_NO_RAISE // Don't raise Z (lower the bed) if Z is "unknown." For beds that fall when Z is powered off.

#define Z_HOMING_HEIGHT 10  // (mm) Minimal Z height before homing (G28) for Z clearance above the bed, clamps, ...
                             // Be sure you have this distance over your Z_MAX_POS in case.

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

// @section machine

// The size of the print bed
#define X_BED_SIZE 235
#define Y_BED_SIZE 247

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 290

```

shitcreek, 2019-07-19:
Could you post your entire config file? Not enough info here to see the issue. Thanks.

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

wookie666, 2019-07-21:
Will do shortly

boelle, 2019-07-21:
and sorry for the close, it was a stupid mistake :-(

wookie666, 2019-07-21:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3415144/Configuration_adv.zip)
This is the current config

boelle, 2019-09-24:
@wookie666 i assume the issue is still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle