MarlinFirmware/Marlin #14655: pin maps for tronxy x5sa reveal

ftoz1, 2019-07-17:
Hi, after several days of conversation to tronxy, they send me schematics pin maps for smt32 board. It should help community to port marlin on tronxy printers
[x5sa-main_board-2.pdf](https://github.com/MarlinFirmware/Marlin/files/3401484/x5sa-main_board-2.pdf)

InsanityAutomation, 2019-07-17:
I believe this matches the Chithu controller, as it is one of their boards. The time to reverse engineer the touchscreen has been my only holdup getting into this....

ftoz1, 2019-07-17:
its not exactly same board as chitu f mini, but some versions of x5sa has it. It has different stepper driver, but maybe should be close.

ftoz1, 2019-07-19:
From what I check, pins match with pins_chitu3d.h but not sure about LCD and touch.

pinchies, 2019-07-28:
Interesting, but not unexpected - the pinout for SD and LCD is nearly identical as JGAurora A5S/A1. Only some differences in LCD backlight and chip enable/chip select pins. Wondering if the same JGAurora bootloader might work too?

ftoz1, 2019-07-28:
> Komu: MarlinFirmware/Marlin <Marlin@noreply.github.com>
> "

I cant find aurora scheme, i compare it with robin board, lcd has same 
connection as robin, sd little different, but i notice that many other 
boards had 6 pin header for programing, my board not and scheme not show 
signals on it. So havnt sure how to connect to be right.
---------- Původní e-mail ----------

(https://github.com/MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=AMUL2CYNMK7K5BNUITHG7KLQBXM2TA5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD27DNMQ#issuecomment-515782322)
"

pinchies, 2019-07-28:
JGAurora A5S reverse engineered pinout is [here](https://jgaurorawiki.com/a5s). Some errata with the unused endstop pins I haven't corrected yet, but otherwise should be mostly right.

ftoz1, 2019-07-28:
aurora code work with atmlh eeprom? Tronxy has similar one too, but i am not
sure which data it store.
---------- Původní e-mail ----------

Datum: 28. 7. 2019 20:12:32
"

(https://github.com/MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=AMUL2C7AT4GO5JYQNSA5QYLQBXOJPA5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD27DTYQ#issuecomment-515783138)
"

Keltere, 2019-12-10:
So the lcd of the tronxy can now be used with any other motherboard?

ftoz1, 2019-12-10:
Probably not, must check schematics for LCD buc connections and the board. 
If wires and connector same, then y can try it. If not match can damage LCD 
or board. Seems china LCD has no standart for that and its better to buy 
printer board with LCD if plan to run it.
---------- Původní e-mail ----------

Datum: 10. 12. 2019 13:12:22
"

(https://github.com/MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=AMUL2CYYIAH4GM7BK3ZBRUTQX6BSBA5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEGPAU5A#issuecomment-564005492)
"

xC0000005, 2020-01-29:
@ftoz1 or @InsanityAutomation, have either of you successfully initialized the Tronxy/Chitu board's LCD? I have a known working init sequence, but I suspect my FSMC configuration isn't right.

InsanityAutomation, 2020-01-29:
I havnt physically messed with the board, I have a machine sitting untouched on a shelf. I have been following @ftoz1 's progress which seems to have stopped over the holidays. It was up and working and I wouldnt mind getting into it at some point to help wrap it up when other projects get closed up.

ftoz1, 2020-02-05:
Answer isnt so easy. To start is check u8g_dev_tft_320x240_upscale_from_128x
64.cpp. there are all LCD init sequences. But there is LCD id read from 04h 
register and when LCD id is unknown, it not run init, y still see blank 
screen. in marlin 2.0.0 is on 460 line. Also is good add lines:  
SERIAL_ECHOPAIR("LCD manufacturer ID: 0x", (lcd_id & 0xF000)>>12);
  SERIAL_ECHO((lcd_id & 0x0F00)>>8);

  SERIAL_ECHO((lcd_id & 0x00F0)>>4);
  SERIAL_ECHOLN(lcd_id & 0x000F); 

It prints your lcd id into serial console and then y can add to if.

I am do my own, because marlin use ug8_lib, which is primary monochome lcd 
library. So i delete all files in lcd and rebuild all gui.
---------- Původní e-mail ----------

Datum: 29. 1. 2020 17:22:57
"

(https://github.com/MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=AMUL2C5SCCYRGNWQNPBLKD3RAGUN3A5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKH2DCY#issuecomment-579838347)
"

xC0000005, 2020-02-05:
I decompiled Chitu’s bootloader (it’s how I reverse engineered their encryption scheme), so I know every LCD it supports (there’s four or five, I don’t recall right off hand). How are you configuring FSMC and SRAM?

ftoz1, 2020-02-05:
Sorry, i am confused by community. I do my own. I dont agree with marlin 
structure. For me its too mess now. I deleted all files from lcd dir to work
clean gui. I have it finished it now, but still lot of work. connect to 
functions. Its hard undestant some things and some ppl imagine that edit few
lines and all will be work. Simply i am tired by discord and talk about. If 
people read code instead of forum, they should understand, what is hell 
work.
---------- Původní e-mail ----------

Datum: 29. 1. 2020 17:31:34
"

(https://github.com/MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=AMUL2C2CBTREL5JSA2XBMG3RAGVOBA5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKH3FDI#issuecomment-579842701)
"

ftoz1, 2020-02-05:
First i do nothing with bootloader. Y can use original, but need find adress
on stock FW, So need dump stock FW edit in hex and find gap between 
bootloader and FW. At me is 0x08008800, but tronsy x5sa is not exactly chitu
bord, look like clone of it. There need to put marlin to check by some 
serial monitor to boot right. if boot right, y must find first your lcd id 
and mod that upcale file to boot original marlin. if your LCD will work, can
mod next. My ldscrips are same as chitu ram (rwx) : ORIGIN = 0x20000000, 
LENGTH = 64K - 40
  rom (rx)  : ORIGIN = 0x08008800, LENGTH = 512K - 34K, py scrip i use from 
jgaurora

}
---------- Původní e-mail ----------

Datum: 5. 2. 2020 19:31:33

notifications&email_token=AMUL2C5SCCYRGNWQNPBLKD3RAGUN3A5CNFSM4IEPCY42YY3
PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKH2DCY#
issuecomment-579838347)
GLRAGUN3ANCNFSM4IEPCY4Q)
MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=
AHVGS4JOIJ6ZGO5NVWZS2TTRBMAMHA5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW
NIWLRBMAMHANCNFSM4IEPCY4Q>.

(https://github.com/MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=AMUL2C6QIE3KBGPEWCXKZLDRBMAX7A5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEK4PSAI#issuecomment-582547713)
"

InsanityAutomation, 2020-02-05:
It is unfortunate that some of the language barrier differences as well as some points on both sides where people were too touchy caused some issues on Discord. Its unfortunate. We all agree that code is old and has had a ton stacked on and is in need of a rewrite. It will move up the list as the 32 bit space stabilizes im sure but for now its working so things that aren't are getting attention.

That said I for one feel good progress was being made and would hate to see it lost, and would like to see it make its way in any functional form into the code base here. A good writeup of the process would be used my many im sure, even in the state is was in last time I saw it!

xC0000005, 2020-02-05:
Agreed. FTOz, I’ve already done exactly this process (If you build using the chit_f103 environment, you’ll see your linker script and mine are near identical, and the resulting update.cbd file is encrypted for disk update.) 
I’m not asking for you to explain everything, 

I simply want to know how you’re configuring FSMC and SRAM. 

Trust me, I’ve already got code running on the Tronxy and Chitu boards, either at 0x8000000 or using the proprietary bootloader. And yes, at 115200 baud, at boot, the LCD_ID function in the built in bootloader will dump the id (it also does it in the GCODE handler and a couple other places). 

If you have code of any sort, I’d be happy to read it instead of asking questions, since thus far, I’ve simply read the assembly in binaryninja to work out how things operate on that board.

tpruvot, 2020-02-05:
there are now 3 types of FSMC devices handled... st7789v ili9328 and ili9341 (all are native 320x240)

the remaining common ones have a bigger resolutions (480x320), so imo they need a specific cpp unit to handle them properly...

xC0000005, 2020-02-05:
Right - that’s why I extracted the init sequences used by the bootloader, those used by the tronxy and chitu firmware (Chitu says Tronxy customized their firmware), and compared to the manufacturer’s recommendations. Then I’ll dump the sequence into a handler similar to the others. My current one is 9844, but adding the others shouldn’t be much more work.

ftoz1, 2020-02-05:
Its true i not touch it ftom holiday, but its not lost. I dont want get out 
it now, when its not state point version. Its now gui without function 
connect, so press move do nothing and i dont want to people punish me that 
dont work. When i connect to functions, i can reveal some code, but 
sometimes i doubt if has sence. Still many problems on current marlin. I 
interestest now in RTOS marlin console to port it and start some clean.
---------- Původní e-mail ----------

Datum: 5. 2. 2020 20:06:02
"

(https://github.com/MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=AMUL2C6NS6IAFFAVMBGXVGDRBMEZNA5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEK4TBFQ#issuecomment-582561942)
"

tpruvot, 2020-02-05:
yep, ili9488 is the big new from bigtreetech, sold with the robin2

ftoz1, 2020-02-05:
its good touch this, clean hw structure, no dual system. But i am also 
little tired by this, people have plenty board and still buy some new. Dont 
know, what they find. This board maybe run FSMC by 2x, but if add asm 
routines to old board, it speed up 10x.
---------- Původní e-mail ----------

Datum: 5. 2. 2020 20:35:15
"

(https://github.com/MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=AMUL2CZUCO4XC5NGXVF47ZDRBMIG3A5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEK4WE3A#issuecomment-582574700)
"

ftoz1, 2020-02-05:
Y dont need change nothing from FSMC, that only set mcu gpios as output. 
Most important are that function there LCD_IO_WriteData, LCD_IO_WriteReg and
Multiply also. that is base of all. Those LCD works as copy memory to 
memory. By writing values y choose lcd register and write data y send data 
by bus into reg. There is nothing like framebuffer. It simply copy byte per 
byte from memory into lcd matrix.
---------- Původní e-mail ----------

Datum: 5. 2. 2020 20:32:56

AHVGS4PC6IHF3UQJ4YPMADTRBMHQXA5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW
ZDRBMHQXANCNFSM4IEPCY4Q>.

(https://github.com/MarlinFirmware/Marlin/issues/14655?email_source=notifications&email_token=AMUL2C5FBLKPPMP7OVV2ZSDRBMH6JA5CNFSM4IEPCY42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEK4V4SA#issuecomment-582573640)
"

rhapsodyv, 2020-06-01:
BOARD_CHITU3D doesn't work with X5SA?

ftoz1, 2020-06-01:
Dont know, i use some some files from jgsaurora and own pin file according 
original schematics
---------- Původní e-mail ----------

Datum: 1. 6. 2020 4:45:56
"

(https://github.com/MarlinFirmware/Marlin/issues/14655#issuecomment-636583787)
"

rhapsodyv, 2020-06-03:
@ftoz1 
https://github.com/MrPoke21/Marlin-Chitu-board-support

ftoz1, 2020-06-03:
Whats work on this? Work properly sd card and serial console? Original git 
is too much buggy and need a lot of tweaking to get solid firmware.
---------- Původní e-mail ----------

Datum: 3. 6. 2020 17:15:36
"

https://github.com/MrPoke21/Marlin-Chitu-board-support
(https://github.com/MrPoke21/Marlin-Chitu-board-support)

(https://github.com/MarlinFirmware/Marlin/issues/14655#issuecomment-638263861)
"

avion23, 2020-06-05:
> @ftoz1
> https://github.com/MrPoke21/Marlin-Chitu-board-support

Did you  try that? With my Tronxy X5SA the device stays dead after flashing.

I reenabled the script for the bootloader encryption. And I changed the print statements to print(...) to satisfy python3

ftoz1, 2020-06-05:
No, it look like fork of original github and look nothing usefull. Also 
original github is quite useless for that if not want spent years with 
twaeking. For chitu board is there https://github.com/xC0000005/Marlin/tree/
LerdgeXAndK. trupvot has alphawise git, and pinchies jgsaurora git. its 
similar hw and y can try routines from that.
---------- Původní e-mail ----------

Datum: 5. 6. 2020 13:52:18
"

" 
https://github.com/MrPoke21/Marlin-Chitu-board-support
(https://github.com/MrPoke21/Marlin-Chitu-board-support)
" 

(https://github.com/MarlinFirmware/Marlin/issues/14655#issuecomment-639434190)
"

xC0000005, 2020-06-05:
I have built the code linked below - it runs just fine as long as you flash it without a bootloader (it does not run with the Chitu bootloader). We should be able to use this as a pins definition for @jmz’s FSMC lcd code as well.

rhapsodyv, 2020-06-06:
The PINS is *almost* exactly the same of my board, but not all. I think we need more board definitions. One Chitu Board seems doesn't fit all.

ftoz1, 2020-06-06:
Schematics is base of that project. Without that ist pain and random work. I
also stuck in one point, where was mistake in schematics and thats hard 
diagnose, beacause of thousand possibilities. Most ppl want fully working 
marlin FW just now, but they dont know, which LCD type they have, which bus 
it use and where to set.
---------- Původní e-mail ----------

Datum: 6. 6. 2020 2:37:30
"

(https://github.com/MarlinFirmware/Marlin/issues/14655#issuecomment-639918858)
"

closed 2020-06-12 by thinkyhead