MarlinFirmware/Marlin #14656: [BUG bugfix-2.0.x] On AZSMZ mini and AZSMZ lcd problem if enable SD ONBOARD

xamxamxam, 2019-07-17:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

xamxamxam, 2019-07-17:
Hello
I have  AZSMZ mini and AZSMZ lcd , I tried to enable ONBOARD in SD in configuration_adv.h but the display stopped working.
I suppose a PINS definition problem ....
I rolled back the change ad the display returned ok.
Any suggestion?
Thanks

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

xamxamxam, 2019-07-23:
Hello
Here they are
Thanks a lot

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3422078/Config.zip)

boelle, 2019-09-24:
@xamxamxam is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle