MarlinFirmware/Marlin #14658: chamber LCD icon overlap

forthereg, 2019-07-17:
Thank you SOOO much for the heated chamber support and display on LCD, its awesome.
can i please just have a little bump to the left please :)
the over lap occurred both With 12864 full graphic display . AND cr10 stock display  (one cable plug)  ( i2c ? ) 
it stayed with both 1 or 2 extruders and using the alternative bed or chamber or fan icons . :)
thank you 
[image]

shitcreek, 2019-07-18:
For now, you can simply add
#define STATUS_CHAMBER_Y _ _ _  

where _ _ _  is the position you want.
define it in your custom _Statusscreen.h or in the dogm_statusscreen.h

forthereg, 2019-07-20:
> For now, you can simply add
> #define STATUS_CHAMBER_Y _ _ _
> 
> where _ _ _ is the position you want.
> define it in your custom _Statusscreen.h or in the dogm_statusscreen.h

Thank you .  i added a    #define STATUS_CHAMBER_Y 50
in dogm_statusscreen.h and it didnt seem to move.  where is that other file located ?

shitcreek, 2019-07-20:
_Statusscreen.h should be where your config file is. If you don't have one then you're using the default status screen. You can grab one from one of the examples. You'd also need to enable CUSTOM_STATUS_SCREEN_IMAGE.

 But it should work in either place (haven't it myself). Did you reflash the machine?

forthereg, 2019-07-20:
> Did you reflash the machine?

How do you reflash a machine ?    hahaha 
I got a SKR pro with no epprom , and just added an external eprom to it . i think ive done like 30 flashes in the past 4 hours .
so #define STATUS_CHAMBER_Y 50  doesnt have to go in any specific folder, i placed it in my Pins.h 
the problem was that its supposed to be chamber X and not chamber Y .
when you edit the Y coardinates it will just ignore your request and not move it.
thank you so much for your help,

forthereg, 2019-07-20:
so a paste of  #define STATUS_CHAMBER_X 55  into pins.h  and  ...
[image]

shitcreek, 2019-09-08:
We've just added fixes for this issue and some. Please try out the latest version.
#15194 #14981

closed 2019-09-08 by shitcreek