MarlinFirmware/Marlin #14672: [BUG] BigTreeTech SKR Pro - Broken compilation with new ststm32

Evg33, 2019-07-18:
platformio.ini
```
[env:BIGTREE_SKR_PRO]
#platform = ststm32
platform = ststm32@5.4.3

```

tpruvot, 2019-07-18:
what is broken ? some pastebin ?

Evg33, 2019-07-18:
```

...
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\LL\stm32yyxx_ll_usart.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\LL\stm32yyxx_ll_usb.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\LL\stm32yyxx_ll_utils.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\PortNames.c.o
CIompiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\analog.c.o
n file included from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/pinmap.h:22:0,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PeripheralPins.h:34,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\variants\BIGTREE_GENERIC_STM32F407_5X/variant.h:37,     
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\system\STM32F4xx/stm32f4xx_hal_conf.h:4,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal.h:30,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f4xx.h:250,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\stm32_def.h:35,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.h:33,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.c:30:
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:16:12: error: 'PortA' undeclared here (not in a function)
   PA_0  = (PortA << 4) + 0x00,
            ^~~~~
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:33:12: error: 'PortB' undeclared here (not in a function); did you mean 'PortA'?
   PB_0  = (PortB << 4) + 0x00,
            ^~~~~
            PortA
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:50:12: error: 'PortC' undeclared here (not in a function); did you mean 'PortB'?
   PC_0  = (PortC << 4) + 0x00,
            ^~~~~
            PortB
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:68:12: error: 'PortD' undeclared here (not in a function); did you mean 'PortC'?
   PD_0  = (PortD << 4) + 0x00,
            ^~~~~
            PortC
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:86:12: error: 'PortE' undeclared here (not in a function); did you mean 'PortD'?
   PE_0  = (PortE << 4) + 0x00,
            ^~~~~
            PortD
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\PortNames.c.o] Error 1
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c: In function 'get_adc_internal_channel':
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c:243:10: error: 'PADC_TEMP' undeclared (first use 
in this function); did you mean 'PADC_BASE'?
     case PADC_TEMP:
          ^~~~~~~~~
          PADC_BASE
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c:243:10: note: each undeclared identifier is reported only once for each function it appears in
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c:248:10: error: 'PADC_VREF' undeclared (first use 
in this function); did you mean 'PADC_TEMP'?
     case PADC_VREF:
          ^~~~~~~~~
          PADC_TEMP
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c:253:10: error: 'PADC_VBAT' undeclared (first use 
in this function); did you mean 'PADC_VREF'?
     case PADC_VBAT:
          ^~~~~~~~~
          PADC_VREF
*** [.pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\analog.c.o] Error 1
======================================================= [ERROR] Took 47.85 seconds =======================================================

=============================================================== [SUMMARY] ===============================================================  
Environment BIGTREE_SKR_PRO             [FAILED]
================================================= 1 failed, 0 succeeded in 47.99 seconds ================================================= 
Процесс терминала завершен с кодом выхода: 1
```

Evg33, 2019-07-18:
But with "platform = ststm32@5.4.3" compilation ok.

tpruvot, 2019-07-18:
yep indeed, seems to break the STM32F407xx

Evg33, 2019-07-21:
How to fix compilation with latest ststm32 ?
Now we have a patch with the old version of ststm32 in palatformio.ini.
Maybe we should fix for the new version of ststm32 ?

tpruvot, 2019-07-21:
Imo you should dig in ststm32 forums or github... and see what was changed to bug this F407VE

Evg33, 2019-07-21:
Broken at https://github.com/platformio/platform-ststm32/commit/3c27299f954a5e9b084db9e7e91c79cf9a1818aa

tpruvot, 2019-07-21:
yep, so might be the framework-arduinoststm32 version changed

Evg33, 2019-07-22:
issue: https://github.com/platformio/platform-ststm32/issues/267

Evg33, 2019-07-22:
https://github.com/platformio/platform-ststm32/issues/267#issuecomment-513671435
> Hi @Evg33 , @tpruvot ! In the 5.5.0 release, we switched to the latest STM32Duino (v1.6.1). I'm guessing the compilation fails because they changed the mechanism of how `stm32f4xx_hal_conf.h` is handled. Now there is common `stm32fxxx_hal_conf.h` per STM32 family (e.g. `system/STM32F4xx/stm32f4xx_hal_conf.h`)

Evg33, 2019-08-15:
Problem in framework-arduinoststm32
When i change "framework-arduinoststm32" from "~3.10601.190716" to "~3.10500.190222" in file 
 C:\Users\xxx\.platformio\platforms\ststm32\platform.json and platformio.ini to 
```
[env:BIGTREE_SKR_PRO]
platform = ststm32
```
Compilation ok.

Evg33, 2019-08-15:
"version": "~3.10500.190327" ok
"version": "~3.10601.190716" fail

Evg33, 2019-08-29:
@tpruvot delete tools section from buildroot/share/PlatformIO/boards/BigTree_SKR_Pro.json
and compilation ok with platform = ststm32

Thx 2 @valeros https://github.com/platformio/platform-ststm32/issues/266#issuecomment-526292630
```
[env:BIGTREE_SKR_PRO]
#platform = ststm32@5.4.3
platform = ststm32

```

Evg33, 2019-08-30:
Today I can’t repeat a successful compilation :(

tpruvot, 2019-08-30:
its normal: https://travis-ci.org/MarlinFirmware/Marlin/builds

basilfx, 2019-09-16:
@Evg33 can you try #15285?

Evg33, 2019-09-17:
@basilfx 
```
DATA:    [=         ]  13.1% (used 25756 bytes from 196608 bytes)
PROGRAM: [==        ]  15.0% (used 157596 bytes from 1048576 bytes)
BIGTREE_SKR_PRO         SUCCESS   00:03:48.461

```The board starts normally.
Thank you !

![image](https://user-images.githubusercontent.com/7160738/65039426-e98b1100-d95a-11e9-9b5e-b70922eca2d1.png)

PR #15285 (2019-09-17): Upgrade BigTree SKR Pro v1.1 to latest platform ststm32
### Description

Upgrade the `ststm32` platform to the latest version (version 5.6) by making the custom variant of the PlatformIO compatible. I think the other variants can also benefit from this, but I don't own these boards to test.

### Benefits

With this PR, we don't have to pin to an older version of the `ststm32` platform anymore. It is also necessary to make Neopixels work on my board and configuration (with #15274 applied, but without this patch, it boot-loops).

### Related Issues

Fixes #14672. Required for #15274 on the SKR P [... cut]
closed 2019-09-17 by Evg33