MarlinFirmware/Marlin #14675: [BUG] SKR Mac OS files on SD don’t work properly

spilz87, 2019-07-18:
### Description

On SKR 1.1 and 1.3 with Marlin Bugfix 2.0.x, last commit last week,

Using SD card reader on smart full screen lcd,

Gcode Files generated on MacBook Pro and copied on SD card don’t read all lines of the start code, specialty G28 and G29.

### Steps to Reproduce

1. Generate gcode file with cura on Mac osX
2. Save the file on SDcard
3. Put SD card in smart display
4. Select file
5. -> after température ok, no home done, directly mouvement for print 
-> not usable.

6. Take the SD card and connect on windows pc
7. Open the gcode file with a text editor
8. Copy all the gcode
9. Paste gcode in a new file
10. Save the new file in SD card
11. Put SD card in smart display
4. Select this new file
5. -> works fine , homing done etc.

**Expected behavior:**

Same comportement of Marlin for the same gcode data whatever if the file is generate on windows or on MacOsX

**Actual behavior:** 

Files generated on MacOsX don’t work properly 

#### Additional Information

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

spilz87, 2019-07-23:
It was an issue generate by a code I added to extract data from comments in gcode files. 
It worked on windows files because they have 2 char at end line : cr and lf, and Mac file only have one char

closed 2019-07-23 by spilz87

boelle, 2019-07-23:
oh yeah, apple knows best what you want :-D