MarlinFirmware/Marlin #14676: [BUG] [bugfix-2.0.x] Watchdog Reset when saving mesh with G29 S0

ebrithiljonas, 2019-07-19:
### Description

Marlin resets after trying to save a generated mesh with UBL.
Board: RADDS

### Steps to Reproduce

G28
G29 P0
G29 P1
G29 P3 T
G29 S0
-> Watchdog Reset

#### Additional Information

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3409022/config.zip)

jokejoke, 2019-07-19:
Maybe is the same like this https://github.com/MarlinFirmware/Marlin/issues/14575 issue?

ebrithiljonas, 2019-07-19:
I tried it know with this tree [https://github.com/MarlinFirmware/Marlin/tree/f7de453da89091d162f01ddddeb259bf8a682472](https://github.com/MarlinFirmware/Marlin/tree/f7de453da89091d162f01ddddeb259bf8a682472) but it still resets at the saving command G29 S0, I even disabled the SD Card feature

ebrithiljonas, 2019-07-19:
It works, when I lower  #define GRID_MAX_POINTS_X from 12 to 10, then I can probe and save the mesh successfully. But why doesn't it work with 12 or even 15 points?

satiromarra, 2019-07-20:
If I deactivate `SDSUPPORT` it fails, since then I have it activated even if I do not use the SD card.
I have `#define GRID_MAX_POINTS_X 10`

boelle, 2019-09-24:
@ebrithiljonas is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle