MarlinFirmware/Marlin #14680: [BUG] Filament Width Sensor on SKR1.3 shows wrong values (*5V/3.3V)

pseudex, 2019-07-19:
### Description

[Jasons Filament Sensor](https://github.com/jasonmarkham/FWS_V3.2) shows wrong values if connected to the SKR1.3 Board with the LPC1786 CPU.
The Analog Inputs on this Board can only read up to 3.3V. Marlin expects a 5V value and scales it to 5 instead of 3.3V

### Steps to Reproduce

1. Connect Filament Sensor to PIN A0[5] (1.31)
[image]
2. Activate Filament Sensor in Configuration_adv.h
3. flash Firmware
4. M407 (get reading in mm)

**Expected behavior:**
Reading in mm of the Filament
The Sensor outputs a Voltage value 1:1 to the width in mm
1.75mm -> 1.75V

**Actual behavior:** 
Reading is scaled to 5V instead of 3.3V so the diameter is shown with a factor of 1.51
1.75V -> 2.64mm

#### Additional Information

I changed the temperature.cpp in the marlin/src/module folder. This works for me. But there might be a better way to fix this issue.

```cpp
  // Convert raw Filament Width to millimeters
  float Temperature::analog_to_mm_fil_width() {
    #ifdef TARGET_LPC1768
      return current_raw_filwidth * 3.3f * (1.0f / 16383.0f);
    #else    
      return current_raw_filwidth * 5.0f * (1.0f / 16383.0f);
    #endif
  }
```

[Commit on my Server](https://gitea.kaiserflur.de/pseudex/Marlin_SKR1.3/commit/4f7b763d284355ff2fe2760fe330e9eb9ba3e5c0)

For people who want to try the sensor on their SKR1.3 Board be aware that the sensor can output up to 5V. I put a Zener Diode (3.3v) on the output to limit the voltage. Not all of the analog inputs of this CPU are 5V tolerant.

InsanityAutomation, 2019-07-19:
setting up scaling factors for min/max reading to measurement has actually been on my todo list for awhile specifically because of this issue. If someone else gets to it before me by all means!

Titi9237, 2019-08-08:
@pseudex Do you know a way to set Z_MAX_PIN as the analog pin used for the sensor ? I can not see anywhere in Marlin where to put this option for the SKR V1.3 boad.

pseudex, 2019-08-18:
@Titi9237 : I don't think, that this is possible. This doesn't look like an analog input pin.

in the pin definition file (pins_BIGTREE_SKR_V1.3.h) zmax is pin 1.24.
```c
#define Z_MAX_PIN          P1_24
```
In the datasheet on page 15 it says nothing about analog input.
[image]
[datasheet LPC1768](https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf)

the Analog inputs pins are:
- AD0[0]: P1.23: #define TEMP_BED_PIN       0   // A0 (T0) - (67) - TEMP_BED_PIN
- AD0[1]: P1.24: #define TEMP_0_PIN         1   // A1 (T1) - (68) - TEMP_0_PIN
- AD0[2]: P1.25: #define TEMP_1_PIN         2   // A2 (T2) - (69) - TEMP_1_PIN
- AD0[3]: P1.26: #define Y_MAX_PIN          P1_26
- AD0[4]: P1.30: #define BEEPER_PIN       P1_30   // (37) not 5V tolerant
- AD0[5]: P1.31: #define SD_DETECT_PIN  P1_31   // (49) (NOT 5V tolerant)
- AD0[6]: P0.3: not in pin file and not in the schematic
- AD0[7]: P0.2: not in pin file and not in the schematic

[schematic for SKR1.3 board](https://raw.githubusercontent.com/bigtreetech/BIGTREETECH-SKR-V1.3/master/hardware/SKR-V1.3-SCH.pdf)

I would suggest to use Y_MAX_PIN P1.26 if 1.31 doesn't work for you. The second temp sensor (TEMP_1_PIN) is also possible to use.
Be aware they might not be **5V tolerant**. Have a look at the datasheet of the processor.

boelle, 2019-09-24:
@pseudex still having issues?

pseudex, 2019-09-25:
I don't have this issue anymore. But I fixed this in my own fork for my printer.

boelle, 2019-09-25:
@pseudex maybe consider submitting a PR so it can be fixed in marlin?

InsanityAutomation, 2019-09-25:
> 
> 
> I don't have this issue anymore. But I fixed this in my own fork for my printer.

Or at least point us to the fork / commit and we can take it from there! Even a partial implementation is better than whats there currently before I ever get around to a full scaling setup.

pseudex, 2019-09-30:
Everything I changed I described in my initial post. Also I pointed to the commit on my fork

boelle, 2019-10-12:
@pseudex maybe submit a PR to marlin 2.0.x instead so it can get fixed for everyone

stefi01, 2019-10-23:
i used todays bugfix (23rd october) with skr v1.3, it compilled ok and uploaded to the board but the board reset itself every 30 seconds or so in a permanant boot loop kinda thing, i comented out  filament width sensor from marlin and everything works ok again, i tried my settings from marlin 1.1 and then the standard stock settings but no change

boelle, 2019-11-03:
will close this one as @pseudex do not have issues

it would have been nice if @pseudex would make a PR and send it to 2.0.x

closed 2019-11-03 by boelle

ReneSchmidts, 2019-11-09:
@boelle  Sorry, I´m a newbe here. Anyway I´m interested into this topic and got the opinion pseudex was describing his solution in detail, already - see his first comment from the 19th, July - "Additional Information". 
Isn´t it possible to use this as a solution for all?