MarlinFirmware/Marlin #14683: TMC2209 with UART causing motors clicking/ticking

asko58, 2019-07-19:
Description

Switching from TMC2209 Standalone to TMC2209 UART (Software Serial) on Marlin 2.x.x bugfix (as of 07/19/2019) with the exact same hardware and software setup (even same driver current) causes the motors to click/tick at a rate of about 2 Hz, regardless of the motor speed or mode (StealthChop, SpreadCycle). Changing the drivers' current within a range from 100mA to 1100mA will not help, higher currents make the ticking more prominent causing the motors to sometimes stall. Also, using different timing profiles will not affect the phenomenon.

Steps to Reproduce

1. install TMC2209 drivers and wire UART to software serial pins (in my case AUX2 on MKS_Gen_L, Pins 63, 40, ...)

2. Configure Marlin to use TMC2209 drivers, enabling TMC_DEBUG and MONITOR_DRIVER_STATUS, leave HybridThreshold and Sensorless Probing/homing off

3. Start printer and issue command to move an axis.

Additional Information

Playing around with various settings such as SquareWave and Acceleration/Jerk will not affect the problem. Going back to standalone mode will however solve it.
It seems to me that the drivers are receiving control signals from the board at a rate of about two per second, causing them to briefly stall and move on, causing the ticking sounds.

Config is:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3411639/config.zip)

M122 gives me this:
16:47:10.297 : X	Y	Z	E
16:47:10.297 : Address		0	0	0	0
16:47:10.324 : Enabled		false	false	false	false
16:47:10.326 : Set current	650	650	650	650
16:47:10.348 : RMS current	642	642	642	642
16:47:10.370 : MAX current	905	905	905	905
16:47:10.373 : Run current	20/31	20/31	20/31	20/31
16:47:10.376 : Hold current	10/31	10/31	10/31	10/31
16:47:10.407 : CS actual		10/31	10/31	10/31	10/31
16:47:10.435 : PWM scale	14	12	10	12
16:47:10.457 : vsense		1=.18	1=.18	1=.18	1=.18
16:47:10.478 : stealthChop	true	true	true	true
16:47:10.508 : msteps		16	16	16	16
16:47:10.528 : tstep		max	max	max	max
16:47:10.528 : pwm
16:47:10.528 : threshold
16:47:10.531 : [mm/s]
16:47:10.552 : OT prewarn	false	false	false	false
16:47:10.552 : OT prewarn has
16:47:10.555 : been triggered	false	false	false	false
16:47:10.582 : off time		4	4	4	4
16:47:10.602 : blank time	24	24	24	24
16:47:10.602 : hysteresis
16:47:10.627 : -end		2	2	2	-3
16:47:10.652 : -start		1	1	1	1
16:47:10.652 : Stallguard thrs	0	0	0	0
16:47:10.655 : DRVSTATUS	X	Y	Z	E
16:47:10.674 : stst		*	*	*	*
16:47:10.694 : olb
16:47:10.713 : ola
16:47:10.732 : s2gb
16:47:10.769 : s2ga
16:47:10.788 : otpw
16:47:10.808 : ot
16:47:10.828 : 157C
16:47:10.856 : 150C
16:47:10.875 : 143C
16:47:10.906 : 120C
16:47:10.906 : s2vsa
16:47:10.906 : s2vsb
16:47:10.908 : Driver registers:
16:47:10.914 : X	0xC0:0A:00:00
16:47:10.920 : Y	0xC0:0A:00:00
16:47:10.926 : Z	0xC0:0A:00:00
16:47:10.931 : E	0xC0:0A:00:00
16:47:10.941 : Testing X connection... OK
16:47:10.947 : Testing Y connection... OK
16:47:10.955 : Testing Z connection... OK
16:47:10.960 : Testing E connection... OK
16:47:11.035 : echo:DEBUG:INFO,ERRORS

pseudex, 2019-07-19:
I replaced also one TMC2208 with a TMC2209 from Watterott. For this I had to cut a pin and connect a jumper in an unusual way.

This is how to do it for the SKR1.3 Board.
[image]
the yellow jumper is optional and for setting the address
[image]

this is from this manual on anet3d
[how to setup SKR1.3 with watterott TMC2209](https://www.anet3d-forum.de/viewtopic.php?t=4776)

OneOfEleven, 2019-07-19:
I'm getting the same approx 2Hz step skip problem on my delta too (8-bit board), it happens to all 3 motors at the same time.

I suspect it's linked to serial comms back to the PC as that's around 2Hz at times, but not sure which settings makes the problem worse.
Enabling the **#define M114_DETAIL** option in Configuration_adv.h appears to make it worse for one.

teemuatlut, 2019-07-19:
The SW serial implementations prevent interrupts during transmissions, meaning no step pulses generated either. This leads to jerky movement.
So if we want to communicate with the driver while printing, like with Monitoring, then there's no way around it. You'll either have to turn off the feature or upgrade to a platform that doesn't use interrupts for SW serial communication.

There may be ways to alleviate the behavior however. We could try increasing the monitoring frequency from 2Hz to say 4Hz, but only poll one driver at a time.

asko58, 2019-07-20:
Ok, based on what you explain, the machine's behavior makes perfect sense. However, I suggest that there be a warning in the code (or the Marlin Wiki on Trinamic) that using software serial and having the MONITOR feature enabled will not work.
Just an idea: Would using Paul Stoffregen's AltSoftSerial library be of any use to alleviate the problem since it can communicate in parallel?

OneOfEleven, 2019-07-20:
I've just found that the controller fan is causing motor clicks for me once every 2 seconds'ish. If I disable the controller fan (comment out **#define USE_CONTROLLER_FAN**) the motor clicking vanishes and nice smooth movements return.

asko58, 2019-07-21:
I can not confirm this. I had already commented out this line before, so in my case it does not affect the stalling issues.

pseudex, 2019-08-30:
@thisiskeithb 
here is my whole marlin source for my printer: [gitea repo marlin](https://gitea.kaiserflur.de/pseudex/Marlin_SKR1.3/src/branch/MultiColor/Marlin)

shitcreek, 2019-09-06:
Is the consensus here this is a hardware issue and not a Marlin bug?

p1ngb4ck, 2019-09-11:
I have the tmc2209 running successfully on the SKR 1.3 - set up as mentioned in the above post with the fotos and info from the anet3d forum. wha could be important to say : Put in the UART-Jumper and the Jumper with the green circle only, unless you have more than 4 stepperdrivers or want to set the Master-UART Address to 1 for some other reason.
4 UART devices may be put on 1 bus - so if you have only 4, just leave them on bus 0.
If you set the jumper with the orange circle, you set that stepperdriver to use bus 1 - do this for your 5th stepperdriver and remember to modify the SLAVE_ADDRESS accordingly as mentioned in the post right above this one

dearssr, 2019-09-11:
Same situation here: the latest bugfix-2.0.x with MONITOR_DRIVER_STATUS, M114_DETAIL and USE_CONTROLLER_FAN enabled. Any movement on any axis and at any speed make a tick a couple of times per second.

Disabling only MONITOR_DRIVER_STATUS makes it smooth and quiet.
So I can confirm that this works for me.

HW: Mega2560 + RAMPS 1.4 + Bigtreetech TMC2209 UART

p1ngb4ck, 2019-09-13:
That motor clicking can be fixed by removing the SPREAD pin. Just solder it out and try again.
What I actually observed, is that a minimum stepper pulse width of 2 is still required - otherwise the motor will not move at all.
Monitor driver status, uart - everything works here on my predator with skr1.3 and watterott tmc2209. Did not test sensorless homing yet (have experience with that from a self-built printer with tmc2130 connected to mks gen 1.4 via spi

dearssr, 2019-09-15:
@p1ngb4ck, in my case the Bigtreetech drivers have different pinouts:

[image]

Instead of a SPREAD pin I have one of the PDN pins used for UART connection (factory setting).
Anyway, I've tried to disconnect both of them from the board, but it doesn't help me.

One thing I've not tried yet is to switch to the second pin...

boelle, 2019-09-24:
@asko58 still having issues?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

emaayan, 2019-11-15:
i also have this issue with TMC2208, TMC2100 and mks gen 1.4  
as is this guy, https://www.reddit.com/r/3Dprinting/comments/ce9hkx/clicking_motor_noises_after_marlin_20_update

emaayan, 2019-11-15:
> Lack of Activity
> This issue is being closed due to lack of activity. If you have solved the
> issue, please let us know how you solved it. If you haven't, please tell us
> what else you've tried in the meantime, and possibly this issue will be
> reopened.

i also tried using other TMC drivers from different versions still having this issue