MarlinFirmware/Marlin #14684: SD-EEPROM not with Bigtreetech SKR Pro(STM32F4) on-board SD card?

eightheads, 2019-07-19:
### Description

Bigtreetech SKR Pro control board(STM32F4) does not seem to hold SD-based EEPROM to the on-board card.

### Steps to Reproduce

1. Load lastest Marlin 2.0 to SKR Pro
2. Make change via LCD to z-offset, etc, and Save to EEPROM
3. Power cycle board. Settings have now reverted to initial(firmware) values

**Expected behavior:** Settings to EEPROM should save between power cycles

**Actual behavior:** Settings revert to initial firmware values after power cycle
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3412658/Configs.zip)

tpruvot, 2019-07-19:
if you sdcard is empty (no gcode file) it may cause an issue to save/load the eeprom.dat

tpruvot, 2019-07-19:
But im not sure the F4/F7 can emulate that on the sdcard, seems a F1 feature

eightheads, 2019-07-19:
> if your sdcard is empty (no gcode file) it may cause an issue to save/load the eeprom.dat
The onboard SD card just has the boards firmware on it. I do not currently have an external SD in my LCD currently.

tpruvot, 2019-07-19:
on the F4.. its emulated in the SoC flash, not on the SD

dch1921, 2019-07-20:
I'm seeing the same issue with the SKR Pro 1.1 Does not save changes to eeprom with M500

LinoBarreca, 2019-07-26:
why don't you just take the 
Marlin/Marlin/src/HAL/HAL_STM32F1/persistent_store_sdcard.cpp which is a F1 platform feature and put it "a level UP" in the /shared features among the SpiEeprom.cpp and I2CEeprom.cpp defining it as VirtualSDEeprom.cpp?

LinoBarreca, 2019-08-02:
Apparently someone has done it at the "platform" level.
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/pull/21/files
I wouldn't still merge this into marlin because I still think that EEPROM emulation should be handled at a global level and not at a platform level.

TheWeakBerlin, 2019-09-08:
Any news on that Topic?

i just bought the SKR PRO Board and it runs well with Marlin 2.0.x i can store Data with M500 on a SD-Card but after shut down the Printer it will not load the Data (Mesh Data for example) but on the SD the eeprom.dat file is still there.

Do i need to implement the "platform level" solution into my version or is there something i missed?

in Configuration.h i set EEPROM_SETTINGS 
and at pins_BIGTREE_SKR_PRO_V1.1.h i tried both SRAM_EEPROM_EMULATION and FLASH_EEPROM_EMULATION, with no difference in behavior.

Greetings from Germany :D

tpruvot, 2019-09-08:
There was a config storage version change last week end.. to V69. When that happens, the SD or FLASH (or whatever) cant be loaded... need a reset

boelle, 2019-09-24:
@eightheads still having problems?

eightheads, 2019-09-24:
> @eightheads still having problems?

As far as I know this is still an issue in the main Marlin branch.

boelle, 2019-09-24:
@eightheads have you tested it?

reloxx13, 2019-09-24:
still wont load config after restart.

TheWeakBerlin, 2019-09-27:
I still have Problems too, if i use SRAM_EEPROM_EMULATION i get the version error with V69 of the EEPROM and if i use the FLASH_EEPROM_EMULATION it only loads old data but dont want to store new data.

Evg33, 2019-10-02:
#define EEPROM_SETTINGS
#define EEPROM_CHITCHAT       // Give feedback on EEPROM commands. Disable to save PROGMEM.
#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation
no problems

dineshvyas, 2019-10-08:
[image]
[image]
THIS SHOULD WORK

eightheads, 2019-10-08:
> [image]
> [image]
> THIS SHOULD WORK

This would enable Flash based EEPROM emulation, not SD card EEPROM emulation though, correct?

dineshvyas, 2019-10-08:
> > [image]
> > [image]
> > THIS SHOULD WORK
> 
> This would enable Flash based EEPROM emulation, not SD card EEPROM emulation though, correct?

Correct.

AnHardt, 2019-10-08:
The last time i tried EEPROM from SD-card worked. 
But only from the external card-reader on the display panel - not from the on-board-slot.
But that's not a fault of the SD-EEPROM emulation but a more general on-board SD problem.

tpruvot, 2019-10-08:
There was a lot of changes recently on the data storage, we are at v70, but there was other changes in the structure since... each time its reset... Are you sure its not your problem ?

dineshvyas, 2019-10-08:
SD EEPROM is not supported on STM32F4. it is only supported on STM32F1, so not on SKR PRO V1.1

tpruvot, 2019-10-08:
some F4 seems to have it, but maybe not the SKR... due to framework differences cf #14808 & #14844

AnHardt, 2019-10-08:
@dineshvyas 
Don't be ridicules. It works on the external SD at SKR PRO 1.1!

dineshvyas, 2019-10-09:
> @dineshvyas
> Don't be ridicules. It works on the external SD at SKR PRO 1.1!

I have no such intention to ridicule. i just stated what I read. it was about on board sd card. I am not an expert, I am just looking for a solution. I apologize if by any mean I made you feel bad.

AnHardt, 2019-10-09:
The problem with "fake news" is - it spreads faster than the truth.

boelle, 2019-10-12:
@eightheads still having issues?

boelle, 2019-10-12:
> This would enable Flash based EEPROM emulation, not SD card EEPROM emulation though, correct?

correct

dineshvyas, 2019-10-12:
Did anyone try SPI SDCARD or SPI EEPROM? SPI PINS are shared for TMC Stepper Motor Drivers, I2C EEPROM can be hooked?

rflulling, 2019-10-18:
Just jumping in here. New addition to owners of one of these cards. The card has a bootloader.

My perception is that this is THEIR firmware that we cannot change. It managed your new firmware.bin file and changes the name of that file to FIRMWARE.CUR after it's been loaded. This bootloader also denies any direct access to the MCU. I have tried an STM programmer that should have been able to see the bootloader. No game, cant see a thing. 'It's the useless machine where when you press the switch to open the door a hand comes and flips the switch back.' I am sure there are more reasons why the bootloader is needed and or useful. For the moment I can only describe what I see. I have aksed for this booloader to be altered slightly so we can flash the Firmware or at least have access to the SD card in some way on the PC so we do not have to remove it at every tweak to the firmware. So far all contact has been ignored. So far while some user want this fix, many seem content to defend having to pull the card out without giving reason why its better.

I have read that on user was able to upload with BFU. But I was not able t confirm this. Attempts to use resulted in a port not found error. 

The PlatformIO.ini spells out contact with the card as being CDC, and this is reflected in windows devices. Though so far it only accepts commands from the Slicer and nothing more. 

Also keep in mind that BIQU says that their SKR PRO branch is not yet merged. While I have seen evidence that this is happening or looks like it. Certainly many changes have happened that seem to support the SKR PRO. I am not sure what all is done and how far that merger has progressed.

reloxx13, 2019-10-18:
@rflulling its working fine to mount the usb with usb_composite on.

i have it mounted on my pi with octoprint and update it with the octoprint firmware flasher plugin.
it copies the firmware to the mounted sd card and triggers a M997 update firmware command.
it will reboot and flash.

rflulling, 2019-10-21:
In this case you are placing the firmware on the External SD card (SD1), by
activating Marlin support for USB transfer to the external card, correct?
You are not talking directly to the onchip flash or the built in SD card
slot (SD0)? Once you have transferred to SD1 then by use of command M997,
it is looking for the Firmware on your SD1 rather than SD0 where the
bootloader looks for it, correct?

AnHardt, 2019-10-21:
No.
Drawing conclusions from already deleted messages, (likely because of dealing with an other board) does not make any sense.

The BTT SKR PRO bootloder can **only** deal with the **onboard SD**. If can update the firmware for the board and the WiFi board from the onboard SD. Nothing else.
Marlin (currently) can not deal with the onboard SD. (Currently) there is no way to access the onboard SD from Marlin. There is no way to place the firmware.bin on the onboard SD other than copy the files via a PC cardreader. Neither the BTT SKR PRO bootloader nor Marlin (currently) can not flash from the extarnal SD. Since some days Marlin (on the BTT SKR PRO) can reset it selfes by M997. That's a first step to conveniently flash Marlin without pressing the reset button or drawing/reseting the power plug.
To conveniently transfer any file to a accessible SDcard a USB disk device must be added to the existing USB serial device in Marlin. For the used STM core not even a example composite device exist - so it will take a while to realize this.  
The fix, builtin all STM32 processors bootloader, can not flash from any SD, but from USB and one serial, but needs to previously set BOOT0 and BOOT1 pins to a patter to do anything but calling the user program (here the BTT bootloader). Only flashing via a ST-Link adapter is accepted without a special pattern on these pins.

----

Once the boot process has left both of the bootloaders the processor has the total freedom to set about any bit any where. Writing to read only memory (ROM) is still impossible (ST bootloader). Writing to flash is always a bit more complicated but possible as long the target addresses are not write protected via a register in the processor. Non of the involved bootloaders alters one of this protection bits (even if protecting the BTT bootloader would make sense when playing with a ST-Link.).  

---------------

Back to the title of this issue.
> 
[BUG] SD-EEPROM not with Bigtreetech SKR Pro(STM32F4) on-board SD card?

Yes there is a problem. But its likely to be not the fault of EEPROM code but Marlins current inability to access the onboard SD-card at all.

This issues title should be:
"[BUG] Marlin on BigTreeTech SKR PRO can't be configured to use the onboard SD-card"

boelle, 2019-10-26:
@AnHardt that for the suggestion of a better title. Do you know what is wrong so it can be fixed?

rflulling, 2019-10-27:
@AnHardt I think if this were entirely a Marlin code fault, then a person would be able to use standard ST STM32 utilities to access the chip. But we cannot. "It is locked." The bootloader placed by BTT in factory ONLY talks to the onboard SD card (SD0). I have been asking that the bootloader be modified in some way to allow the SD card to be seen externally, even of as a stand in for the onchip flash. Something, anything, so that the card does not have to be removed each and every time. 

Some have reported using DFU and getting an upload, and I have seen a developer with SMSIS-DAP. 
-I have not gotten DFU to work yet. I have not tested SMSIS. However I did notice that a file in a recent nightly built now included a dependency for a SMSIS as part of the serial communications. So I will be tinkering soon. 

I have also read that users have been able to enable the SD card (SD1) on LCD or some other, as I have, the send the firmware file over as any uploaded G-Code. Then use the command to instruct a firmware update from SDCard. 

If I can find a way to do this. All of you Devs reading can be 100% certain I will report it. I do not like to pull the SD card every time. I know I am not alone. 
Although I feel often that as I am posting, most are resigned to giving up, and so I simply cannot drum up the support that I feel should be out there. The kind of noise that would get BTT to take notice and either give an explanation for the lock, or make a fix.

Patag, 2019-10-27:
Same here 👍 @rflulling

rflulling, 2019-10-27:
Another user/dev pointed me at this today. 
[https://github.com/stm32duino/Arduino_Core_STM32/pull/586?fbclid=IwAR1GvDz5HiDAWuQtkBA7rNXwxFFljfxs5uniqTEglpkzOGKOIOf2dSfIKCY](url)

carl1961, 2019-10-28:
your link is not working as is, so here it is. and Thanks for your work at
pushing this.
https://github.com/stm32duino/Arduino_Core_STM32/pull/586?fbclid=IwAR1GvDz5HiDAWuQtkBA7rNXwxFFljfxs5uniqTEglpkzOGKOIOf2dSfIKCY

xC0000005, 2019-10-28:
Does someone have the SKR bootloader? I'll look and see how it accesses the SD card and what state it leaves it in.

Evg33, 2019-10-28:
> Does someone have the SKR bootloader
My dump via st-link:
[fw_stlink.zip](https://github.com/MarlinFirmware/Marlin/files/3781041/fw_stlink.zip)

rflulling, 2019-10-29:
observation seems to indicate that the boot loader only looks at the onboard SD card at reset. It looks for specific files. If they are not found it disconnects the card and continues with startup. If the files are found they are installed. The previous copy is replaced, and the new one is renamed from firmware.bin to FIRMWARE.CUR. After this is done the card is disconnected.

dineshvyas, 2019-10-29:
BTT SKR_PRO is hardware to drive developers/users crazy. My on board card reader is broken due to repeated injecting and inserting. @bigtreetech if you are reading this. Do something about it.

Evg33, 2019-10-29:
@dineshvyas Thanks for the information ! I urgently need to add this:
[image]

LinoBarreca, 2019-10-29:
@Evg33 it should come bundled with the board. LOL

xC0000005, 2019-10-29:
I think the onboard SD card is attached to SPI and that PA4 is the CS pin for it, but I don’t have one of these, so I can’t modify the pins to set SS to PA4 and enable SD support (not SDIO).

Evg33, 2019-10-29:
[image]

[image]

[image]

xC0000005, 2019-10-29:
Has anyone actually tried setting the SS/SPI pins right and just using the onboard SD? We’d still need composite for access to multiple cards (or something similar to the SERIAL macros with instances of the card reader) but this kind of SD access is one of the oldest working ones on STM32.

Evg33, 2019-10-29:
Yes. Not working.

boelle, 2019-11-03:
@eightheads many updates have passed since your last reply, please retest with latest 2.0.x

eightheads, 2019-11-04:
> @eightheads many updates have passed since your last reply, please retest with latest 2.0.x

@boelle I am not currently running the SKR Pro on a printer so cannot test at the moment. However, as you can see multiple others have been having this issue still.

LinoBarreca, 2019-11-04:
@xC0000005 How is it supposed to work if stm32 lacks sdio.h/cpp? I mean...The hal doesn't have the support for it.... When I get my board I'll work on it.

xC0000005, 2019-11-04:
The HALSTM32 doesn’t have working SDIO implementation - I’ve been using it on STM32F103, STM32F070, and STM32F407 machines with the SD cards hooked up via SPI (the onboard card on the BTT SKR Pro is hooked up this way). I did get the SDIO ports on a lerdge board working using the code here -
https://github.com/xC0000005/Marlin/tree/LerdgeXAndK/Marlin/src/HAL/HAL_STM32 <https://github.com/xC0000005/Marlin/tree/LerdgeXAndK/Marlin/src/HAL/HAL_STM32>
the two files are the SDIO ones. It doesn’t use DMA yet and occasionally hits an error, but it would be a good start toward SDIO for the unified HAL.

LinoBarreca, 2019-11-04:
https://github.com/stm32duino/STM32SD
We just need to use it inside marlin...maybe that's what I'll work on next.
I'll create the interface in the generic stm32 hal to use the official stm32 SD support.

xC0000005, 2019-11-04:
That includes the parameter order fix for both F4 and F103 so that’s good a perfectly good way to use it.

LinoBarreca, 2019-11-13:
I am working on it.
Once I'm done marlin will be able to support multiple SPI buses simultaneously and on-board sd card

davidcgu, 2019-11-14:
following!!!!!

tjengbudi, 2019-11-21:
> I am working on it.
> Once I'm done marlin will be able to support multiple SPI buses simultaneously and on-board sd card

Waiting for this...

LinoBarreca, 2019-11-21:
> 
> 
> > I am working on it.
> > Once I'm done marlin will be able to support multiple SPI buses simultaneously and on-board sd card
> 
> Waiting for this...

It's on hold at the moment because I am waiting for the board.
Thing turned out more difficult than expected because the code should work "in theory", at least for the basic part, but from user reports it doesn't so I need to debug it on hardware and see what's the real problem behind it. Meanwhile you can have a look at the code [here.](https://github.com/LinoBarreca/Marlin/tree/MultiSPI) 
This afternoon I rebased the branch merging multiple commits into one. once I'll have the board here I will finish this, hopefully, in a weekend.
Since it required a "logic rewrite" of some pieces of the core (actually marlin is tied to "one controller-one spi bus" logic) it will need to be ported back to other HALs once complete.
Nothing difficult, just tedious but it will give A LOTS of advantages. For example you'll be able to have multiple "devices" (like on a computer) USBdrive/onboardSD/screenSD which can be used all together at the same time. And the thing is it will not be just limited to storage but also with sensors/drivers/displays/eeproms and other spi peripherals.

LinoBarreca, 2019-11-26:
any testers which can build and flash a debug version and can extract and upload the serial output?

LinoBarreca, 2019-11-27:
It's done! :muscle: :tada:

eightheads, 2019-11-27:
> It's done! 💪 🎉

Thanks @LinoBarreca !

[42 more replies]

closed 2020-01-07 by boelle