MarlinFirmware/Marlin #14686: confirmation request

boubouexpress, 2019-07-20:
Hello
I have a RADDS with LCD RADDS original
I have two napes of 70cm
I activate in marlin 2.0 LCD RADDS
see the picture

the length of the tablecloth may cause this problem or I have not configured correctly

merci
//=============================================================================
//============================= LCD and SD support ============================
//=============================================================================

// @section lcd

/**
 * LCD LANGUAGE
 *
 * Select the language to display on the LCD. These languages are available:
 *
 *    en, an, bg, ca, cz, da, de, el, el-gr, es, eu, fi, fr, gl, hr, it,
 *    jp-kana, ko_KR, nl, pl, pt, pt-br, ru, sk, tr, uk, zh_CN, zh_TW, test
 *
 * :{ 'en':'English', 'an':'Aragonese', 'bg':'Bulgarian', 'ca':'Catalan', 'cz':'Czech', 'da':'Danish', 'de':'German', 'el':'Greek', 'el-gr':'Greek (Greece)', 'es':'Spanish', 'eu':'Basque-Euskera', 'fi':'Finnish', 'fr':'French', 'gl':'Galician', 'hr':'Croatian', 'it':'Italian', 'jp-kana':'Japanese', 'ko_KR':'Korean (South Korea)', 'nl':'Dutch', 'pl':'Polish', 'pt':'Portuguese', 'pt-br':'Portuguese (Brazilian)', 'ru':'Russian', 'sk':'Slovak', 'tr':'Turkish', 'uk':'Ukrainian', 'zh_CN':'Chinese (Simplified)', 'zh_TW':'Chinese (Traditional)', 'test':'TEST' }
 */
#define LCD_LANGUAGE fr

/**
 * LCD Character Set
 *
 * Note: This option is NOT applicable to Graphical Displays.
 *
 * All character-based LCDs provide ASCII plus one of these
 * language extensions:
 *
 *  - JAPANESE ... the most common
 *  - WESTERN  ... with more accented characters
 *  - CYRILLIC ... for the Russian language
 *
 * To determine the language extension installed on your controller:
 *
 *  - Compile and upload with LCD_LANGUAGE set to 'test'
 *  - Click the controller to view the LCD menu
 *  - The LCD will display Japanese, Western, or Cyrillic text
 *
 * See http://marlinfw.org/docs/development/lcd_language.html
 *
 * :['JAPANESE', 'WESTERN', 'CYRILLIC']
 */
#define DISPLAY_CHARSET_HD44780 JAPANESE

/**
 * Info Screen Style (0:Classic, 1:Prusa)
 *
 * :[0:'Classic', 1:'Prusa']
 */
#define LCD_INFO_SCREEN_STYLE 0

/**
 * SD CARD
 *
 * SD Card support is disabled by default. If your controller has an SD slot,
 * you must uncomment the following option or it won't work.
 *
 */
//#define SDSUPPORT

/**
 * SD CARD: SPI SPEED
 *
[... 315 lines cut ...]
// (For CR-10 owners who want to replace the Melzi Creality board but retain the display)
//
//#define CR10_STOCKDISPLAY

//
// ANET and Tronxy Graphical Controller
//
// Anet 128x64 full graphics lcd with rotary encoder as used on Anet A6
// A clone of the RepRapDiscount full graphics display but with
// different pins/wiring (see pins_ANET_10.h).
//
//#define ANET_FULL_GRAPHICS_LCD

//
// AZSMZ 12864 LCD with SD
// https://www.aliexpress.com/store/product/3D-printer-smart-controller-SMART-RAMPS-OR-RAMPS-1-4-LCD-12864-LCD-control-panel-green/2179173_32213636460.html
//
//#define AZSMZ_12864

//
// Silvergate GLCD controller
// http://github.com/android444/Silvergate
//
//#define SILVER_GATE_GLCD_CONTROLLER
[image]

boelle, 2019-07-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

boubouexpress, 2019-07-21:
hello
merci

RudyBenner, 2019-07-21:
I see no obvious errors.

I believe you suspect the long ribbon cables could be picking up noise?
I would agree that is possible. I suggest you make a spiral of them, side by side (transposition)
to minimize EMI pickup, do not put next to other cables.

I see you are using the auto fans feature. Perhaps you can disable those lines to see if there
is a difference. One can easily put them back.

I did not check the pins file for conflicts.

I hope my english is adequate.

rudy

boubouexpress, 2019-07-21:
thank you
I am french

49/5000
I order connectors to shorten it

boelle, 2019-07-21:
i had similar issue with a full graphic display, had to shorten the wires a lot
but screened cables could also be something to test

boubouexpress, 2019-07-23:
c the length of the sheet that causes the interference.

closed 2019-07-23 by boubouexpress