MarlinFirmware/Marlin #14687: Bltouch wont auto home or level

XxZorro13xX, 2019-07-20:
The BLTouch was working fine yesterday on ender 3, then it started acting up.

When starting a print: The printer heats up the bed and nozzle like normal, then proceeds to find the center of the bed and finally moves up, deploys the bltouch (as if its going to work) then the strange thing happens.  The probe immediately pops up and the whole thing halts. 

When Auto-Homing: Find the center of the bed then proceeds to deploy the probe and once again immediately pops right back up as if it has encountered the bed.

I have even went so far as to reload the firmware onto my ender 3 but alas nothing is working, any ideas?

reloxx13, 2019-07-20:
Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

boelle, 2019-07-21:
We're not a support group

This Issue Queue is for Marlin bug reports and development-related
issues, and we prefer not to handle user-support questions here. (As
noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/
1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-ijust-have-a-question).) For best results getting help with configuration
and troubleshooting, please use the following resources:

-RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
-[Tom's 3D Forums](https://discuss.toms3d.org/)
-Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
-Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
-[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a
bug in Marlin, then you should post a [bug report](https://github.com/
MarlinFirmware/Marlin/issues/new/choose).

closed 2019-07-21 by boelle